SmingHub/Sming #166: IR-lib possible bug

fabiangl, 2015-07-07:
I have been trying out IR_lib, I use Gpio0 to read Infrared Control Remote and seems to work ok. On the other hand i use Gpio2 to send the code on Gpio2 and it doesn't work, seems not to produce the 38-40 Khz needed.
Is it possible to produce this kind of frequency with any Gpio ?

In IRremote.cpp from IR_lib they use:

void bitbangOutput(int time){
int loops = round(time/(cycle_duration));
float half_cycle = cycle_duration/2;
int i=0;
for(i=0; i<loops; i++){
digitalWrite(IR_OUT_PIN, HIGH);
delayMicroseconds(half_cycle);
digitalWrite(IR_OUT_PIN, LOW);
delayMicroseconds(half_cycle);
}
}

void IRsend::enableIROut(int khz) {
// Enables IR output. The khz value controls the modulation frequency in kilohertz.
irReadTimer.stop(); //TODO: Maybe this shouldn't be here
pinMode(IR_OUT_PIN, OUTPUT);
digitalWrite(IR_OUT_PIN, LOW); // When not sending, we want it low
cycle_duration=(float)1000/khz;
}

condemil, 2015-09-08:
I have the same issue - IRread is working fine, I can get the code of my remote and it matches the protocol, but IRsend is not, looks like bitbanging happens faster than it should, as I can't see led blinking until I increase cycle duration.

hreintke, 2016-03-25:
Closing due to no progress.

closed 2016-03-25 by hreintke