SmingHub/Sming #167: Eclipse project files & changes

0xPIT, 2015-07-08:
Please remove the Eclipse project files from the project, they bloat every changeset, it's annoying.

kireevco, 2015-07-08:
+

kireevco, 2015-07-09:
The problem with eclipse users is that they rely heavily on project settings. 

I'd like to migrate to docker/netbeans, do this way all examples will be project-less and easy to add to netbeans.

alonewolfx2, 2015-07-11:
actually its not a bug. @kireevco ?

kireevco, 2015-07-11:
@alonewolfx2 from the user prospective it is. What do you think is it?

alonewolfx2, 2015-07-11:
@kireevco i still using eclipse :)

alonewolfx2, 2015-07-11:
maybe it can generalization or improvement but not a bug.

0xPIT, 2015-07-11:
To me, its a Bug that affects Developer Experience. An improvement could arise only when you change the actual code, not the IDE project files. They should simply not be in the repo.

kennpie, 2015-07-13:
Not a bug. More an expression of user preference.

I'm using the CHERTS 'UDK'  so I've found it convenient to have eclipse project files with the sming examples. I also like Netbeans.

Maybe the project files could be placed in a separate directory (one for eclipse, one for netbeans, one for vi ;-)  )?

kireevco, 2015-07-27:
@kennpie what should happen is that sming example should not have __any__ project files. There should be a skeleton Example for Eclipse and Netbeans project on the other hand.

alonewolfx2, 2015-07-28:
@kireevco still it doesnt a bug you know that.

anakod, 2015-08-04:
What problem with eclipse projects is?
May be just add to global ignore project properties files what isn't needed for work.

kireevco, 2015-08-18:
relates to #167

hreintke, 2015-11-29:
No updates for a long time.
If issue still exist please submit a new issue

closed 2015-11-29 by hreintke