SmingHub/Sming #170: last commit for telnet not working

alonewolfx2, 2015-07-11:
```
19:29:26 **** Build of configuration Sming for project Telnet_TCPServer_TCPClient ****
make rebuild 
C+ app/application.cpp
app/application.cpp:4:41: fatal error: SmingCore/Network/TcpStream.h: No such file or directory
 #include <SmingCore/Network/TcpStream.h>
                                         ^
compilation terminated.
make: *** [out/build/app/application.o] Error 1

19:29:27 Build Finished (took 1s.115ms)
```

hreintke, 2015-07-14:
@alonewolfx2 @hrsavla : The include should have been removed from the application.
I will submit a PR for that.
If you want to use before that just remove the one line
```
#include <SmingCore/Network/TcpStream.h>
```

alonewolfx2, 2015-07-14:
ok i will try that

kireevco, 2015-07-27:
@alonewolfx2 Is this working now for you?

alonewolfx2, 2015-07-28:
Yes. Issue solved.

closed 2015-07-28 by alonewolfx2