SmingHub/Sming #171: How to include external libs?

Loooopy, 2015-07-12:
Hi,

I am trying to use the compiled u8glib.a (with include file u8g.h) from nodemcu, but it fails when linking,
the following error i can see in the console: "...xtensa-lx106-elf/bin/ld.exe: cannot find -lu8glib".

Where should i declare it so it can be found?

I have added the line below into the "makefile-user.mk" to succefully compile it:
EXTRA_LIB= u8glib

but linking is still a problem, any clue what I am missing?

kireevco, 2015-08-23:
Sorry for a delayed response. @Loooopy can you show us where do you put your library? The error means that the linker can't find a library.

hreintke, 2015-10-05:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-05 by hreintke

alonewolfx2, 2015-10-05:
put libraries into the sming/libraries and compile it. you dont need to add extra libs flag.