SmingHub/Sming #172: How to feed WDT -  WDT.alive() doesn't seems to work

robotiko, 2015-07-12:
How can I prevent a long running task from resetting device due to the wdt?
I have a loop and I call the WDT.alive()  but doesn't seem to work at all. 
Tried on sdk 1.1.1 and 1.2.

Loooopy, 2015-07-13:
Check out this: http://tech.scargill.net/esp8266-debug/
Since Espressif SDK 1.01 there are some api changes.

void pp_soft_wdt_stop();    // close software watchdog
void pp_soft_wdt_restart();    // reset software watchdog

robotiko, 2015-08-10:
DIrect call to Espressif  SDk seem to work (at least with SDK 1.2.0)  but the WDT.alive() doesn't.

kireevco, 2015-08-23:
@Loooopy @robotiko any findings on this?

robotiko, 2015-08-23:
@kireevco, stated in the last comment. 
WDT.alive () doesn't work, but  there is a direct SDK solution  that works for SDK 1.2.0  
calling system_soft_wdt_stop();  before the long operation 
and system_soft_wdt_restart(); every  x time  (ie on each cycle on a long loop)
works.

kireevco, 2015-08-23:
@robotiko perfect summary, thank you.
But should't long operations be avoided as much as possible? Not doing so will block everything.

robotiko, 2015-08-23:
@kireevco 
Sure. But if needed, WTC.alive() should work as expected or not be available at all.
The point is that current implementation doesn't work and the functionality is present

hreintke, 2015-11-29:
No update for a long time
Please retest in current develop branch and if issue persists submit a new issue.

closed 2015-11-29 by hreintke

alonewolfx2, 2015-11-29:
@hreintke @robotiko this issue solved with this commit https://github.com/SmingHub/Sming/commit/bffdfc51c80c9e5917a8d19f1d3be4c843bb38b3