SmingHub/Sming #174: DNS record not found

flexiti, 2015-07-13:
After 2 and more hours of work DHT example reading loop ( every 2 minutes) I have : 

8:17:52.338, connect to: api.thingspeak.com
8:17:52.338, realloc 121 -> 252
8:18:05.993, DNS record _not_ found: api.thingspeak.com
8:18:05.993, -TCP connection
8:18:05.993, Failed

Using direct IP work all the time

kireevco, 2015-07-27:
@flexiti Please provide us your code and your device type that you are using.

Thank you

flexiti, 2015-07-28:
Now I have another symptoms (more code?), in any case, as you get an error like this below- from this moment each call is completed with error until  a soft reset.
it happens sometimes after an hour and sometimes several.
In my opinion it is associated with errors in the network (I have many devices on my WiFi)
I'll try to prepare in my free time some sources simplified to suit the tests. I use 3 pieces:ESP01 + DHT22 ,  ESP01 + BS1820 and ESP12 +BS1820+ BMP180 . In all the symptom it is similar, generally hard for me to get continuous work more than 24 hours.
Also I will try to get a more extensive log.

Debug with error:
==============================================
Download: http://api.thingspeak.com:80/update?key=YOXULSNC3XXXXXXX&field6=23.9&field2=52.7&field3=13&status=dev_ip:192.168.0.188
+TCP connection
connect to: api.thingspeak.com
realloc 131 -> 262
MB_TCPConnection_DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
MB_HttpClient_onFinished Write error code: 0, Mode: 0, Finish state: 4, Body: 

onDataSent  --> ResponseCode: 0
onDataSent  --> ResponseString: 
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: none
onDataSent  --> ResponseState: 4
onDataSent  --> Successful: Error

MB TcpConnection error: -8
MB_TCPConnection_staticOnError err

Normal debug:
============================
Download: http://api.thingspeak.com:80/update?key=YOXULSNC3RXXXXXX&field6=23.9&field2=52.9&field3=11&status=dev_ip:192.168.0.188
+TCP connection
connect to: api.thingspeak.com
realloc 131 -> 262
MB_TCPConnection_DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 133
onReadyToSendData: 2
***************************
MB_HttpClient_onReceive buf:
HTTP/1.1 200 OK
Server: nginx/1.7.5
Date: Tue, 28 Jul 2015 14:20:35 GMT
Content-Type: text/html; charset=utf-8
Connection: close
Vary: Accept-Encoding
Status: 200 OK
X-Frame-Options: ALLOWALL
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers: origin, content-type, X-Requested-With
Access-Control-Max-Age: 1800
ETag: "82dee7f6ff2493db007197ab9591bf7a"
Cache-Control: max-age=0, private, must-revalidate
X-Request-Id: 40c3816d-8cf1-4ebb-81f8-528d929ba2b0

68359
***************************
MB_HttpClient_onReceive code: 200
MB HttpClient_onReceive Header pos: 546
Server === nginx/1.7.5
Date === Tue, 28 Jul 2015 14:20:35 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "82dee7f6ff2493db007197ab9591bf7a"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === 40c3816d-8cf1-4ebb-81f8-528d929ba2b0
TCP received: 555 bytes
onReadyToSendData: 1
MB HttpClient_onReceive buf = NULL
TCP received: (null)
MB_HttpClient_onFinished Write error code: 200, Mode: 0, Finish state: 3, Body: 

Send Time   --> 28.07.2015 16:20:37
onDataSent  --> ResponseCode: 200
onDataSent  --> ResponseString: 68359
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: nginx/1.7.5
onDataSent  --> ResponseState: 3
onDataSent  --> Successful: Success 

TCP connection closing
-TCP connection

flexiti, 2015-07-29:
Hi,

I will publish source on GitHub.

Can you tell me how best to publish it on Git and  not publish Git Wi-Fi passwords,...... ?

I'll be grateful for the tip or best practices.

 

 

Best Regards and thanks

 

Mariusz

anakod, 2015-07-29:
> Can you tell me how best to publish it on Git and  not publish Git Wi-Fi passwords
You can define ssid and password variables globally in Eclipse Environment settings for all your projects.

flexiti, 2015-07-29:
I tried before  but the compiler does not see
[image]

anakod, 2015-07-29:
I think you set it in Environment section

flexiti, 2015-07-29:
ok, thanks

flexiti, 2015-07-29:
I found problem, the test device is DHT22 and ESP01,
 Loop with two servers every 2 minutes conection to other
Connection to a server that does not exist, and then connection to the server thingspeak and from that moment the connection with thingspeak is no longer working (always finishState = 4)

Can anyone verify this?

flexiti, 2015-07-30:
My log file, before and after non existing server connection (non existing server marked as KEIL) 

THINGSPEAK
Humidity: 48.70 %	Temperature: 23.20 *C
*http*api.thingspeak.com*80*/update*?key=YOXULSNC3RXXXXXX&field6=23.2&field2=48.7&status=dev_ip:192.168.0.188*

Download: http://api.thingspeak.com:80/update?key=YOXULSNCXXXXXX&field6=23.2&field2=48.7&status=dev_ip:192.168.0.188
connect to: api.thingspeak.com
realloc 121 -> 252
MB_TCPConnection_DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result: 0
OnConnected
TCP connected
onReadyToSendData: 0
MB_TcpCoonection_flush
MB_TcpCoonection_flush
TcpClient request completed
TCP sent: 123
onReadyToSendData: 2
***************************
MB_HttpClient_onReceive buf:
HTTP/1.1 200 OK
Server: nginx/1.7.5
Date: Thu, 30 Jul 2015 14:08:04 GMT
Content-Type: text/html; charset=utf-8
Connection: close
Vary: Accept-Encoding
Status: 200 OK
X-Frame-Options: ALLOWALL
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers: origin, content-type, X-Requested-With
Access-Control-Max-Age: 1800
ETag: "5fb42dc17861645234c3bbcd9e5ab69e"
Cache-Control: max-age=0, private, must-revalidate
X-Request-Id: 3573fe88-e539-4511-9dbb-0c44e70f9614

70988
***************************
MB_HttpClient_onReceive code: 200
MB HttpClient_onReceive Header pos: 546
Server === nginx/1.7.5
Date === Thu, 30 Jul 2015 14:08:04 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "5fb42dc17861645234c3bbcd9e5ab69e"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === 3573fe88-e539-4511-9dbb-0c44e70f9614
TCP received: 555 bytes
onReadyToSendData: 1
MB HttpClient_onReceive buf = NULL
TCP received: (null)
MB_TcpClient_close state=2
MB_HttpClient_onFinish finishState=3
MB_HttpClient_onFinished Write error code: 200, Mode: 0, Finish state: 3, Body: 

Send Time   --> 30.07.2015 16:08:08
onDataSent  --> ResponseCode: 200
onDataSent  --> ResponseString: 70988
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: nginx/1.7.5
onDataSent  --> ResponseState: 3
onDataSent  --> Successful: Success 
Free heap size = 23832

MB_TcpClient_onFinished state=3
MB_TcpClient_onFinished finishState=3
TCP connection closing
-TCP connection
 
KEIL
Humidity: 48.80 %	Temperature: 23.20 *C
*http*192.168.0.98*1002*/u*?k&f6=23.2&f2=48.8&s*

Download: http://192.168.0.98:1002/u?k&f6=23.2&f2=48.8&s
+TCP connection
connect to: 192.168.0.98
TcpConnection::connect result: 0
realloc 57 -> 188
MB_TcpClient_onError err=-8
MB_HttpClient_onFinish finishState=4
MB_HttpClient_onFinished Write error code: 0, Mode: 0, Finish state: 4, Body: 

Send Time   --> 30.07.2015 16:09:09
onDataSent  --> ResponseCode: 0
onDataSent  --> ResponseString: 
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: none
onDataSent  --> ResponseState: 4
onDataSent  --> Successful: Error
Free heap size = 25376

MB_TcpClient_onFinished state=4
MB_TcpClient_onFinished finishState=4
MB TcpConnection error: -8
MB_TCPConnection_staticOnError err=-8
 
THINGSPEAK
Humidity: 49.00 %	Temperature: 23.10 *C
*http*api.thingspeak.com*80*/update*?key=YOXULSNC3XXXXXXXX&field6=23.1&field2=49.0&status=dev_ip:192.168.0.188*

Download: http://api.thingspeak.com:80/update?key=YOXULSNCXXXXXXX&field6=23.1&field2=49.0&status=dev_ip:192.168.0.188
+TCP connection
connect to: api.thingspeak.com
realloc 121 -> 252
MB_TCPConnection_DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result: 0
OnConnected
TCP connected
onReadyToSendData: 0
MB_TcpCoonection_flush
MB_TcpCoonection_flush
TcpClient request completed
TCP sent: 123
onReadyToSendData: 2
***************************
MB_HttpClient_onReceive buf:
HTTP/1.1 200 OK
Server: nginx/1.7.5
Date: Thu, 30 Jul 2015 14:10:04 GMT
Content-Type: text/html; charset=utf-8
Connection: close
Vary: Accept-Encoding
Status: 200 OK
X-Frame-Options: ALLOWALL
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers: origin, content-type, X-Requested-With
Access-Control-Max-Age: 1800
ETag: "b0221b0f49d4249158f04fcc993dbe26"
Cache-Control: max-age=0, private, must-revalidate
X-Request-Id: c4c4db17-6969-4b26-804e-923525812da0

70990
***************************
MB_HttpClient_onReceive code: 200
MB HttpClient_onReceive Header pos: 546
Server === nginx/1.7.5
Date === Thu, 30 Jul 2015 14:10:04 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "b0221b0f49d4249158f04fcc993dbe26"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === c4c4db17-6969-4b26-804e-923525812da0
TCP received: 555 bytes
onReadyToSendData: 1
MB HttpClient_onReceive buf = NULL
TCP received: (null)
MB_TcpClient_close state=2
MB_HttpClient_onFinish finishState=4
MB_HttpClient_onFinished Write error code: 0, Mode: 0, Finish state: 4, Body: 

Send Time   --> 30.07.2015 16:10:09
onDataSent  --> ResponseCode: 0
onDataSent  --> ResponseString: 70990
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: nginx/1.7.5
onDataSent  --> ResponseState: 4
onDataSent  --> Successful: Error
Free heap size = 23760

MB_TcpClient_onFinished state=4
MB_TcpClient_onFinished finishState=4
TCP connection closing
-TCP connection

flexiti, 2015-07-30:
Btw. Previous logs are direct IP address, when I use http://api.thingspeak.com still have problem with DNS after one or few hours. If it hapens, continues until reset.
I don't know if it is the same source of problem. So I think  at the beginning we need to concentrate on the error after  of non-existent server connections

DNS error example.
------------------------------------------------------------
THINGSPEAK
Humidity: 49.40 %	Temperature: 22.60 *C
*http*api.thingspeak.com*80*/update*?key=YOXULSXXXXXXXXXXX&field6=22.6&field2=49.4&status=dev_ip:192.168.0.188*

Download: http://api.thingspeak.com:80/update?key=YOXULSNCXXXXXXXXX&field6=22.6&field2=49.4&status=dev_ip:192.168.0.188
+TCP connection
connect to: api.thingspeak.com
realloc 121 -> 252
MB_TCPConnection_DNS record _not_ found: api.thingspeak.com
-TCP connection
MB_TcpClient_close state=1
MB_HttpClient_onFinish finishState=4
MB_HttpClient_onFinished Write error code: 0, Mode: 0, Finish state: 4, Body: 

Send Time   --> 30.07.2015 17:25:03
onDataSent  --> ResponseCode: 0
onDataSent  --> ResponseString: 
onDataSent  --> PostBody: 
onDataSent  --> ResponseHeader: none
onDataSent  --> ResponseState: 4
onDataSent  --> Successful: Error
Free heap size = 26000

MB_TcpClient_onFinished state=4
MB_TcpClient_onFinished finishState=4

flexiti, 2015-08-03:
Simple bug example

https://github.com/flexiti/HttpClient_Bug

hreintke, 2015-08-03:
@flexiti : I just tried your Simple bug example and noticed the following : 
This is the debug output from first request : 
```
  I'm CONNECTED
Download: http://api.thingspeak.com:80/update?key=7XXUJWCWYTMXKN3L&field1=1
connect to: api.thingspeak.com
realloc 78 -> 209
DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 80
onReadyToSendData: 2
Header pos: 546
Server === nginx/1.7.5
Date === Mon, 03 Aug 2015 15:37:14 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "7fb8ceb3bd59c7956b1df66729296a4c"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === e73895cd-cab3-46c5-8d10-ee1f2efe3cb0
TCP received: 554 bytes
onReadyToSendData: 1
TCP received: (null)
ThingSpeak server connection: Success sent
Server response: '1485'
==============
TCP connection closing
-TCP connection
```
Then I see the the request to the non-exsisting server : 
```
Download: http://192.168.0.8:80/update?key=7XXUJWCWYTMXKN3L&field1=4
+TCP connection
connect to: 192.168.0.8
TcpConnection::connect result:, 0
realloc 71 -> 202
Non existing server connection: Failed
Server response: ''
==============
TCP connection error: -8
```
And the following request to thingspeak : 
```
Download: http://api.thingspeak.com:80/update?key=7XXUJWCWYTMXKN3L&field1=5
+TCP connection
connect to: api.thingspeak.com
realloc 78 -> 209
DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 80
onReadyToSendData: 2
Header pos: 546
Server === nginx/1.7.5
Date === Mon, 03 Aug 2015 15:38:54 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "6d3a1e06d6a06349436bc054313b648c"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === 8355c175-aa5d-4b3d-9b3f-3aa48914e705
TCP received: 554 bytes
onReadyToSendData: 1
TCP received: (null)
ThingSpeak server connection: Failed
Server response: '1487'
==============
TCP connection closing
-TCP connection
```
Although the onDataSent(HttpClient& client, bool successful), is called with bool = false (not successful). It looks to me that the actual communication is working correctly.
Server response from the first : Server response: '1486'
Server response form the second : Server response: '1487'

Can you verify that ? Is a big difference in looking for the underlying issue.
Disclaimer : I am not sure whether I can solve the issue but trying to find the reason

flexiti, 2015-08-03:
Yes, Ansver from thingspeak server is ok (or only looks ok?), but Sming-HttpClient interpreting is as a fail. Why ? I suppose some variables that are not properly initialized before the start of the connection or previous defective call something 'destroys'

hreintke, 2015-08-03:
I've seen this code in the httpserver_bootstrap example.
```
	if (downloadClient.isProcessing()) return; // Please, wait.

	if (downloadClient.isSuccessful())
		dowfid++; // Success. Go to next file!
	downloadClient.reset(); // Reset current download status
```
So there is an addtional call to reset() after completion.

Could you update your "simple bug program" to : 
```
void onDataSent(HttpClient& client, bool successful)
{

	if (server)
	  Serial.print("ThingSpeak server connection: ");
	else
	  Serial.print("Non existing server connection: ");

	if (successful)
		Serial.println("Success sent");
	else
		Serial.println("Failed");

	String response = client.getResponseString();
	Serial.println("Server response: '" + response + "'");
	if (response.length() > 0)
	{
		int intVal = response.toInt();

		if (intVal == 0)
			Serial.println("Sensor value wasn't accepted. May be we need to wait a little?");
	}

	if (server)
		server = false;
	else
		server = true;
	thingSpeak.reset();

	Serial.println("==============");
}
```
Notice the added thingSpeak.reset().
And test again (I cannot do that at this moment) and see whether that gives other results ?

flexiti, 2015-08-03:
Problem is deeper (not HttpClient class but maybe  in TCPconnection class )
Bool Successfull is false because TcpClient.onFinished return state=4. (normal finish state is 3)

ps. thingSpeak.reset nothing changes

hreintke, 2015-08-04:
I don't know whether this is a separate issue or the cause of the above but I see in 
```
bool TcpConnection::internalTcpConnect(IPAddress addr, uint16_t port)
{
	NetUtils::FixNetworkRouting();
	err_t res = tcp_connect(tcp, addr, port, staticOnConnected);
	debugf("TcpConnection::connect result:, %d", res);
	return res == ERR_OK;
}
```
It returns the result of tcp_connection but that will also return true if the connection fails due to abort/timeout/other  (is statement from lwip documentation)
If there is a connection error there is an internal callback tcp_err which should handle this but that only sends debug message and closes tcpclient.

flexiti, 2015-08-04:
For me this is weird:
when you connect to a nonexistent server, we have an error (it's obvious), 
but.. 
next there is a good server connection, (even obtain the correct answer from the server !!!!) but still error in status

flexiti, 2015-08-04:
I found a solution to  "not existing server" problem. (https://github.com/flexiti/HttpClient_Bug )
 it's probably solves few other problems. 
On network  error, the "asyncTotalLen" remained  not cleared
What is your opinion?

SimngCore-Network-TcpClient.cpp
----------------------------------------------
void TcpClient::onError(err_t err)
{
	      asyncTotalLen = 0;  //this should be added in my opinion
	      asyncTotalSent = 0; // and this ?? 

	state = eTCS_Failed;
	onFinished(state);
	TcpConnection::onError(err);
}

hreintke, 2015-08-04:
First : I was wrong in my statement above on connect, sorry for the confusion, but..
I have found the cause of the issue.
Sming saves the "send strings" in an internal buffer and only later does the actual send on tcp, counts the "chars to be sent" (asyncTotalLen) and "chars actual send" (asyncTotalSent) to decide later whether all went OK.
At TcpClient Close, these are both reset to zero. 
However, when the connect fails, there count "chars to be sent" has been increased but "chars actual sent" not (cannot be sent) and also tcpclose is not called (it is never "opened") -> no reset of values. 
With the next connection, "chars to be sent" is increased again, "chars send" also with the same amount -> they still are unequal at the end -> Sming decides to give state = unsuccessful.
You also see that wen you do : 
thingspeak connect -> OK
non-existing connect -> Fail
thingspeak connect -> Fail
thingspeak connect -> OK

This is TcpClose code : 
```
void TcpClient::close()
{
	debugf("TCPClient close : state = %d, Sent = %d, Len = : %d", state,asyncTotalSent,asyncTotalLen);
	if (state != eTCS_Successful && state != eTCS_Failed)
	{
		state = (asyncTotalSent == asyncTotalLen) ? eTCS_Successful : eTCS_Failed;
		asyncTotalLen = 0;
		asyncTotalSent = 0;
		onFinished(state);
		debugf("TcpClient Close : asyncTotal reset, state = %d", state);
	}

	// Close connection only after processing
	TcpConnection::close();
}
```
This is debuglog : 
```
  I'm CONNECTED
Download: http://api.thingspeak.com:80/update?key=7XXUJWCWYTMXKN3L&field1=1
connect to: api.thingspeak.com
TCPClient Send : asyncLen = 78, len = 78, asyncSent = 0
realloc 78 -> 209
TCPClient Send : asyncLen = 80, len = 2, asyncSent = 0
TCPClient Send : asyncLen = 80, len = 0, asyncSent = 0
DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCPClient onSent : asyncSent = 80, len = 80, asyncLen = 80
TCP sent: 80
onReadyToSendData: 2
Header pos: 546
Server === nginx/1.7.5
Date === Tue, 04 Aug 2015 11:03:43 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "21fe5b8ba755eeaece7a450849876228"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === 2e00bd08-47ee-4468-b5fe-a6b869bdd779
TCP received: 554 bytes
onReadyToSendData: 1
TCP received: (null)
TCPClient close : state = 2, Sent = 80, Len = : 80
ThingSpeak server connection: Success sent
Server response: '1633'
==============
TcpClient Close : asyncTotal reset, state = 3
TCP connection closing
-TCP connection
Download: http://192.168.0.8:80/update?key=7XXUJWCWYTMXKN3L&field1=2
+TCP connection
connect to: 192.168.0.8
TcpConnection::connect result:, 0
TCPClient Send : asyncLen = 71, len = 71, asyncSent = 0
realloc 71 -> 202
TCPClient Send : asyncLen = 73, len = 2, asyncSent = 0
TCPClient Send : asyncLen = 73, len = 0, asyncSent = 0
Non existing server connection: Failed
Server response: ''
==============
TCP connection error: -8
Download: http://api.thingspeak.com:80/update?key=7XXUJWCWYTMXKN3L&field1=3
+TCP connection
connect to: api.thingspeak.com
TCPClient Send : asyncLen = 151, len = 78, asyncSent = 0
realloc 78 -> 209
TCPClient Send : asyncLen = 153, len = 2, asyncSent = 0
TCPClient Send : asyncLen = 153, len = 0, asyncSent = 0
DNS record found: api.thingspeak.com = 184.106.153.149
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCPClient onSent : asyncSent = 80, len = 80, asyncLen = 153
TCP sent: 80
onReadyToSendData: 2
Header pos: 546
Server === nginx/1.7.5
Date === Tue, 04 Aug 2015 11:04:32 GMT
Content-Type === text/html; charset=utf-8
Connection === close
Vary === Accept-Encoding
Status === 200 OK
X-Frame-Options === ALLOWALL
Access-Control-Allow-Origin === *
Access-Control-Allow-Methods === GET, POST, PUT, OPTIONS, DELETE, PATCH
Access-Control-Allow-Headers === origin, content-type, X-Requested-With
Access-Control-Max-Age === 1800
ETag === "6dd4e10e3296fa63738371ec0d5df818"
Cache-Control === max-age=0, private, must-revalidate
X-Request-Id === 51d5cb0b-0c23-447e-b44f-de73c24100df
TCP received: 554 bytes
onReadyToSendData: 1
TCP received: (null)
TCPClient close : state = 2, Sent = 80, Len = : 153
ThingSpeak server connection: Failed
Server response: '1634'
==============
TcpClient Close : asyncTotal reset, state = 4
TCP connection closing
-TCP connection
```
Knowing the reason, now to solve. I will work on that but from tomorrow on I am out for a couple of days, so solution might take a little longer.

flexiti, 2015-08-04:
read my last post :)

hreintke, 2015-08-04:
Yes I saw it wen I pushed the "comment" button. 
Same conclusion.

hreintke, 2015-08-04:
Or for solution add the
asyncTotalLen = 0; //this should be added in my opinion
asyncTotalSent = 0; // and this ?? 
in the TcpConnect function.
You might you still want to know the values after tcp_err

hreintke, 2015-08-04:
And we need to check whether the buffer/stream is deleted correctly otherwise there will be memory leakage

flexiti, 2015-08-04:
yes,can You ? ..... and reset  in  TcpConnect must be done twice

bool TcpClient::connect(String server, int port)
and
bool TcpClient::connect(IPAddress addr, uint16_t port)

hreintke, 2015-08-04:
Maybe best location is in TcpClient::onfinished.
That is called from tcpclient::onerror and is also the location where stream is deleted and set to nullptr.
Then all re-initialization is in one location.

hreintke, 2015-08-04:
@flexiti : I submitted a PR to solve this to the latest idea that we had.
I am out for the coming days, please feel free to update this PR or submit another if wished or necessary

flexiti, 2015-08-07:
This problem is solved, now only #200

closed 2015-08-07 by flexiti