SmingHub/Sming #175: Can't get the cheap SSD1306 from aliexpress running with sming

Loooopy, 2015-07-13:
Hi,

I have tried to get the SSD1306 running in sming but I dot get any output.
(http://www.aliexpress.com/item/1Pcs-Yellow-blue-double-color-128X64-OLED-LCD-LED-Display-Module-For-Arduino-0-96-I2C/32219359544.html)

I am using the ESP8266 ESP-01 and can see that in the wiki it should not work (Only ESP-03), but in the code i cant see why it should not.

In nodemcu i get output on the display but it uses u8g lib there.

Tavalin, 2015-07-23:
Can this be closed? 

I use an I2C version of that display on an ESP-12 and it works fine.  If you're using an SPI display I imagine the ESP-01 does not have enough pins.

kireevco, 2015-08-23:
Closing this for now.

closed 2015-08-23 by kireevco