SmingHub/Sming #176: undefined reference to `system_get_rst_info' in latest builds

piperpilot, 2015-07-14:
I'm having problems with the latest builds.  I have rolled back commits and found it to be related to commit https://github.com/anakod/Sming/commit/1c376f5c64cd22ae3b46347c5fbf51303f4fb5f9 when I compile anything other than basic_blink I get the following:

```
09:08:06 **** Build of configuration Sming for project HttpServer_ConfigNetwork ****
make all 
C+ app/application.cpp
C+ /opt/sming/Sming/appinit/user_main.cpp
AR out/build/app_app.a
LD out/build/app.out
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x8): undefined reference to `vtable for WebSocket'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0xc): undefined reference to `WebSocket::WebSocket(HttpServerConnection*)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x10): undefined reference to `WebSocket::initialize(HttpRequest&, HttpResponse&)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x31): undefined reference to `WebSocket::WebSocket(HttpServerConnection*)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o): In function `HttpServer::initWebSocket(HttpServerConnection&, HttpRequest&, HttpResponse&)':
C:\DEV\ESP8266\Sming\Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::initialize(HttpRequest&, HttpResponse&)'
/opt/sming/Sming/compiler/lib//libsming.a(CommandHandler.o):(.text._ZN14CommandHandler19procesStatusCommandE6StringP13CommandOutput+0x20): undefined reference to `system_get_rst_info'
/opt/sming/Sming/compiler/lib//libsming.a(CommandHandler.o): In function `CommandHandler::procesStatusCommand(String, CommandOutput*)':
C:\DEV\ESP8266\Sming\Sming/Services/CommandProcessing/CommandHandler.cpp:70: undefined reference to `system_get_rst_info'
/opt/sming/Sming/compiler/lib//libsming.a(RTC.o): In function `RtcClass::RtcClass()':
C:\DEV\ESP8266\Sming\Sming/SmingCore/Platform/RTC.cpp:26: undefined reference to `system_get_rst_info'
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

09:08:07 Build Finished (took 1s.425ms)
```

I'm on a MAC and obviously there's no C:\DEV so it looks like something got compiled with an absolute path?  I look at that commit and see it has a new libsming.a file...probably something wrong with that.  I'll try to compile libSming myself, but wanted to call out this problem so it could get fixed...everything worked prior to that commit.

piperpilot, 2015-07-14:
When I try to compile the framework itself I get:

```
09:12:42 **** Build of configuration Sming for project SmingFramework ****
make all 
AS system/irq_check.s
CC system/flashmem.c
CC Services/SpifFS/spiffs.c
CC Services/SpifFS/spiffs_cache.c
CC Services/SpifFS/spiffs_check.c
CC Services/SpifFS/spiffs_gc.c
CC Services/SpifFS/spiffs_hydrogen.c
CC Services/SpifFS/spiffs_nucleus.c
CC Services/libemqtt/libemqtt.c
CC Libraries/Adafruit_GFX/glcdfont.c
CC Libraries/Adafruit_ILI9341/hspi.c
C+ system/esp_cplusplus.cpp
C+ system/stringconversion.cpp
C+ Wiring/FakePgmSpace.cpp
C+ Wiring/I2cMaster.cpp
C+ Wiring/IPAddress.cpp
C+ Wiring/Print.cpp
C+ Wiring/SplitString.cpp
C+ Wiring/Stream.cpp
C+ Wiring/WMath.cpp
C+ Wiring/WShift.cpp
C+ Wiring/WString.cpp
C+ SmingCore/Clock.cpp
C+ SmingCore/DataSourceStream.cpp
C+ SmingCore/Digital.cpp
C+ SmingCore/ESP8266EX.cpp
C+ SmingCore/FileSystem.cpp
C+ SmingCore/HardwareSerial.cpp
C+ SmingCore/Interrupts.cpp
C+ SmingCore/PWM.cpp
C+ SmingCore/SPI.cpp
C+ SmingCore/SystemClock.cpp
C+ SmingCore/Timer.cpp
C+ SmingCore/Wire.cpp
C+ appinit/user_main.cpp
C+ SmingCore/Network/FTPServer.cpp
C+ SmingCore/Network/FTPServerConnection.cpp
C+ SmingCore/Network/HttpClient.cpp
C+ SmingCore/Network/HttpFirmwareUpdate.cpp
C+ SmingCore/Network/HttpRequest.cpp
C+ SmingCore/Network/HttpResponse.cpp
C+ SmingCore/Network/HttpServer.cpp
C+ SmingCore/Network/HttpServerConnection.cpp
C+ SmingCore/Network/MqttClient.cpp
C+ SmingCore/Network/NetUtils.cpp
C+ SmingCore/Network/NtpClient.cpp
C+ SmingCore/Network/TcpClient.cpp
C+ SmingCore/Network/TcpConnection.cpp
C+ SmingCore/Network/TcpServer.cpp
C+ SmingCore/Network/TelnetServer.cpp
C+ SmingCore/Network/URL.cpp
C+ SmingCore/Network/UdpConnection.cpp
C+ SmingCore/Network/WebSocket.cpp
C+ SmingCore/Platform/AccessPoint.cpp
C+ SmingCore/Platform/RTC.cpp
SmingCore/Platform/RTC.cpp: In constructor 'RtcClass::RtcClass()':
SmingCore/Platform/RTC.cpp:4:39: error: 'system_get_rst_info' was not declared in this scope
  rst_info* info = system_get_rst_info();
                                       ^
SmingCore/Platform/RTC.cpp:5:25: error: 'struct rst_info' has no member named 'reason'
  hardwareReset = (info->reason == REASON_WDT_RST);
                         ^
SmingCore/Platform/RTC.cpp:5:35: error: 'REASON_WDT_RST' was not declared in this scope
  hardwareReset = (info->reason == REASON_WDT_RST);
                                   ^
make: *** [out/build/SmingCore/Platform//RTC.o] Error 1

09:12:51 Build Finished (took 9s.280ms)
```

Tavalin, 2015-07-14:
@piperpilot The system_get_rst_info is related to your SDK version, try updating your SDK to V1.1 or higher to solve this.  The C:\DEV thing seems like a red herring, you may have another issue as that shouldn't be related to system_get_rst_info.

piperpilot, 2015-07-14:
I tried compiling with:

esp_iot_sdk_v1.1.0_15_05_26
esp_iot_sdk_v1.2.0_15_07_03

Both have the same problem.

Its actually manifesting itself in HttpServer examples on the latest build as as:

```
19:54:50 **** Build of configuration Sming for project HttpServer_ConfigNetwork ****
make all 
C+ app/application.cpp
C+ /opt/sming/Sming/appinit/user_main.cpp
AR out/build/app_app.a
LD out/build/app.out
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x8): undefined reference to `vtable for WebSocket'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0xc): undefined reference to `WebSocket::WebSocket(HttpServerConnection*)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x10): undefined reference to `WebSocket::initialize(HttpRequest&, HttpResponse&)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o):(.text._ZN10HttpServer13initWebSocketER20HttpServerConnectionR11HttpRequestR12HttpResponse+0x31): undefined reference to `WebSocket::WebSocket(HttpServerConnection*)'
/opt/sming/Sming/compiler/lib//libsming.a(HttpServer.o): In function `HttpServer::initWebSocket(HttpServerConnection&, HttpRequest&, HttpResponse&)':
C:\DEV\ESP8266\Sming\Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::initialize(HttpRequest&, HttpResponse&)'
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1
```

If I roll back to older builds everything is fine.

Tavalin, 2015-07-15:
@piperpilot Could you see if https://github.com/anakod/Sming/commit/e47d5c27d5ab61ddf5e41b45f07efe1f0add7b63 fixes the above error for you? You'll still need to use an SDK at 1.1 or higher in order to resolve the system_get_rst_info issue.

piperpilot, 2015-07-15:
Both problems still exist.  I'm using the 1.1.0 SDK (esp_iot_sdk_v1.1.0_15_05_26) and have moved to HEAD on Master... clean and compile.

Building Sming, it doesn't complain about "no member named 'reason'" any more, but it still complains about the missing declaration:

```
C+ SmingCore/Platform/RTC.cpp
SmingCore/Platform/RTC.cpp: In constructor 'RtcClass::RtcClass()':
SmingCore/Platform/RTC.cpp:4:39: error: 'system_get_rst_info' was not declared in this scope
  rst_info* info = system_get_rst_info();
                                       ^
SmingCore/Platform/RTC.cpp:5:35: error: 'REASON_WDT_RST' was not declared in this scope
  hardwareReset = (info->reason == REASON_WDT_RST);
                                   ^
make: *** [out/build/SmingCore/Platform//RTC.o] Error 1
```

The websocket reference error is the same.

Tavalin, 2015-07-15:
The message seems to suggest your not using the right SDK or your paths are somehow pointing to an incorrect SDK.  Could you give a bit more info about your setup/environment please?

piperpilot, 2015-07-15:
Sure thing...First off, I'm on OSX 10.10.3.  I installed the SDK from the package listed here:

https://github.com/anakod/Sming/wiki/MacOS-Quickstart

Just to make sure I did everything properly.  I think downloaded the 1.1.0 SDK and symlinked it over.

```
Curtiss-MBP:esp-open-sdk cpope$ ls -la
total 19696
drwxr-xr-x  11 cpope     staff       374 Jul 14 19:49 .
drwxr-xr-x@  8 transfer  wheel       272 Jul 13 22:40 ..
-rw-r--r--   1 cpope     staff      1847 May 26 09:58 License
drwxr-xr-x  15 cpope     staff       510 Jul 14 17:07 esp_iot_sdk_v1.0.1
drwxr-xr-x  11 cpope     staff       374 May 26 11:38 esp_iot_sdk_v1.1.0
-rw-r--r--   1 cpope     staff  10057897 Jul 14 19:48 esp_iot_sdk_v1.1.0_15_05_26.zip
drwx------@  6 cpope     staff       204 Jul 14 12:11 esp_iot_sdk_v1.2.0_15_07_03
drwxr-xr-x   8 cpope     staff       272 Apr 25 21:48 esptool
-rw-r--r--   1 cpope     staff     14805 May 26 11:49 release_note.txt
lrwxr-xr-x   1 cpope     staff        18 Jul 14 19:49 sdk -> esp_iot_sdk_v1.1.0
drwxr-xr-x   9 cpope     staff       306 Apr 25 21:48 xtensa-lx106-elf
```

I double checked the code and its looking at ESP_HOME which is set for /opt/esp_open_sdk in my case...inside that its looking for sdk which is what the symlink is set to.

I can't see that its using a wrong SDK version but I must be wrong.  I'm building from eclipse and if I roll back about 10-15 commits the websocket error goes away, but I still have the system_get_rst_info error.  If anyone has OSX and has something I can check, I'm happy to share the info.  I don't really need to build the core Sming...if we can get past the websocket build issue, I'd be happy.  I have dug into that and can't see any issues, must be something simple I'm missing.  OSX is a case sensitive OS...but it looks like all the includes are proper case.

kireevco, 2015-07-15:
@piperpilot how do you set ESP_HOME and SMING_HOME? How do you run your build?

piperpilot, 2015-07-15:
ESP_HOME and SMING_HOME are set via the plist in eclipse.app per the quick start.  I just double checked and re-added them.  See the screenshot of my Info.plist file.  I also tried overriding SMING_HOME and ESP_HOME in the main Makefile to ensure it was getting picked up properly with the same results.

[image]

Build is done from the eclipse target.

I have tried to confirm it is using the proper SDK by doing the following:

-  Removed all old versions
-  Tested...same errors

kireevco, 2015-07-15:
Let's test it with a command line. Please run `make` from the project root.

piperpilot, 2015-07-15:
@kireevco Looks like we were typing at the same time...per my previous EDIT above...I tested from the command line with same results.

piperpilot, 2015-07-23:
Any more suggestions guys?  I'm still having both of the issues above...

1.  Can't compile SmingFramework... I get the system_get_rst_info error
2.  Can't compile most examples, get "undefined reference" errors with respect to WebSocket

I pulled the latest master and the problem is still there.  If I roll back to July 10 commit: 266df44f70065bcf2c3016f356da80cdde4f5fea everything compiles fine for the examples, but I still get the same error on SmingFramework.

Tavalin, 2015-07-23:
I've just looked through the SDK changes...you need to make sure you're using v1.1.1 or higher (i.e. not v.1.1.0).  I'm still 99% sure that the system_get_rst_info part is related to that.

piperpilot, 2015-07-23:
OK...I tried again with a fresh download of 1.2.0, symlinked /opt/esp-open-sdk/esp_iot_sdk_v1.2.0 to /opt/esp-open-sdk/sdk... did a clean of the sdk... made sure all paths were correct and it compiled fine.  This also fixed the example projects.  I'm not sure what the magic was...the only thing different I did this time was a "make clean" inside the sdk directory.  But its working, so I will close the issue.  Thanks everyone for the help!

closed 2015-07-23 by piperpilot

lgcpovoas, 2015-08-09:
Hello there, can you please tell me how to compile esp-open-sdk_v1.2.0? or do you have a direct download link for mac version? I cant get its compilation to go through

kireevco, 2015-08-09:
@lgcpovoas did try https://bintray.com/artifact/download/kireevco/generic/esp-open-sdk-1.2-macos-x86_64.tar.gz ?

lgcpovoas, 2015-08-10:
oh man, thanks. Worked like a charm.
What a pain to 'learn' to build these packages!

kireevco, 2015-08-10:
@lgcpovoas building those packages is a pain. Can't say better. Glad it helped!