SmingHub/Sming #177: Please confirm what version of Espressif SDK is required for Sming

kennpie, 2015-07-14:
Hi,

Please give your 'official' blessing on what version of the Espressif SDK will work with Sming

The readme.md still states
Based on Espressif SDK v1.0.1b1

But elsewhere (eg http://www.esp8266.com/viewtopic.php?p=22445#p22445) in response to my post above it, it's implied that Sming can deal with the latest SDK.

I tried a while back to upgrade to v1.1 of the SDK, but stuff broke, so I retreated to the safety of 1.0.1b1, and here I sit, til you say it's ok to upgrade. ;-)

kireevco, 2015-07-14:
Until we move on with basic testing and ci that is hard to do. Please vote under "stability" ticket here and under "move to community group" ticket.

kennpie, 2015-07-25:
I'm sorry for not replying sooner.

> *Please vote under "stability" ticket here and under "move to community group" ticket.*

I'm a github n00b; what does this mean?

Anyway, I remain concerned that Sming is not keeping up with the espressif SDK releases. The latest **three** releases from espressif (non_OS, mesh, RTOS) seem potent; what will this mean to Sming?

I appreciate that there isn't yet CI, and testing is a lot of effort... but if you were to post some guidelines and goals, and make a public call for testers, then maybe it would get done

I'm also concerned that I haven't seen much recent forum activity regarding Sming. It's still my favourite for the ESP8266, but all the community love seems to be going to the Arduino port and Lua. How can we raise the profile and activity around Sming?

anakod, 2015-07-26:
By some reason separated Sming thread still wasn't created on esp8266.com forum.
Already many many peoples ask about that. I think will be better to ask forum admin about that.

kennpie, 2015-07-28:
Richard has just created a new forum area for Sming on esp8266.com: http://www.esp8266.com/viewtopic.php?f=35&t=4251

kennpie, 2015-08-07:
After some extensive testing, (ok, not really, I just tried it) it has been empirically determined
determined that Sming compiles just fine with the espressif SDK 1.20. And despite what the Sming README says, it probably doesn't compile any more with 1.01b.

closed 2015-08-07 by kennpie