SmingHub/Sming #180: spiff fails adding files and flash leaves the device memory corrupt when adding many files

robotiko, 2015-07-16:
Adding several files in the device (over 200KB),  spiffs gives errors  while adding last files.
However the build completes without errors but flashing the device, the filesystem gets corrupt and cannot fin any file when running the program. 
Tested on a 4MB device, looks like the issue is in the spiff rom size creation that is fixed before adding the files: "Creating rom spiff_rom.bin of size 196608 bytes".

kireevco, 2015-07-17:
@robotiko please provide a `ls` / `dir` of the 'webFiles' directory, as well as output of your build process. Also can you confirm where spiffy.exe is taken from.

Thank you.

alonewolfx2, 2015-07-18:
@robotiko also we have new spiffy and you can change spiffs size from commandline. for example "spiffy 404800 files" and you can use more than 200kb if you have 4MB flash.

kireevco, 2015-08-23:
Please use spiffy from @alonewolfx2 https://github.com/alonewolfx2/spiffy

or my binary build of it:
https://bintray.com/kireevco/generic/spiffy/view
closing this for now

closed 2015-08-23 by kireevco