SmingHub/Sming #181: Access to WebSocket for sending messages

hreintke, 2015-07-16:
When using the websocket callbacks, they all have a "WebSocket" parameter.
This Websocket parameter however is not valid outside the callback function due to the use of the vector "websocketlist" within httpserver & websocket.
Therefore the socket involved cannot be saved locally and it is not possible to send messages which are not a reply to an incoming message.

MrAPierce, 2015-07-17:
HttpServer has a method getActiveWebSockets, usage as below, I use this in a timer callback, could be used where ever you like.

	WebSocketsList &clients = server.getActiveWebSockets();
	for (int i = 0; i < clients.count(); i++)
	{
		clients[i].sendString("hello");
		uint8_t data[5] = { 1,2,3,4,5 };
		clients[i].sendBinary(data,sizeof(data));
	}

Is this what you after?

hreintke, 2015-07-19:
@MrAPierce : No this does not solve the issue. In this way you only send to all active websockets.
I want to send to a specific socket as they all could act differently

hreintke, 2015-07-19:
@anakod : As I also want to use it within sming core (CommandProcessing and later Debug output) I checked whether I could use : 
```
if (wsConnect) wsConnect(wsocks[wsocks.indexOf(sock)]);
```
But got compiling errors which could not solve.
Solution I found was : Update Vector class with : 
```
template <class Element>
void Vector<Element>::addElement(Element* objp)
{
  if (_size == _capacity)
    ensureCapacity(_capacity + _increment);
  if (_size < _capacity)
    _data[ _size++ ] = objp;
};
```
And then update to HttpServer::initWebSocket : 
```
bool HttpServer::initWebSocket(HttpServerConnection& connection, HttpRequest& request, HttpResponse& response)
{
	if (!wsEnabled)
		return false;

	WebSocket *sock = new WebSocket(&connection);
	if (!sock->initialize(request, response))
		return false;

	connection.setDisconnectionHandler(HttpServerConnectionDelegate(&HttpServer::onCloseWebSocket, this)); // auto remove on close
	response.sendHeader(connection); // Will push header before user data

	wsocks.addElement(sock);
	if (wsConnect) wsConnect(*sock);
//	if (wsConnect) wsConnect(wsocks[wsocks.indexOf(*sock)]);

	if (request.getQueryParameter("command") == "true")
	{
		commandExecutor = new CommandExecutor(sock);
	}

	return true;
}
```
What is your opinion on this solution ?

ADiea, 2015-08-24:
this will couple websockets and commandexecutor. IMHO we should keep things as decoupled as possible in the framework and tie them together as needed in individual applications

closed 2015-11-12 by hreintke