SmingHub/Sming #185: [MqttClient] does not publish message

STASIUK, 2015-07-23:
MqttClient:
Not working method for publishing a message without timer.
Ex:
void publishMessage(String text)
{
	if (mqtt.getConnectionState() != eTCS_Connected)
		startMqttClient(); // Auto reconnect

	Serial.println("Let's publish message now!");
	mqtt.publish("main/frameworks/sming", text); // or publishWithQoS
}

publishMessage("hello world");  <- does not publish

Give me some example.

Tavalin, 2015-07-23:
You need to give a bit more info from the logs.  Has your module connected to the wifi network? Has the module connected to the broker?

STASIUK, 2015-07-23:
connected with PG, channel 11
dhcp client start...
ip:192.168.1.67,mask:255.255.255.0,gw:192.168.1.1
I'm CONNECTED
MQTT start connection
connect to: test.mosquitto.org
timeout updating: 70 -> 65535
subscription 'main/status' registered
realloc 23 -> 170
MQTT start connection
timeout updating: 65535 -> 65535
subscription 'main/status' registered
MQTT start connection
timeout updating: 65535 -> 65535
subscription 'main/status' registered
Let's publish message now!
realloc 123 -> 308
DNS record found: test.mosquitto.org = 85.119.83.194
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 179
onReadyToSendData: 2
> MQTT status: MQTT_MSG_CONNACK (len: 4)
TCP received: 4 bytes
onReadyToSendData: 1
TCP connection error: -9

What is wrong here?

STASIUK, 2015-07-23:
A possible small example a publishing messages without timer?

anakod, 2015-07-23:
-9 is ERR_RTE

Did you complete waiting succesfull WiFi connection before start publishing?

hreintke, 2015-10-05:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-06 by STASIUK