SmingHub/Sming #189: Example of receive and parse JSON via Serial

piperpilot, 2015-07-26:
Does anyone have a good example of how to receive and parse JSON received via serial from another microcontroller?  Idea being that it can be manipulated/stored and then served up via http server.

anakod, 2015-07-26:
It should be same as parse JSON from file. PLease look https://github.com/anakod/Sming/blob/master/HttpServer_ConfigNetwork/include/AppSettings.h example

piperpilot, 2015-07-26:
Thanks...yeah, I got that much...I'm more interested in if anyone has an efficient way to detect incoming JSON via the serial stream and then hand it off to the parser when the full message has been received.  I am working on a method to look at characters... for every { increase a counter and every } decrease it...then when the counter is zero...take the message.

Anyone have a better solution?

anakod, 2015-07-26:
I think this example should be useful for input stream: https://github.com/anakod/Sming/blob/master/Basic_Serial/include/SerialReadingDelegateDemo.h
You can check input stream for NULL char and than split reading.

avr39-ripe, 2015-07-27:
https://github.com/avr39-ripe/ValveControl/blob/master/app/application.cpp#L52
Here is my dirty, but fully working code to receive and parse some json from arduino. Feel free to ask anything if you find it useful :)

kireevco, 2015-07-27:
I like that.

@avr39-ripe hey, do you think you could submit an example for Sming that will contain two codebases: sming and arduino?

kireevco, 2015-08-23:
@avr39-ripe we need your heroic help here! :)

avr39-ripe, 2015-08-23:
I will try to compose such example from pieces of my own test code..

kireevco, 2015-08-23:
This will be very very helpful. Thank you!

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke