SmingHub/Sming #190: Heap overflow by usage of JsonObjects

Bennedetto, 2015-07-27:
There seems to be an problem with the consistency of *JsonObjects* (or/and *JsonObjectStreams*). This Example is a simplification of the Example *HttpServer_ConfigNetwork* (compare to its *AppSettings.h*). The problem also occurs all over the usage of *JsonObjectStreams*. The heap will overflow after a short amount of time (depending of the amount of used memory of each instance of the Json-Stuff and of course how often you use the JsonObjects).

```
#include <SmingCore/SmingCore.h>

Timer procTimer;

void forceError()
{
	JsonObjectStream* stream = new JsonObjectStream();
	//the line above will block ~200 Bytes of Heap forever.

	JsonObject& json = stream->getRoot();

	JsonArray& netlist = json.createNestedArray("nestedArray");

	for (int i = 0; i < 20; i++)
	{
		JsonObject &item = netlist.createNestedObject();
		item.add("id", i);
		item.add("title1", "value1");
		item.add("title2", "value2");
		item.add("title3", "value3");
	}
	Serial.print("New Heap size: "); Serial.println(xPortGetFreeHeapSize());
}

void init()
{
	Serial.begin(115200 );
	Serial.systemDebugOutput(false); // Enable debug output to serial
	procTimer.initializeMs(1000, forceError);
	procTimer.start();
	Serial.println("done");
}
```

hreintke, 2015-08-03:
@Bennedetto : I think the issue is not in JsonObject(stream) but in your application.
I updated the forceError to : 
```
void forceError()
{
    JsonObjectStream* stream = new JsonObjectStream();
    //the line above will block ~200 Bytes of Heap forever.

    JsonObject& json = stream->getRoot();

    JsonArray& netlist = json.createNestedArray("nestedArray");

    for (int i = 0; i < 20; i++)
    {
        JsonObject &item = netlist.createNestedObject();
        item.add("id", i);
        item.add("title1", "value1");
        item.add("title2", "value2");
        item.add("title3", "value3");
    }
    delete stream;
    Serial.print("New Heap size: "); Serial.println(xPortGetFreeHeapSize());
}
```
Notice the added statement :  delete stream.
Your forceError creates a new object :  JsonObjectStream* stream = new JsonObjectStream();
But never returns the claimed memory for that.

Can you try this and report back ?

Bennedetto, 2015-08-09:
You are right. I took the wrong example - sorry. The problem can been seen at the SMING Example "HttpServer_ConfigNetwork". As soon as the AppSettings.save() method gets called, the heap will be filled and never freed because (in my beginner s mind) of the addCopy - instructions .. (if i remove that line, the heap stays intact

This is the correct example:

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

void save() {
	DynamicJsonBuffer jsonBuffer;
	JsonObject& root = jsonBuffer.createObject();
	JsonObject& network = jsonBuffer.createObject();
	root["network"] = network;
	network["gateway"] =  "222.111.333.444";

	network.addCopy("ip", "111.222.333.444"); //this line will leak heap..

	//not necessary to get the problem
	//fileSetContent("myOwnFileName.conf", root.toJsonString());
}

Timer dbT;
void debugOut() {
	//loop is just to speed up the error occurance
	for (int i=0; i<10;i++) save();

	Serial.println(xPortGetFreeHeapSize());
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
	Serial.systemDebugOutput(false); //just to get not too much reconnects msgs..

	debugOut();
	dbT.initializeMs(250, debugOut);
	dbT.start(true);

}

```

kireevco, 2015-08-23:
Closing. Thank you @hreintke !

closed 2015-08-23 by kireevco