SmingHub/Sming #193: IoT protocol AllJoyn.

STASIUK, 2015-07-30:
Newly Microsoft presented windows 10 and with it the windows 10 sdk.
It includes Open Source protocol for the Internet of things which is called AllJoyn.
He is becoming popular and enables all device (pc, tablet, tv, arduino, raspberry, esp8266 and other) communicate with each other one protocol.
I ask you to consider a proposal to add this protocol in the esp8266 module. 
What do you think?

alonewolfx2, 2015-08-02:
i think it would be very good

kireevco, 2015-08-02:
@STASIUK Sounds good. 
But do you think this is more important than working HTTPS #99 or stable HTTP #154 ?

Sming team doesn't have enough resources and unfortunately right now we need to set priority for everything.

alonewolfx2, 2015-08-02:
1-Stable http
2-https
3-stable mqtt
4-..

kireevco, 2015-08-02:
@alonewolfx2 yeah, I agree with the priorities list.

ajaybhargav, 2015-08-26:
+1 for this.. I will see if I can port thin client to Sming

jcherrabi, 2015-08-26:
+10

STASIUK, 2015-08-26:
+10!!!! For .net developers and other!

closed 2016-01-01 by STASIUK