SmingHub/Sming #194: I2C read not working from arduino slave

brunnels, 2015-07-31:
I2C read from slave isn't working from arduino 328p or arduino mega as the slave.  Trying to do a single byte read or multibyte read all fail.  Simplest demonstration of the problem is here:
https://www.arduino.cc/en/Tutorial/MasterReader

Here's example Sming application.cpp.  It won't ever break out of the while loop in the MasterReader example so I just did 6 reads instead.  I have also tried with and without beginning and ending the transmission.

``` c
#include <user_config.h>
#include <SmingCore/SmingCore.h>
Timer loopTimer;

void loop() {
  Wire.beginTransmission(0x29);
  Wire.requestFrom(0x29, 6);
  for (uint8_t i = 0; i < 5; i++) {
    Serial.print((char)Wire.read());
  }
  Serial.println((char)Wire.read());
  Wire.endTransmission();
}

void init() {
  Serial.begin(115200);
  Wire.pins(5, 4);
  Wire.begin();
  loopTimer.initializeMs(5000, loop).start();
}
```
First read using that example I get: 
h2???? and then only ???????

I have successfully read from arduino slave on ESP8266 hardware with nodemcu firmware so this can work.

alonewolfx2 and others have also attempted and experienced the same results.

kireevco, 2015-08-23:
@alonewolfx2 @anakod @hreintke any thoughts?

open-home-automation, 2015-09-13:
I also had problems connecting an MCP23017 via i2c with Wire. Detecting was possible, but reading from it was crap. I ended up including Zarya's I2C driver (https://github.com/zarya/esp8266_i2c_driver) in my application and use that now instead of wire. But I hope that one day i2c starts working in Sming too so that I can revert to using wire. Spi has problems too, but those I was able to fix.
@alonewolfx2 @anakod @hreintke is anybody looking into this or would it help if I tried to look into the i2c driver?

avr39-ripe, 2015-09-14:
@open-home-automation , look at my PR #286 try it and giv us some feedback.
What problems do you have with Sming's SPI? I use 74hc595 + 74hc165 via SPI just fine, no changes to SPI itself made.

open-home-automation, 2015-09-15:
@avr39-ripe , I have added my changes in issue #245. Without these I could not get the SD card to work.

hmms, 2015-11-04:
i was trying to read the HTU21DF using the Adafruit library (https://github.com/adafruit/Adafruit_HTU21DF_Library) the 'begin' function works and the chip is detected, but the data requested using the Wire.requestFrom() doesnt seem to work and returns zero bytes, also Wire.Available() does'nt seem to work. are there any work arounds?  I would be glad to contribute to help solve this issue, pointer appreciated! :)

open-home-automation, 2015-11-06:
I just retried the MCP23017 with i2c (Wire) from Sming and it works now, so my previous comment about i2c is not valid anymore.

alonewolfx2, 2015-11-18:
Is it solved @brunnels?

hreintke, 2016-01-27:
Closed due to low activity

closed 2016-01-27 by hreintke

rafkacz, 2017-04-23:
Hi,

I have the same problem that **brunnels**. Can write to arduino slave but reading from is not working always. I have to read 5 bytes from slave - and first [10-20] iterations works well - data retrieved are as expected. But later - the input data are shifted (e.g. 0x00AABBCCDD -> 0XAABBCCDD00 - it's random). 

I've tested the conversation between 2 arduinos - all works fine... 

Any idea why?

Hukuma23, 2017-04-24:
I got the same problem, didn't fix this. My workaround is using serial protocol instead of i2c.

avr39-ripe, 2017-04-24:
@rafkacz @Hukuma23 try recently imported in Sming/develop new i2c master + Wire library ported from esp8266/Arduino and tell us if it solves the problem.

rafkacz, 2017-04-24:
@avr39-ripe - I've just tried develop brunch. Doesn't work at all. I think there is a problem with initial state of SCL/SDA lines in this version. Previously i had  about 3V at these lines after Wire.begin() call - and now it  looks like it is set to logic 0 (i measure about 2,4V at GPIO2 and 1,8V ato GPIO0 - due to external pull up resistors).

Hukuma23, 2017-04-25:
@avr39-ripe thanks. Unfortunately I haven't demo to check this now, I'll do this later

rafkacz, 2017-04-25:
Hello again - i had a hw problem - pull up resistors to big and powered from 3.3V - it was about 2,4-2,5V at SCL/SDA lines and it was not enough for 5V powered slaves . After corrections (pullup resistors powered from 5V and 3,3V zener diode to protect ESP8266) new version of wire lib seems to work perfect.
Thanks a lot.