SmingHub/Sming #196: Page refresh, errors in load files

sobiso, 2015-08-03:
This errors is appear in my project but it is too in HttpServer_Bootstrap. When i go to page in this example, page load and it is great. I reload(refresh) page few time one after the other, heap less than 20000, files have error in load

GET http://192.168.1.25/lz-string.min.js net::ERR_CONNECTION_RESET

When it will take time, heap grows to 21000, refresh page and everything it's allright.

sobiso, 2015-08-03:
My code http://esambo.pl/tmp/prox.tar

sobiso, 2015-08-03:
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
fs.start: size:208 Kb, offset:

Station configuration was updated to: gospo
AP IP succesfully updated
Set AP configuration in background
realloc 1 -> 131
system freq
160
mode : sta(18:fe:34:a0:a4:b7) + softAP(1a:fe:34:a0:a4:b7)
                                                         add if0
                                                                add if1
                                                                       dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
                         bcn 100
                                f 0, addUser: me 123
'/' registered
'/ipconfig' registered
'/ajax/get-networks' registered
'/ajax/connect' registered
'/ajax/send-roller' registered
'/ajax/set-roller' registered
'/ajax/set-router' registered
'/ajax/save-rollers' registered
'/ajax/checkCookies' registered
scandone
        f 0, bcn 0
                  del if1
                         add if1
                                bcn 100
                                       AP configuration was updated on system ready event
scandone
        scan completed: 3 found
reconnect
         f -240, Heap: 
23656
scandone
        reconnect
                 f 0, scandone
                              add 0
                                   aid 3

                                        connected with gospo, channel 6
                                                                       ip:192.168.1.25,mask:255.255.255.0,gw:192.168.1.1
      cnt 
          CONNECT TO:

Heap: 
23776
onAccept state: 0 K=0
Free heap size=23312, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient  192.168.1.1, activeClients = 1
 
path=/
Host === 192.168.1.25
parsed
Request: GET, nodata
TCP received: 38 bytes
onReadyToSendData: 1
send file: index.html (89 bytes)
response sendHeader
response sendBody
READ Template (0)
plain template text pos: 0, len: 89
Stream completed
[... 1559 lines cut ...]
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
response sendBody
Stream completed
TCP connection closing
~TCP connection
-TCP connection
-TCP connection
-TCP connection

sobiso, 2015-08-03:
And i see this errors in console browser

2015-08-03 19:43:56.460edit.html:116 GET http://192.168.1.25/common.js net::ERR_CONNECTION_RESET
2015-08-03 19:43:56.474edit.html:117 GET http://192.168.1.25/lz-string.min.js net::ERR_CONNECTION_RESET
2015-08-03 19:43:57.217edit.html:189 tes
2015-08-03 19:43:57.228edit.html:227 Uncaught ReferenceError: readCookie is not defined

anakod, 2015-08-04:
From log:
```
TCP received: 431 bytes
onReadyToSendData: 1
Default server handler for: '/jquery.js'
found jquery.js.gz
send file: jquery.js.gz (30153 bytes)
response sendHeader
response sendBody

CONNECTION DROPPED
(5584)

TCP sent: 1460
onReadyToSendData: 2
response sendBody
TCP sent: 1460
onReadyToSendData: 2
```
Sming droping connection when available memory **lower than 6000 bytes** to prevent heap overflow.

This issue is due to natural memory constrains of the device. Partially it is related to TCP TIME_WAIT (http://superuser.com/questions/173535/what-are-close-wait-and-time-wait-states), partially to too many concurrent requests. 
Embedded devices, generally, require more optimization and can not be used the same way as a general-purpose server.

We can tackle this issue from multiple ways:
User Application optimiztion:
- Resourse minification
- Optimizing network request count
- Optimize memory usage in the app

Sming Core Low Level Optimization:
- Optimize network buffers
- Implement IRAM cache (store network buffers cache in IRAM)

Let's sync up on gitter with us, we'll see what we can do for you.

kireevco, 2015-08-23:
@sobiso Any update from your side?

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke