SmingHub/Sming #197: If I have 4MB module (esp-12). Need more spiff memory. >256

STASIUK, 2015-08-03:
I have 4MB module (esp-12)  and my files do not fit in 256 KB.
Can I increase the spiff memory size for my module(esp-12). And as?
Maybe somewhere in the settings need to change?

anakod, 2015-08-03:
You can just set SPI_SIZE = 4M in Makefile configuration.
More details here: https://github.com/anakod/Sming/issues/78

STASIUK, 2015-08-03:
Total files take 620kb. I have esp-12 module (4MB).
What can I do to add all the files?

sobiso, 2015-08-03:
What SDK version ?

STASIUK, 2015-08-03:
latest.

STASIUK, 2015-08-04:
please, help.

anakod, 2015-08-05:
How you upload files? Spiffy, FTP, or something else?

STASIUK, 2015-08-05:
I upload through singing Spiffy and FTP. 
total size of files that I want to load the module 620kb.
I have esp-12 module (4MB). 

> Default fs size is 196608 and directory is files.
Creating rom spiff_rom.bin of size 196608 bytes
Adding files in directory files
Unable to read file .
Unable to read file ..
admin.html added to spiffs (4562 bytes)
bootstrap.min.css added to spiffs (122540 bytes)
bootstrap.min.js added to spiffs (36816 bytes)
dashboard.css added to spiffs (1557 bytes)
Unable to read file dashboard.js
Unable to write file index.html - errno -10001
index.html added to spiffs (567 bytes)
Unable to write file jquery.js - errno -10010
jquery.js added to spiffs (84345 bytes)
Unable to write file main.js - errno -10010
main.js added to spiffs (769 bytes)
spiff_rom.bin---------->0x44000

When I upload files via FTP they do not fit.
What do I need to do, to increase the size of the file system to hold all my files?

flexiti, 2015-08-07:
How have you been checking out what type of Flash You have?

See my example #201

flexiti, 2015-08-07:
I have ESP12 and, SPI_SIZE = 4M

"fs.start: size:3824 Kb, offset:0x40000" , so its 3.8 Mb spiff file system

You have only 196 Kb 
"Default fs size is 196608 and directory is files"

sobiso, 2015-08-07:
Change in /Sming/Makefile-project.mk line: 
echo "$(SPIFF_FILES) directory exists. Creating spiff_rom.bin"; \
    	spiffy 
    	mv spiff_rom.bin $(FW_BASE)/spiff_rom.bin; \

to:
echo "$(SPIFF_FILES) directory exists. Creating spiff_rom.bin"; \
    	spiffy 500000 files; \
    	mv spiff_rom.bin $(FW_BASE)/spiff_rom.bin; \

flexiti, 2015-08-07:
first: find file Makefile-project.mk in SmingFramework project
next: change SPI_SIZE to 4M
then: rebulit SmingFramework, next rebulit and download Your project.
Ps. I don't know if You need also change SPI_MODE 

# SPI_MODE: qio, qout, dio, dout  
SPI_MODE ?= qio
# SPI_SIZE: 512K, 256K, 1M, 2M, 4M
SPI_SIZE ?= 4M

anakod, 2015-08-07:
SPI_MODE=dio for large flash, usually.

kireevco, 2015-08-07:
@anakod @flexiti I should probably make a simple user option in Makefile-user.mk for this. Something like
```
# FLASH_SIZE := 512K
# FLASH_SIZE := 1M
# FLASH_SIZE := 2M
FLASH_SIZE := 4M
```

On the other hand,is there any way to detect the device type somehow automatically?

flexiti, 2015-08-07:
The only way  is  like in my example at #201

piperpilot, 2015-08-07:
As far as I can tell SPIFF uses memory mapped SPI access, not direct SPI Read, so the combined user code and SPIFF must fit in the 1MB boundary.  The SPIFF routines would need to be re-written to use direct SPI access to support the above example of 3.8MB SPIFFs

STASIUK, 2015-08-07:
I have does not work!
I ordered in Makefile-project.mk :
>SPI_SIZE = 4M
SPI_MODE = dio

Build project (rebuild, flashinit, flashboot and flash)
and in the terminal, I get:
>fs.start: size:224 Kb, offset:40244000
What am I doing wrong ?

flexiti, 2015-08-07:
You must first rebulid sming, did You ?

STASIUK, 2015-08-07:
How do I do this?

flexiti, 2015-08-07:
[image]

STASIUK, 2015-08-07:
[image]

anakod, 2015-08-08:
You should rebuild your project too, and add SPI configuration details to makefile before that.

cartoonincm, 2015-08-20:
can you create "How to slove this problem "(step-by-step) because i got this problem also.

STASIUK, 2015-08-22:
1. Delete folder C:\Espressif
2. Install https://github.com/CHERTS/esp8266-devkit/releases/tag/v2.0.6

kireevco, 2015-08-23:
@STASIUK were you able to resolve it? What was the list of actions that you took?

leolite1, 2015-09-25:
I also ran into the same issue. The latest spiffy from https://github.com/xlfe/spiffy is hardcoded to 192KB max size. I ended up using a forked version https://github.com/alonewolfx2/spiffy.

closed 2015-09-25 by STASIUK