SmingHub/Sming #198: TCP Client Stops with "TCP connection error: -8"

hrsavla, 2015-08-03:
"TCP connection error: -8" mean ?? error number is either -8 or -9 . I have been receiving this error in most TCP client application like Mqtt, downloadString() (httpclient), ftp (Big files). The occurrence of this is random. But occurs onces in atleast 5 attempts. My usual fix for it has been restarting the TCP client which works, but now analyzing and have auto restart in Tcpclient will be helpful to all.

On further looking on error codes in espconn.h (which is in  this folder C:\tools\sming\Sming\system\include\lwip\app)

#define ESPCONN_ABRT       -8    /* Connection aborted.      */
#define ESPCONN_RST        -9    /* Connection reset.        */
#define ESPCONN_CLSD       -10   /* Connection closed.       */
#define ESPCONN_CONN       -11   /* Not connected.           */

So it seems that esp8266 is aborting /resetting TCP connection. 
** "Is this aborting/resetting of TCP connection due to timeout ? "  
** Also suggest about way to implement "Auto-reconnect".

flexiti, 2015-08-04:
Please read issue:
 https://github.com/anakod/Sming/issues/174
and check https://github.com/flexiti/HttpClient_Bug
Is not this another symptom of the same problems?

hrsavla, 2015-08-04:
Yes @flexiti , all your symptom seems to be pointing to same "Exception" (lets not say call it  bug). We need to handle this "Connection aborted/reset" exception. In my Mqtt client I have a timer periodically checking connection status and reconnecting it. But there should be better way of "Auto Reconnect" in TCPClient.

hreintke, 2015-08-04:
@hrsavla : Currently I thing the concept of the implementation is OK.
It calls the callback oncompled with status true/false based on correct/not correct completion of http request. It's up to the application to reconnect or take other actions.
Of course the status in the callback should be correct (which it not now) to let this work

flexiti, 2015-08-04:
I found a solution to  "not existing server" problem. (https://github.com/flexiti/HttpClient_Bug )
 it's probably solves few other problems. 
On network  error, the "asyncTotalLen" remained  not cleared
What is your opinion?

SimngCore-Network-TcpClient.cpp
----------------------------------------------
void TcpClient::onError(err_t err)
{
	      asyncTotalLen = 0;  //this should be added in my opinion
	      asyncTotalSent = 0; // and this ?? 

	state = eTCS_Failed;
	onFinished(state);
	TcpConnection::onError(err);
}

flexiti, 2015-08-07:
All works more stable now , but still "TCP connection error: -8" appears, we must work on it - why ?.
(no need now restart TCP client,  during the next connection error disappears)

hrsavla, 2015-08-07:
It seems to be some timing problem. This error occurs when TCP client is trying to connect to server. We need to see functioning of connect() function.  It seems that after sending connect request to server, TCP client expects some packet. But if that packet is not received in stipulated time then it give this "error :-8" (ie abort connection). 
This scenario does not happen every time. If the relay time between server and client is long then only you would get this error. You will never get this error if server is in your local network. 
We need to check TcpConnection.cpp for this.

flexiti, 2015-08-07:
strange, so far I have not found the place where this error is set (TCPClient and TcpConnection)

hrsavla, 2015-08-07:
In TcpConnection.cpp  line number 277 (TcpConnection::internalTcpConnect(IPAddress addr, uint16_t port))  This function tries to connect to server. And we get "TcpConnection::connect result:, 0" in our debug screen.  internalTcpConnect() call callback function staticOnConnected()  ["Just bellow it in code"]. If connection is OK then we get "OnConnected" in our debug screen.

But in Error Case it seems staticOnConnected() is not called.

flexiti, 2015-08-07:
this part is ok, even in  error sequence 

-----------------------
Download: http://184.106.153.149:80/update?key=YOXULSNC3XXXXXX&field1=31.3&field7=27.3&field8=1002.1&status=dev_ip:192.168.0.180
TcpClient_connect1
TcpClient_connect2
+TCP connection
connect to: 184.106.153.149
TcpConnection::connect result: 0
TcpClient_sendString
TcpClient_send
len = 67
 asyncTotalSent =0, asyncTotalLen=0
 asyncTotalSent =0, asyncTotalLen=67
TcpClient_sendString
TcpClient_send
 len = 2
realloc 67 -> 198
 asyncTotalSent =0, asyncTotalLen=67
 asyncTotalSent =0, asyncTotalLen=69
TcpClient_sendString
TcpClient_send
len = 0
asyncTotalSent =0, asyncTotalLen=69
 asyncTotalSent =0, asyncTotalLen=69
OnConnected
TcpClient::OnConnected asyncTotalSent =0, asyncTotalLen=69
TCP connected
TcpClient_onReadyToSendData
onReadyToSendData: 0
TcpClient_onReadyToSendData_pushAsyncPart
TcpClient_pushAsyncPart
TcpClient_pushAsyncPart not NULL
TCPConnection_write_Stream
write..read = 1024 avail = 69
TCPConnection_write_data
TCP connection send: 69 (69)
TcpCoonection_flush
TCPConnection_write_Stream total=69
TcpClient_pushAsyncPart_write
TcpCoonection_flush
TcpClient request completed d=1
TcpConnection staticOnConnected err=0
TcpClient_onReadyToSendData
onReadyToSendData: 3
TcpClient_onReadyToSendData_pushAsyncPart
TcpClient_pushAsyncPart
TcpClient_pushAsyncPart stream = NULL
TCPConnection_staticOnError err=-8
TCPConnection_staticOnError con not NULL
TcpClient_onError err=-8

hrsavla, 2015-08-07:
In your case, Tcpconnection has been made NULL . 
What happens at "TcpClient request completed d=1".  Is it part of your code. It seems Tcp connection is getting closed and then you are trying to send data. So you are getting err=-8 .

flexiti, 2015-08-07:
I still do not know where err -8 is set
ps. Here is the function with the "d" :),
Btw. when connection is ok, all log is the same as previous log until  line "TCPConnection_staticOnError err=-8"

void TcpClient::pushAsyncPart()
{
	int d;
	debugf("TcpClient_pushAsyncPart");
	if (stream == NULL)
	{
		debugf("TcpClient_pushAsyncPart stream = NULL");
		return;
	}
	debugf("TcpClient_pushAsyncPart not NULL");
	write(stream);
	debugf("TcpClient_pushAsyncPart_write");
	if ((d=stream->isFinished()))
	{
		flush();
		debugf("TcpClient request completed d=%d",d);
		delete stream; // Free memory now!
		stream = NULL;
	}
}

hreintke, 2015-08-07:
The errorcode comes from the underlying lwip implementation. Is is coming from there to sming via static onerror and onerror.

flexiti, 2015-08-07:
I have also  such a sentence, it will be difficult to get a reason

hreintke, 2015-08-07:
Think you have to get into tcp tracing using wireshark to get to the details. 
But even then you would only see the reason and no possibility to fix. Unless you would go to use a local updated lwip

hrsavla, 2015-08-07:
@hreintke Lwip is implented by Espressif sdk ?

flexiti, 2015-08-07:
I think so, because we have only liblwip.a

flexiti, 2015-08-07:
but in example we have lwip source, you need to check whether the compilation replace the library, something seems to me that once I tried anyway

flexiti, 2015-08-07:
Yes, it possible to change liblwip.a to own one, with debug.
However, I will log on Monday ;)

flexiti, 2015-08-14:
http://bbs.espressif.com/viewtopic.php?p=3221 :)

dmarkey, 2015-08-20:
Anyone make any progress here?

anakod, 2015-08-20:
> http://bbs.espressif.com/viewtopic.php?p=3221
It's gread news!

flexiti, 2015-08-21:
Problem still exist, I am busy these days,  maybe @hrsavla  did some progress

hrsavla, 2015-08-21:
@dmarkey @flexiti I had put up this issue, so that every one can document the scenario due to which this error -8/-9 is occuring. And some better solution can be found. 
One of the scenario I found due to which this error is occurring and its possible solution / workaround  is listed by me in https://github.com/anakod/Sming/issues/216 
With  access to lwip (http://bbs.espressif.com/viewtopic.php?p=3221). I hope @anakod / @hreintke can suggest a way to implement event closed / closing. 
The latest issue added by @dmarkey  regarding Mqtt client also seems to be generated due to same  error.

hreintke, 2015-08-21:
I am out until next Wednesday but will check then the possibility

dmarkey, 2015-08-21:
Is this a problem with all the SDKs and wrappers or only Sming?

flexiti, 2015-08-21:
I was tested it with SDK 0.95 and 1.3  - the same

dmarkey, 2015-08-21:
Hmm, previous Sming versions? Arduino SDK has the same problem I wonder?

flexiti, 2015-08-21:
I think that is something at the junction sming - sdk

dmarkey, 2015-08-21:
I have 2 MQTT servers in 2 different locations. This error only happens with one and not the other. :/

kireevco, 2015-08-23:
@hrsavla were you able to resolve your initial issue?

flexiti, 2015-08-26:
I have a test environment that the problem is repeatable. ("TCP connection error: -8" every 2-5 connection).
This is not a problem of any particular program, but I think that particular environment (many Wifi devices ?)
Although the error occurs, however, the information reaches the server.
The error is signaled from the library liblwip. Does anyone know what further can be done to find out the actual reason? I can easily get this error for test  (Unfortunately, I failed to turn on log liblwip )

dmarkey, 2015-08-29:
Anyone manage to compile in the lwip they released?

flexiti, 2015-09-07:
in my case helped change the method of use DS1820
Look   #249

dmarkey, 2015-09-07:
@flexiti Hmm, what exactly would we need to do to the TCP stack to fix this?

flexiti, 2015-09-07:
check if You use function delay()  in your aplication   eg. if your application uses Ds1820 from example You must change this .
How kind of applications do you have?

Using a long delay() in the application negatively affects the TCP stack.

dmarkey, 2015-09-08:
Mine is here: https://github.com/dmarkey/smarthome_esp8266controller_sming

I dont use delay anywhere as far as I can tell :(

flexiti, 2015-09-08:
Surely you have some other reason of instability.
I will try to test it in my free time (this is # 222 anyway?)

dmarkey, 2015-09-08:
Yes, possible duplicate.

dmarkey, 2015-10-06:
Any change with 1.4 SDK?

dmarkey, 2015-10-09:
Seems to be the same on 1.4 :(

flexiti, 2015-10-09:
I know this effect, I try to track down.
A few changes that I introduced improved the case, but the problem is still, however, now already a much smaller scale.
I'll be looking for.

dmarkey, 2015-10-09:
Anyone any luck compiling/linking the OSS version of lwip released by espressif?

dmarkey, 2015-10-11:
I'm seeing similar unreliability with Arduino esp8266 :(

hreintke, 2015-12-21:
Probably unrelated to Sming but to Espressif network implementation

closed 2015-12-21 by hreintke