SmingHub/Sming #203: Bug in RTC class

danielnilsson9, 2015-08-07:
Hi

Have done som experiments again with the NtpClient, fixed a few bugs and so on, will make a PR later.

Anyway, seems like something is not correct in the SystemClock/RTC class, most likely the RTC.

Checkout this log:
>Sming. Let's do smart things!
>Station configuration was updated to: wlan
>mode : sta(18:fe:34:a5:73:73) + softAP(1a:fe:34:a5:73:73)
>add if0
>add if1
>dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
>bcn 100
>f 0, scandone
>add 0
>aid 1
>cnt 

>connected with wlan, channel 7
>dhcp client start...
>ip:192.168.1.154,mask:255.255.255.0,gw:192.168.1.1
>NTP client request time.
>DNS IP lookup in progress.
>UDP LocalPort: 4097
>UDP connect to 216.239.34.15:123
>time updated? 1
>Time received: 07.08.2015 22:03:36
>Time synchronized RTC: 07.08.2015 22:03:41
>Timer processing: interval counter: 1
>Timer processing: interval counter: 2
>NTP client request time.
>DNS IP lookup in progress.
>UDP connect to 216.239.34.15:123
>time updated? 1
>Time received: 07.08.2015 22:13:35
>Time synchronized RTC: 07.08.2015 22:23:34
>Timer processing: interval counter: 1
>Timer processing: interval counter: 2
>NTP client request time.
>DNS IP lookup in progress.
>UDP connect to 216.239.34.15:123
>time updated? 1
>Time received: 07.08.2015 22:23:35
>Time synchronized RTC: 07.08.2015 22:33:33
>Timer processing: interval counter: 1
>Timer processing: interval counter: 2
>NTP client request time.
>DNS IP lookup in progress.
>UDP connect to 216.239.34.15:123
>time updated? 1
>Time received: 07.08.2015 22:33:35
>Time synchronized RTC: 07.08.2015 22:43:36

Have done a few minor changes to ntp class, some more debug outputs as you can se above, have also fixed timer class to support larger intervals, limit was around 7 minutes before as stated in issue #67.

NtpClient is running with 10 minute update interval. The callback function looks like this:

>void onNtpReceive(NtpClient& client, time_t timestamp) {
>	SystemClock.setTime(timestamp, eTZ_UTC);
>
>	Serial.print("Time received: ");
>	Serial.println(DateTime(timestamp + 2*3600).toFullDateTimeString());
>
>
>	Serial.print("Time synchronized RTC: ");
>	Serial.println(SystemClock.getSystemTimeString(eTZ_Local));
>}

Nothing complicated, whats funny is that we call SystemClock.setTime() and later SystemClock.getTime() but then it doesn't match what was received from the ntp client.

First time its off by 5 seconds (roughly how long the boot up takes), later on its off by 10 minutes (interval between requests) and it then continues to be that.

As I don't fully understand how the rtc apis work on this chip I can't se what the problem is but it must be in RTC.cpp.

Any ideas?

danielnilsson9, 2015-08-07:
Think I found the problem now, seems to be the wrong order in this function. First loadTime, then calculate offset and then adding the elapsed time since last save to rtc memory. I think it should be loadTime(), updateTime() and the calc offset, it seems to work at least.

>bool RtcClass::setRtcSeconds(uint32_t seconds) {
>	RtcData rtcTime;
>	loadTime(rtcTime);
>	uint64_t offsetNs = ((uint64_t) seconds * NS_PER_SECOND) - rtcTime.time;
>	rtcTime.time += offsetNs; // adjust time in nanoseconds by the ntp time offset in nanoseconds
>	updateTime(rtcTime);
>	return saveTime(rtcTime);
>}

Tavalin, 2015-08-09:
Hi @danielnilsson9 

Sorry I hadn't seen this sooner. I was having a look at this method the other day and think it could be rewritten as follows:

```
bool RtcClass::setRtcSeconds(uint32_t seconds) {
RtcData rtcTime;
loadTime(rtcTime);
uint64_t time= ((uint64_t) seconds * NS_PER_SECOND);
rtcTime.time = time;
updateTime(rtcTime);
return saveTime(rtcTime);
}
```
If you results are reproducible, i.e. always seems to go out of sync, could you try the above and see if it fixes it please?

Thanks

danielnilsson9, 2015-08-09:
Nope that can't work either, problem is that you load the time from rtc memory which hasn't been updated, then you set time to the synced time, and then you calculate the numer of us that has elapsed since last time it was saved in rtc memory i suppose and add that pratically adding the elpsed time since last save to the time passed as argument. 

This works:
>bool RtcClass::setRtcSeconds(uint32_t seconds) {
>	RtcData rtcTime;
>	loadTime(rtcTime);
>	updateTime(rtcTime);
>	uint64_t offsetNs = ((uint64_t) seconds * NS_PER_SECOND) - rtcTime.time;
>	rtcTime.time += offsetNs; // adjust time in nanoseconds by the ntp time offset in nanoseconds
>	return saveTime(rtcTime);
>}

But I believe some of the codde could be skipped, haven't tested though.

Tavalin, 2015-08-09:
OK. Could you try the following then please:

```
bool RtcClass::setRtcSeconds(uint32_t seconds) {
RtcData rtcTime;
loadTime(rtcTime);
updateTime(rtcTime);
uint64_t time= ((uint64_t) seconds * NS_PER_SECOND);
rtcTime.time = time;
return saveTime(rtcTime);
}
```

danielnilsson9, 2015-08-09:
Yes that works too, I guess thats better since there isn't really any reason to calc offset and then add that.

I have a PR I'm gonna make this evening fixing bugs in NtpClient, Timer and then RTC, I will include this fix if it is OK?

Tavalin, 2015-08-09:
That's great, thank you!

Tavalin, 2015-08-11:
Looks like we can close now that #206 is merged?

danielnilsson9, 2015-08-12:
Yes, its fixed.

closed 2015-08-12 by danielnilsson9