SmingHub/Sming #205: make: *** No rule to make target `/c/tools/sming/Sming/Makefile-windows.mk'.  Stop.

mshill, 2015-08-09:
Having a basic problem getting a fresh install of Sming to build. I followed the Windows setup guide, but when I open the eclipse examples, open the make targets, and run the "all", I get the following error:
10:13:15 **** Build of configuration Sming for project Basic_Serial ****
make all 
C:\tools\sming\Sming/Makefile-project.mk:49: /c/tools/sming/Sming/Makefile-windows.mk: No such file or directory
make: *** No rule to make target `/c/tools/sming/Sming/Makefile-windows.mk'.  Stop.

10:13:15 Build Finished (took 94ms)

On Windows 10 if that makes a difference. Apologize in advance if this has been answered somewhere else, I couldn't find info on it in my searches.
Thanks!
Mike

mshill, 2015-08-09:
Seems to be a environment issue. Not optimal, but if I hard code c:/tools/sming/Sming/ in line 49 of c:/tools/sming/Sming/Makefile-project.mk, I can make it further. Next message I get is:

10:45:33 **** Build of configuration Sming for project Basic_Serial ****
make all 
mkdir: cannot create directory `out/build//c/tools/sming/Sming/appinit': No such file or directory
make: *** [out/build//c/tools/sming/Sming/appinit] Error 1

10:45:33 Build Finished (took 203ms)

kireevco, 2015-08-09:
@mshill 
1. Start `cmd.exe`
2. Please paste output of 
```
echo %SMING_HOME%
echo %ESP_HOME%
dir %SMING_HOME%
dir %ESP_HOME%
```

mshill, 2015-08-09:
C:\Users\daUser>echo %SMING_HOME%
C:\tools\sming\Sming

C:\Users\daUser>echo %ESP_HOME%
c:/Espressif

C:\Users\daUser>dir %SMING_HOME%
 Volume in drive C has no label.
 Volume Serial Number is 1CB7-C716

 Directory of C:\tools\sming\Sming

08/09/2015  08:59 AM    <DIR>          .
08/09/2015  08:59 AM    <DIR>          ..
06/24/2015  08:53 PM             8,597 .cproject
06/24/2015  08:53 PM               815 .project
08/09/2015  08:59 AM    <DIR>          .settings
08/09/2015  08:59 AM    <DIR>          appinit
08/09/2015  08:59 AM    <DIR>          apptest
08/09/2015  08:59 AM    <DIR>          compiler
08/09/2015  08:59 AM    <DIR>          include
08/09/2015  08:59 AM    <DIR>          Libraries
06/24/2015  08:53 PM             7,687 License.txt
06/24/2015  08:53 PM             8,323 Makefile
06/24/2015  08:53 PM               554 Makefile-bsd.mk
06/24/2015  08:53 PM               541 Makefile-linux.mk
06/24/2015  08:53 PM               533 Makefile-macos.mk
08/09/2015  10:45 AM            13,847 Makefile-project.mk
06/24/2015  08:53 PM               540 Makefile-windows.mk
08/09/2015  08:59 AM    <DIR>          Services
08/09/2015  08:59 AM    <DIR>          SmingCore
08/09/2015  08:59 AM    <DIR>          system
08/09/2015  08:59 AM    <DIR>          Wiring
               9 File(s)         41,437 bytes
              12 Dir(s)  61,968,445,440 bytes free

C:\Users\daUser>dir %ESP_HOME%
Invalid switch - "Espressif".

kireevco, 2015-08-09:
Please Change `c:/Espressif` to `c:\Espressif`

mshill, 2015-08-09:
After changing the "/" to "\" in c:\tools\sming\Sming\Makefile-project.mk, per above, I now get an interesting new output:

11:02:45 **** Incremental Build of configuration Sming for project Basic_Serial ****
make all 
mkdir: cannot create directory `out/build//c/tools/Sming/Sming/appinit': No such file or directory
make: *** [out/build//c/tools/Sming/Sming/appinit] Error 1

11:02:45 Build Finished (took 147ms)

"out/build//c/tools/Sming/Sming/appinit" obviously can't be created, so I'm looking for that in the makefiles - any hints would be welcome... ;)

kireevco, 2015-08-09:
@mshill 
1. Are you testing from cmd.exe? If not, please open `cmd.exe` and run make in your project
2. Please make sure to your SMING_HOME, ESP_HOME environment variables are up to date: set them and restart your computer.

mshill, 2015-08-09:
Ok, rebooted and ran make from the command line.Still get the same error:

C:\tools\sming.examples\Basic_Serial>echo %SMING_HOME%
C:\tools\sming\Sming

C:\tools\sming.examples\Basic_Serial>echo %ESP_HOME%
c:\Espressif

C:\tools\sming.examples\Basic_Serial>make all
mkdir: cannot create directory `out/build//c/tools/sming/Sming/appinit': No such
 file or directory
make: *** [out/build//c/tools/sming/Sming/appinit] Error 1

mshill, 2015-08-09:
Trying to make sense of the output. Changed line 271 in c:\tools\sming\Sming\Makefile-project.mk from:
BUILD_DIR	:= $(addprefix $(BUILD_BASE)/, $(MODULES))
to:
BUILD_DIR	:= $(addprefix $(MODULES)/, $(BUILD_BASE))

and got something that looked a little more executable:

15:17:50 **** Incremental Build of configuration Sming for project Basic_Serial ****
make all 
mkdir: cannot create directory `/c/tools/sming/Sming/appinit/out/build': No such file or directory
make: *** [/c/tools/sming/Sming/appinit/out/build] Error 1

15:17:50 Build Finished (took 139ms)

I'm not sure where the output directory is supposed to be, so I might just be throwing a wrench in the system... Seems like it should be in the project folder under build?

kireevco, 2015-08-09:
@mshill current make file configuration is well tested. 

Can you please provide us with output of the following command:
`where make` ?

mshill, 2015-08-09:
Ah ha! This looks interesting...

C:\tools\sming.examples\Basic_Serial>where make
C:\WinAVR-20100110\utils\bin\make.exe
C:\Program Files (x86)\MHV AVR Tools\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe
c:\GnuWin32\bin\make.exe

mshill, 2015-08-10:
Ok, reset all files to default, changed the path to ensure the "tools" exec were first in the chain - voila! It's all working now.

For those who run into the same problem (at least in Windows 10):
1) start a cmd.exe as administrator
2) type "rundll32.exe sysdm.cpl,EditEnvironmentVariables" at the cmd prompt
3) move all path references to "c\tools..." to first in the chain of paths
4) restart (best if you want to use Eclipse for the building)

Hope that saves some people some time...

Thanks @kireevco!

kireevco, 2015-08-10:
@mshill Yeah, that's another common issue - having another make.exe in the path.
The solution - move your mingw bin directory in the path variable so it will take precedence before other make.exe.

Another solution - use docker.

Glad that you made it work!

realtimeup, 2015-10-09:
My problem:
run cmd.exe as Administrator
c:\tools\sming\Sming make all 
No problem
c:\tools\sming\Basic_Blink make all 
return error

C:\tools\sming\Basic_Blink>echo %ESP_HOME%
c:\Espressif

C:\tools\sming\Basic_Blink>echo %SMING_HOME%
C:\tools\sming\Sming

C:\tools\sming\Basic_Blink>make all
C:\tools\sming\Sming/Makefile-project.mk:49: /c/tools/Sming/Sming/Makefile-windo
ws.mk: No such file or directory
make: *** No rule to make target '/c/tools/Sming/Sming/Makefile-windows.mk'.  Stop.

same error at Eclipse with sming examples.

Thank you.

kireevco, 2015-10-10:
@realtimeup please provide diagnostics as specified here https://github.com/SmingHub/Sming/wiki/Troubleshooting-Windows

realtimeup, 2015-10-10:
C:\>echo %PATH%
C:\tools\mingw64;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin;C:\app\oracl
e\product\12.1.0\dbhome_2\bin;C:\ProgramData\Oracle\Java\javapath;C:\Windows\sys
tem32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\
v1.0\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Pr
ogram Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsof
t SDKs\TypeScript\1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\
Program Files\TortoiseGit\bin;C:\Program Files (x86)\Microsoft SQL Server\100\To
ols\Binn\;C:\Program Files\Microsoft SQL Server\100\Tools\Binn\;C:\Program Files
\Microsoft SQL Server\100\DTS\Binn\;C:\Program Files\TortoiseSVN\bin;C:\Program
Files (x86)\Microsoft ASP.NET\ASP.NET Web Pages\v1.0\;C:\Program Files (x86)\Mic
rosoft SQL Server\100\Tools\Binn\VSShell\Common7\IDE\;C:\Program Files (x86)\Mic
rosoft SQL Server\100\DTS\Binn\;C:\Program Files\Microsoft\Web Platform Installe
r\;C:\Program Files (x86)\Android\android-sdk\platform-tools;C:\Program Files (x
86)\Android\android-sdk\tools;C:\Program Files\nodejs\;C:\Program Files (x86)\Gi
t\bin;C:\Program Files (x86)\Git\cmd;C:\Program Files (x86)\OpenVPN\bin;C:\Progr
am Files (x86)\Skype\Phone\;C:\ProgramData\chocolatey\bin;C:\Program Files\Eclip
se Foundation\4.4.2\eclipse;C:\Users\leand_000\AppData\Roaming\npm;C:\Program Fi
les (x86)\WinAnt\bin;C:\Program Files\Java\jdk1.8.0_45\bin

C:\>echo %SMING_HOME%
C:\tools\sming\Sming

C:\>echo %ESP_HOME%
C:\Espressif

C:\>dir %SMING_HOME%
 Volume in drive C has no label.
 Volume Serial Number is 961D-2DCC

 Directory of C:\tools\sming\Sming

09/10/2015  15:59    <DIR>          .
09/10/2015  15:59    <DIR>          ..
25/06/2015  00:53             8.597 .cproject
25/06/2015  00:53               815 .project
08/10/2015  09:28    <DIR>          .settings
08/10/2015  09:28    <DIR>          appinit
08/10/2015  09:28    <DIR>          apptest
08/10/2015  09:28    <DIR>          compiler
08/10/2015  09:28    <DIR>          include
08/10/2015  09:28    <DIR>          Libraries
25/06/2015  00:53             7.687 License.txt
25/06/2015  00:53             8.323 Makefile
25/06/2015  00:53               554 Makefile-bsd.mk
25/06/2015  00:53               541 Makefile-linux.mk
25/06/2015  00:53               533 Makefile-macos.mk
10/10/2015  09:19            13.840 Makefile-project.mk
25/06/2015  00:53               540 Makefile-windows.mk
10/10/2015  09:12    <DIR>          out
08/10/2015  09:28    <DIR>          Services
08/10/2015  09:28    <DIR>          SmingCore
08/10/2015  09:28    <DIR>          system
08/10/2015  09:28    <DIR>          Wiring
               9 File(s)         41.430 bytes
              13 Dir(s)  15.238.144.000 bytes free

C:\>dir %ESP_HOME%
 Volume in drive C has no label.
 Volume Serial Number is 961D-2DCC

 Directory of C:\Espressif

08/10/2015  09:27    <DIR>          .
08/10/2015  09:27    <DIR>          ..
01/09/2015  00:17            18.828 changelog.txt
21/05/2015  21:52             2.061 Command line Unofficial Development Kit for
Espressif ESP8266.lnk
16/01/2015  09:28               186 DevKit web site.url
08/10/2015  09:27    <DIR>          docs
[... 102 lines cut ...]
C+ Libraries/BMP180/BMP180.cpp
C+ Libraries/MMA_7455/MMA_7455.cpp
C+ Libraries/WS2812/WS2812.cpp
C+ Libraries/BH1750FVI/BH1750FVI.cpp
C+ Libraries/LiquidCrystal/LiquidCrystal_SR.cpp
C+ Libraries/LiquidCrystal/LiquidCrystal_SR3W.cpp
C+ Libraries/LiquidCrystal/LiquidCrystal.cpp
C+ Libraries/LiquidCrystal/LCD.cpp
C+ Libraries/LiquidCrystal/LiquidCrystal_SR2W.cpp
C+ Libraries/LiquidCrystal/I2CIO.cpp
C+ Libraries/LiquidCrystal/LiquidCrystal_I2C.cpp
C+ Libraries/LiquidCrystal/FastIO.cpp
AR out/build/app_app.a
Installing libsming
Done

C:\tools\sming\Sming>cd..

C:\tools\sming>cd Basic_Blink

C:\tools\sming\Basic_Blink>make
C:\tools\sming\Sming/Makefile-project.mk:49: /c/tools/Sming/Sming/Makefile-windo
ws.mk: No such file or directory
make: *** No rule to make target '/c/tools/Sming/Sming/Makefile-windows.mk'.  St
op.

realtimeup, 2015-10-10:
I alter file "c:\tools\sming\Sming\Makefile-project.mk" line 49 "include $(SMING_HOME)/Makefile-windows.mk  " to "include c:\tools\Sming\Sming\Makefile-windows.mk" and make command execute ok.

asez73, 2015-10-19:
Hi @kireevco ,
I have the same problems and my path are slightly different: Windows 8.1 64 bits and Mingw64
So here are:
1- errors messages in eclipse
>Description	Resource	Path	Location	Type
make: *** No rule to make target `/c/tools/Sming/Sming/Makefile-windows.mk'.  Stop.	Basic_Blink		 	C/C++ Problem
make: /c/tools/Sming/Sming/Makefile-windows.mk: No such file or directory	Basic_Blink		line 49, external location: c:\tools\Sming\Sming\Makefile-project.mk	C/C++ Problem

2- another, probably linked error message at setup (basically error: uncorrect syntax. The default option is not allowed more than 2 times.):
>c:\tools\sming\Sming>SETX PATH /M C:\mingw\bin;C:\mingw\msys\1.0\bin;%PATH%
Erreur : Syntaxe incorrecte. L’option par défaut n’est pas autorisée plus de 2 fois.
Entrez "SETX /?" pour afficher la syntaxe.

or its equivallent with 64 bits invocations
>c:\tools\sming\Sming>SETX PATH /M C:\mingw64\bin;C:\mingw64\msys\1.0\bin;%PATH%
Erreur : Syntaxe incorrecte. L’option par défaut n’est pas autorisée plus de 2 fois.
Entrez "SETX /?" pour afficher la syntaxe.

3- the various debugging info:

>c:\tools\sming\Sming>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Perl64\site\bin;C:\Perl64\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Calibre
2\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\10.0\DLLShared\;C:\Program Files\TortoiseHg\;C:\Program Files (x86)\Windows Live\
Shared;C:\Program Files (x86)\Java\jre7\bin;C:\MinGW\bin;C:\cygwin\bin;C:\Program Files (x86)\WinSCP\;C:\Program Files (x86)\Skype\Phone\;C:\Users\Alain\.dnx\bin;C:\Program Files\Microsoft DNX\
Dnvm\;C:\Program Files\nodejs\;C:\SysGCC\esp8266\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x86)\nodejs\;C:\Program Files\Git\cmd;C:\Program File
s\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Eclipse Foundation\4.4.2\eclipse;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin;C:\Tcl\bin;C:\U
sers\Alain\Anaconda;C:\Users\Alain\Anaconda\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Perl64\site\bin;C:\Perl64\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\Syste
m32\WindowsPowerShell\v1.0\;C:\Program Files\Calibre2\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\10.0\DLLShared\;C:\Program Fi
les\TortoiseHg\;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\Java\jre7\bin;C:\MinGW\bin;C:\cygwin\bin;C:\Program Files (x86)\WinSCP\;C:\Program Files (x86)\Skype\Phone\;C:\
Users\Alain\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\nodejs\;C:\SysGCC\esp8266\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x
86)\nodejs\;C:\Program Files\Git\cmd;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Eclipse Foundation\4.4.2\eclipse;;C:\Tcl\bin;C:
\Users\Alain\Anaconda;C:\Users\Alain\Anaconda\Scripts;C:\Users\Alain\AppData\Roaming\npm;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\Alain\AppData\Roaming\npm;C:\Program Files (x86)\M
icrosoft VS Code\bin;C:\tools\mingw64\bin;

>c:\tools\sming\Sming>echo %SMING_HOME%
c:\tools\Sming\Sming

>c:\tools\sming\Sming>echo %ESP_HOME%
c:\Espressif

>c:\tools\sming\Sming>dir %SMING_HOME%
 Le volume dans le lecteur C n’a pas de nom.
 Le numéro de série du volume est C699-EB73

> Répertoire de c:\tools\Sming\Sming

>19/10/2015  10:55    <DIR>          .
19/10/2015  10:55    <DIR>          ..
25/06/2015  05:53             8 597 .cproject
25/06/2015  05:53               815 .project
19/10/2015  10:55    <DIR>          .settings
19/10/2015  10:55    <DIR>          appinit
19/10/2015  10:55    <DIR>          apptest
19/10/2015  10:55    <DIR>          compiler
19/10/2015  10:55    <DIR>          include
19/10/2015  10:55    <DIR>          Libraries
25/06/2015  05:53             7 687 License.txt
25/06/2015  05:53             8 323 Makefile
25/06/2015  05:53               554 Makefile-bsd.mk
25/06/2015  05:53               541 Makefile-linux.mk
25/06/2015  05:53               533 Makefile-macos.mk
25/06/2015  05:53            13 841 Makefile-project.mk
25/06/2015  05:53               540 Makefile-windows.mk
19/10/2015  10:55    <DIR>          Services
19/10/2015  10:55    <DIR>          SmingCore
19/10/2015  10:55    <DIR>          system
19/10/2015  10:55    <DIR>          Wiring
               9 fichier(s)           41 431 octets
              12 Rép(s)  73 455 456 256 octets libres

[... 13 lines cut ...]
19/10/2015  10:46    <DIR>          ESP8266_RTOS_SDK
19/10/2015  10:46    <DIR>          ESP8266_SDK
19/10/2015  10:46    <DIR>          ESP8266_SDK_094
19/10/2015  10:46    <DIR>          ESP8266_SDK_095
19/10/2015  10:46    <DIR>          ESP8266_SDK_101
19/10/2015  10:46    <DIR>          ESP8266_SDK_110
19/10/2015  10:46    <DIR>          ESP8266_SDK_120
16/10/2015  09:40    <DIR>          ESP8266_SDK_130
21/11/2014  11:22               180 Espressif web site.url
16/10/2015  09:12    <DIR>          examples
16/10/2015  09:12    <DIR>          extra
25/03/2015  09:16           137 750 Install.ico
19/10/2015  10:46    <SYMLINKD>     sdk [c:\Espressif\ESP8266_SDK]
19/10/2015  10:46         1 404 674 unins000.dat
19/10/2015  10:42         1 312 112 unins000.exe
19/10/2015  10:46            23 001 unins000.msg
21/11/2014  10:45           147 878 Uninstall.ico
19/10/2015  10:45    <DIR>          utils
16/10/2015  09:11    <DIR>          xtensa-lx106-elf
               9 fichier(s)        3 046 670 octets
              17 Rép(s)  73 457 586 176 octets libres

>c:\tools\sming\Sming>where make
C:\SysGCC\esp8266\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe

lxkarthi, 2015-12-15:
is there any solution for this?

kireevco, 2015-12-15:
@lxkarthi solution is to make sure you are using proper tools. 
For example @asez73 has `C:\SysGCC\esp8266\bin\make.exe` in his path, and make.exe from mingw comes only after this.

So try moving all mingw stuff to the front in your PATH.

lxkarthi, 2015-12-15:
Thanks. I figured it out. my cygwin path was before mingw path. So it was using cygwin make. so \ and / problems arose. Now I removed other paths to make.exe in %PATH%. It works now.  This is what I did in cmd.exe
> where make
C:\cygwin64\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe
> echo %PATH%
C:\cygwin64\bin;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin;
> setx PATH "C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin;C:\cygwin64\bin;"
It works now. Thanks.

hreintke, 2016-01-27:
Solved.

closed 2016-01-27 by hreintke