SmingHub/Sming #209: ftp 502 Size not Supported

crosofg, 2015-08-11:
In the HpptServer_ConfigNetwork upon uploading files on the ftp server, I get the following error.
Tcp Server onClient 
USER: 'me'
> 331 OK

PASS: '123'
checkUser: me 123
> 230 OK

SYST: ''
> 215 Windows_NT: Sming Framework

PWD: ''
> 257 "/"

TYPE: 'I'
> 250 OK

SIZE: '/'
> 502 Not supported

QUIT: ''
> 221 OK

anakod, 2015-08-11:
What FTP client you using? And why size is needed?

crosofg, 2015-08-11:
that is the erro lof ins the serial port. I am trying to implement this example https://github.com/anakod/Sming/tree/master/HttpServer_ConfigNetwork
when index.html is not present. 
As per the source code 
>fileSetContent("index.html", "<h3>Please connect to FTP and upload files from folder 'web/build' (details in code)</h3>");

I entered the url ftp://esp_ip:21 
username me
password 123
and following is the output on serial port.
Tcp Server onClient 
USER: 'me'

331 OK
PASS: '123'
checkUser: me 123

230 OK
SYST: ''

215 Windows_NT: Sming Framework
PWD: ''

257 "/"
TYPE: 'I'

250 OK
SIZE: '/'

502 Not supported
QUIT: ''

221 OK

anakod, 2015-08-11:
SIZE command isn't supported in SMing FTP now. Also it shouldn't be required command in my opinion.

crosofg, 2015-08-11:
so how do i upload web files?

kireevco, 2015-08-23:
@crosofg 
> What FTP client are you using?

Seems like SIZE command is issued by some client. Can you please try FileZilla and report back?

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke