SmingHub/Sming #210: Sming not linking to libpwm.a  (SDK PWM)

hrsavla, 2015-08-12:
Sming PWM api is using sofware timer which has low priority and causes visible problem when trying to drive "LEDs". So I am trying to implement esp sdk pwm api. (new one added in SDK1.11, which is using hw_timer). My trial code is bellow
```
/*Definition of GPIO PIN params, for GPIO initialization*/
#define PWM_0_OUT_IO_MUX PERIPHS_IO_MUX_MTDI_U
#define PWM_0_OUT_IO_NUM 12
#define PWM_0_OUT_IO_FUNC  FUNC_GPIO12

#define PWM_1_OUT_IO_MUX PERIPHS_IO_MUX_MTDO_U
#define PWM_1_OUT_IO_NUM 15
#define PWM_1_OUT_IO_FUNC  FUNC_GPIO15

#define PWM_2_OUT_IO_MUX PERIPHS_IO_MUX_MTCK_U
#define PWM_2_OUT_IO_NUM 13
#define PWM_2_OUT_IO_FUNC  FUNC_GPIO13

#define PWM_3_OUT_IO_MUX PERIPHS_IO_MUX_MTMS_U
#define PWM_3_OUT_IO_NUM 14
#define PWM_3_OUT_IO_FUNC  FUNC_GPIO14

#define PWM_4_OUT_IO_MUX PERIPHS_IO_MUX_GPIO5_U
#define PWM_4_OUT_IO_NUM 5
#define PWM_4_OUT_IO_FUNC  FUNC_GPIO5

#include <user_config.h>
#include <SmingCore/SmingCore.h>
\\Sming Not recognizing pwm.h functions. So need to include bellow.
#include <../../../../Espressif/ESP8266_SDK/include/pwm.h>  

void init()
{
	  uint32 pwm_duty_init[8] = {0};

        uint32 io_info[][3] =
{{PWM_0_OUT_IO_MUX,PWM_0_OUT_IO_FUNC,PWM_0_OUT_IO_NUM},
{PWM_1_OUT_IO_MUX,PWM_1_OUT_IO_FUNC,PWM_1_OUT_IO_NUM},
{PWM_2_OUT_IO_MUX,PWM_2_OUT_IO_FUNC,PWM_2_OUT_IO_NUM}};
pwm_init(500,pwm_duty_init,3,io_info);
}
```
When I compile, I am getting bellow error
```
CLEAN SUCCESSFUL (total time: 3s)
C+ app/application.cpp
C+ /c/tools/Sming/Sming/appinit/user_main.cpp
AR out/build/app_app.a
LD out/build/app.out
out/build/app_app.a(application.o):(.text._Z4initv+0xc): undefined reference to `pwm_init(unsigned int, unsigned int*, unsigned int, unsigned int (*) [3])'
out/build/app_app.a(application.o): In function `init()':
C:\tools\MyCode\ESP_PWM/app/application.cpp:35: undefined reference to `pwm_init(unsigned int, unsigned int*, unsigned int, unsigned int (*) [3])'
collect2.exe: error: ld returned 1 exit status
make.exe": *** [out/build/app.out] Error 1

BUILD FAILED (exit value 2, total time: 13s)
```
It seems that linker is not able to find libpwm.a  (this is in ESP_SDK/include)

There is similar discussion on Espressif form (http://bbs.espressif.com/viewtopic.php?f=7&t=572&hilit=pwm+pwm+linker)

Could you please point how to link this file in Sming.

hreintke, 2015-08-12:
@hrsavla : I think a simular issue came up on gitter a few weeks ago by @Tavalin.
```
@hrsavla have you got an example using the SDK PWM API?
So far I've tried amending the sming core makefile to add libpwm and I've added the pwm.h but if I try and use one of the functions like pwm_start() the compiler or the linker complains that it's undefined or not found
```
Did some experiments then with the following results : 
```
@Tavalin : no guarantee that this is the solution but..
I added pwm.h to an example
I added pwm to makefile-project (LIBS = pwm microc microgcc hal phy pp net80211 lwip wpa main sming $(EXTRA_LIBS))
I added the line "extern void pwm_start();" to the sming file : esp_systemapi.h
I added the call : pwm_start() in the example
And compile and link is without error 
```
That is the way I got it compiled -> I did not run it.

Tavalin, 2015-08-12:
Yes I had the same problem but have been too busy to try to get it working.  @hreintke could we inclucde pwm.h in esp_systemapi.h instead of doing "extern void pwm_start();"?

hrsavla, 2015-08-12:
@hreintke Thanks, I got the code compiled and Ran it successfully. I will try to write code with functions similar to Sming PWM.cpp and make PR for it. 
@Tavalin Just including pwm.h causes confusion for compiler between sming PWM.h and sdk pwm.h (It seems, compiler looks at files without case sensitivity).

closed 2015-08-12 by hrsavla

Tavalin, 2015-08-12:
@hrsavla excellent news! Maybe if you're able to create substitute functions for Sming PWM.h using SDK PWM then maybe we can get rid of current Sming PWM or rename to SoftwarePWM?