SmingHub/Sming #212: the updated version is not build!

STASIUK, 2015-08-12:
I installed sming through Choco.
Then replaced "C:\tools\sming" folder, file archives with github.com/anakod/Sming.
I get errors when build any projects.

kireevco, 2015-08-12:
@STASIUK Please do troubleshooting steps from [image] and provide us with with the output.

STASIUK, 2015-08-12:
Fine. Now, as the module flash Basic_Blink, get out a
>exception (0):
epc1=0x746e4565, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 20816, room 16 
tail 0
chksum 0x27
load 0x3ffe8000, len 1344, room 8 
tail 8
chksum 0x7e
csum 0x7e



JustACatRu, 2015-08-13:
Maybe the problem is in UDK version 2.0.7?
Please, read gitter starting from this message: https://gitter.im/alonewolfx2/Sming?at=55cbc693057d8c9d3a6d1467
Try using UDK version 2.0.6.

alonewolfx2, 2015-08-13:
id you tried full flash erase and re flash firmware?