SmingHub/Sming #213: Basic Air Update -Power Down

crosofg, 2015-08-13:
Is they any way to make the over the air update failsafe? If power goes down in the middle of writing to  flash, is there a way to revert back?

crosofg, 2015-08-14:
Can we add an enhancement which partitions the memory into two and copies the firmware in the unused part.And if the write process is successful only then esp will reboot with that firmware. Just like the user1.bin and user2.bin as given in the espressif documentation

kireevco, 2015-08-23:
@anakod @alonewolfx2 @hreintke 
This is a sane suggestion, that everyone was talking about multiple times. What stops us from doing this?

alonewolfx2, 2015-08-23:
time :)

hreintke, 2015-08-27:
@kireevco : I have no experience in OTA and memory/flash allocation. Focusing on tcp/http/debug/commandprocessing for now.

robotiko, 2015-09-28:
Closing: New OTA update using rBoot is failsafe and the recommended way for OTA.

closed 2015-09-28 by robotiko