SmingHub/Sming #216: Clean way of Closing TCP Client / TCP Connection

hrsavla, 2015-08-18:
Closing TCP client connection by close() which is used by Mqtt / tcp client / http client / Websocket client takes some time about 3-4 seconds. Closing of TCP client can occur due to user input or server failure/timeout, router shut down, etc.  If an attempt is made to reconnect the client before closing of connection then TCP connection error -8 (reset/abort) occurs. 

After every close() of tcp client socket I am waiting for 5 second before new connection is made. But is there a clean way close Tcp client and ensure connection .

anakod, 2015-08-18:
I think will be fine implement additional state: **closing**, and after that, after confirmation apply **closed** state.

kireevco, 2015-08-23:
@anakod Would you have time to work on this soon?

hreintke, 2015-08-25:
@hrsavla : Can you provide the/a complete application which shows the error(s) and I can run here ?
I will then check whether I can find reason/solution for this.

hreintke, 2015-11-29:
Related to #449. Follow up will be there

closed 2015-11-29 by hreintke