SmingHub/Sming #218: Allow disable of SPIFF area reflash

JustACatRu, 2015-08-18:
File: \Sming\Sming\Makefile-project.mk String number: 320 In "flash: all" area.
```
$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin $(SPIFF_START_OFFSET) $(FW_BASE)/spiff_rom.bin
```

So we are reflashing spiff_rom.bin all the time we reflash the fw.
But why do we need it?
It takes additional time to flash. And when debugging something it is really annoying.
Maybe add some parameter to makefile? To create/flash spiff_rom.bin or not.

avr39-ripe, 2015-08-18:
I made target flash_app in my repo for flashing spiff-less projects.. but when some project NEEDS spiffy fs image it AFAIK MUST be written every time main firmware flashes due to spiffy fs start depends on main fw size. Somebode already suggest to disable fs mount in user_main for fs-less app..
We must find some good way to disable/enable fs support in main fw and to enable/disable reflashing spiffy.bin each time.

JustACatRu, 2015-08-18:
Does spiffy.exe have some parameters or not?
I want to pass it the size of fs that I want it to create.
But do not know how. And it always creates a file 192 KB size. And we flash it every time...
While testing I need only about 3 KB and all other is just 0xFF...
What a waste of time and flash cycles? :)

ADiea, 2015-08-18:
Hmm i think #208 partially fixes this issue. Let me clarify, you also need a flag to disable every time spiffs flash but keep spiffs support on?

JustACatRu, 2015-08-19:
I've made some changes in \Sming\Sming\Makefile-project.mk" for "$(SPIFF_BIN_OUT):" block:
```
$(SPIFF_BIN_OUT):
	# Generating spiffs_bin
	$(vecho) "Checking for spiffs files"
	$(Q) if [ -a "spiff_rom.bin" ] ; then \
			echo "./spiff_rom.bin found - use it."; \
			cp spiff_rom.bin $(FW_BASE)/spiff_rom.bin; \
	else \
			if [ -d "$(SPIFF_FILES)" ]; then \
					echo "$(SPIFF_FILES) directory exists. Creating spiff_rom.bin"; \
					spiffy; \
					mv spiff_rom.bin $(FW_BASE)/spiff_rom.bin; \
			else \
					echo "No files found in ./$(SPIFF_FILES)."; \
					echo "Creating empty spiff_rom.bin ($$($(GET_FILESIZE) $(SMING_HOME)/compiler/data/blankfs.bin) bytes)"; \
					cp $(SMING_HOME)/compiler/data/blankfs.bin $(FW_BASE)/spiff_rom.bin; \
			fi; \
	fi;
	$(vecho) "spiff_rom.bin---------->$(SPIFF_START_OFFSET)"
```
I've added block, which checks: do we have precompiled spiff_rom.bin file in the project root dir - if yes - it just copies it to $(FW_BASE)/spiff_rom.bin and use it.
What is the profit?
For me: I've made small spiff_rom.bin 2KB in size for my project.
2KB flashes much faster than 192KB...

What do you think? Can we add this changes to Sming?
It does not break default functionality.

kireevco, 2015-08-19:
@JustACatRu Before spiff_rom.bin was required to format fs to fix issues (remind which one @anakod @alonewolfx2 ?)
What we should do here is confirm if it is still required, and if not, add a conditional to the 
`$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin $(SPIFF_START_OFFSET) $(FW_BASE)/spiff_rom.bin` part, and remove
```
 else \
                    echo "No files found in ./$(SPIFF_FILES)."; \
                    echo "Creating empty spiff_rom.bin ($$($(GET_FILESIZE) $(SMING_HOME)/compiler/data/blankfs.bin) bytes)"; \
                    cp $(SMING_HOME)/compiler/data/blankfs.bin $(FW_BASE)/spiff_rom.bin; \
``` 
part.

Thoughts?

JustACatRu, 2015-08-19:
@kireevco Yes, if we do not need blankfs.bin anymore, then this last "else" section may be removed.
But I myself more interested in first "if" part :)

alonewolfx2, 2015-08-19:
@kireevco blankfs.bin not necessary when use spiff_rom.bin. we can remove in makefile.

kireevco, 2015-08-19:
@alonewolfx2 I'm talking about `cp $(SMING_HOME)/compiler/data/blankfs.bin $(FW_BASE)/spiff_rom.bin; \` when blankfs is used instead of spiff_rom.bin (when there is no `files/` directory)

alonewolfx2, 2015-08-19:
@kireevco ohh right. if there is no file in files folder you need to flash blankfs.bin for clearing spiffy area

kireevco, 2015-08-19:
do we need a large one or we can have a very small one, like 2K? If yes,
let's change blankfs.bin to 2K.

alonewolfx2, 2015-08-19:
i think min 4k but i need to try or you can try if you have module

JustACatRu, 2015-08-19:
@kireevco I have no problem flashing my 2KB spiff_rom.bin - it works on device.
But I need that it be my own spiff_rom.bin - I have some small files in it :)
If spiffy.exe can create smaller file (or reduce it after adding all files in it) - then we can just use spiffy.exe all the time. But I think that spiffy.exe does not have such ability and creates 192KB file all the time (with many 0xFF's in the end)...

alonewolfx2, 2015-08-19:
@JustACatRu maybe we can add no file option in spiffy for small blankfs. what do you think?

JustACatRu, 2015-08-19:
@alonewolfx2, I think, that the right behavior of spiffy.exe should be like this:
- Check some given by parameters directory for files in it
-- if found
--- create correct spiff_rom.bin with this files in it
--- reduce spiff_rom.bin size to fit this files only (why do we need those 0xFF in the end of this file?)
-- if no files in dir
--- then just create some empty small spiff_rom.bin of given by parameters size (let's say 4KB)

ADiea, 2015-08-19:
@alonewolfx2 @kireevco doesn't spiffy support arguments to set filesystem size?

alonewolfx2, 2015-08-19:
@ADiea my spiffy build (we are using my build) has support spiffy size. 
@JustACatRu you are right. i will look and add this when i have free time

JustACatRu, 2015-08-19:
@alonewolfx2, great! Thank you!

anakod, 2015-08-19:
It will be great to make autoatic size calculation for file system (by given files)

tprochazka, 2015-08-23:
And must be spiff area before program area? Can spiff be at the end of flash storage, at the fixed address and program from the beginning?

tprochazka, 2015-08-23:
Also is very useful to change flash speed by
COM_SPEED_ESPTOOL ?= 921600

But reflashing spiff_rom.bin all the time also cause "First init file system" which take a really long on ESP12E.

kireevco, 2015-08-23:
The problem will be solved by putting a smaller blank.bin file instead of the one that we have right now.
It will take @alonewolfx2 forever, @JustACatRu can you please look into http://github.com/alonewolfx2/spiffy and add a change to generate an empty 4k file?

tprochazka, 2015-09-01:
But original request was prevent to reflash SPIFF area again and again. Smaller blank.bin doesn't solve situation when you need to use quite large SPIFF area, but you don't want to reflash it after every change in the code. Especially if you are working for example on the app that collect data and store it to the SPIFF, every small change in the code clear it completely.

tprochazka, 2015-09-01:
Best solution is possibility to init empty SPIFF for example 1MB (on 4MB module) and keep it forever intouch. And make changes only by FTP.

tprochazka, 2015-10-31:
It is at least possible do something like that?
I mean re flash program and keep spiff partition?

raburton, 2015-10-31:
If you are using `spiffs_mount` (instead of `spiffy_mount_manual`) the start position is not fixed. This means that changes to your app, which cause the rom size to change, might shift the start offset of the spiff file system on the flash, so it needs to be flashed every time.

If you are convinced the rom size hasn't changed you can use the command `DISABLE_SPIFFS=1 make flash` to temporarily disable spiffs when you flash, but if your filesystem doesn't work properly afterwards you'll know why.

closed 2015-11-09 by raburton