SmingHub/Sming #222: MQTT TCP connection closes randomly..

dmarkey, 2015-08-20:
onReadyToSendData: 3
TCP received: (null)
TCP connection closing
-TCP connection

Then on the mosquitto side:

onReadyToSendData: 3
TCP received: (null)
TCP connection closing
-TCP connection

Using the latest Sming and the latest SDK.

dmarkey, 2015-08-21:
http://dmarkey.com/~dmarkey/file.pcap here is a trace with the problem. Seems to be not handling a re-transmission properly?

kireevco, 2015-08-23:
@dmarkey thank you so much for providing trace. @anakod what is requried to troubleshoot / fix this?

tve, 2015-10-14:
I'm not seeing any problem highlighted by the packet trace. There's a packet missing in the trace according to wireshark. In the end, the server closes the connection cleanly. If the missing packet (according to the sequence numbers) wasn't sent by the esp8266 I would suspect a buffering problem.

dmarkey, 2015-10-14:
Well the socket isnt closed cleanly on the server side:

Socket error on client SmartController-10364675, disconnecting.

m0bygit, 2016-01-21:
@dmarkey @ADiea I had a similar problem with Mqtt. My problem was solved with commit 19bcf85add9e6fb4cbe7c61065c4129d9abcb496.

hreintke, 2016-03-25:
Closed. Fixed by #550

closed 2016-03-25 by hreintke