SmingHub/Sming #223: make: *** [out/firmware/spiff_rom.bin] Error 1

cartoonincm, 2015-08-20:
i got form "HttpServer_WebSockets" only.
other example project work for me.
Can you help me to solve please.

Checking for spiffs files
files directory exists. Creating spiff_rom.bin
/bin/sh: spiffy: command not found
mv: rename spiff_rom.bin to out/firmware/spiff_rom.bin: No such file or directory
make: *** [out/firmware/spiff_rom.bin] Error 1

anakod, 2015-08-20:
You should install spiffy

cartoonincm, 2015-08-20:
how to install that?

cartoonincm, 2015-08-20:
I found turtorial in "Linux QuickStart"
link this 
----------------------------------------------------------------------------------------------------------------------------------------
wget https://bintray.com/artifact/download/kireevco/generic/spiffy-linux-1.0.1.tar.gz
tar -zxf spiffy-linux-1.0.1.tar.gz && sudo mv spiffy /usr/local/bin/
sudo chmod +rx /usr/local/bin/spiffy
----------------------------------------------------------------------------------------------------------------------------------------

But i don't know about OS X

kireevco, 2015-08-23:
@caquino Please try a prev. version from here: https://bintray.com/artifact/download/kireevco/generic/spiffy-1.0.3-macos-x86_64.tar.gz

I'll build 1.0.4 as soon as I can.

piperpilot, 2015-08-24:
You can also clone spiffy from here:

https://github.com/alonewolfx2/spiffy.git

Create a build directory inside of the spiffy folder, run make, copy the spiffy executable to your /usr/bin/ directory and make sure it is executable.

cartoonincm, 2015-08-24:
@piperpilot  wow wow now it worked thanks a lot

piperpilot, 2015-08-24:
I will add these instructions to the OSX quick start.  Go ahead and close your issue if this is solved for you.

cartoonincm, 2015-08-24:
in my side it worked every thing. thank you so much. :)

anakod, 2015-08-24:
Great. I think we should improve our quick start guides and related information.

closed 2015-08-24 by anakod