SmingHub/Sming #224: spiff build looks in files directory instead of SPIFF_FILES location

piperpilot, 2015-08-21:
There is a SPIFF_FILES variable that does a check to see if the folder exists, but it isn't honored when running spiffy.  Most of the projects that have files are configured to use web/build or web/dev.  It would make more sense to honor those.  Also spiffy is called with defaults and no file size specified.  We should probably calculate 1/2 of the FS space based on the settings of SPI_SIZE.

I am happy to make these changes and submit a pull request, but I wanted to make sure that there wasn't a different intention on using the default files vs the config variable.

kireevco, 2015-08-23:
@piperpilot which spiffy version are you using? did you use choco to install?
Did you check https://bintray.com/kireevco/generic/spiffy/1.0.4/view#files ?

piperpilot, 2015-08-24:
To be honest, I'm not sure which version of spiffy I'm using...I think its one I found on github.  We should probably post the above link, or better yet have a github repo for whatever spiffy is recommended for use with Sming.  This will be a big point of confusion for anyone wanting to use spiffs.

No, I didn't use choco to install.  I'm on a mac, I downloaded the source, built it and put it into my bin dir.  I will replace my current version with the above.

piperpilot, 2015-08-24:
Closing this issue based on the acceptance of my pull request to fix this...previously spiffy was being called from the makefile with "spiffy; \"  I changed it to accept the parameters you list:

"spiffy $(SPIFF_SIZE) $(SPIFF_FILES); \"

I also added the ability to specify in the user makefile.

closed 2015-08-24 by piperpilot