SmingHub/Sming #226: ends Heap size in http.

STASIUK, 2015-08-22:
I did notice that every request free Heap size decreases.

kireevco, 2015-08-23:
@STASIUK Please provide us with a test case? Also, maybe a screencast via [jing](https://www.techsmith.com/jing.html)

STASIUK, 2015-08-24:
```
void onGetGroups(HttpRequest &request, HttpResponse &response)
{
	JsonObjectStream* stream = new JsonObjectStream();
	JsonObject& json = stream->getRoot();
	json.add<int>("Count", groupController.groupList.count());
	JsonArray& groups = json.createNestedArray("groups");
	for(int i = 0; i < groupController.groupList.count(); i++){
		JsonObject& groupObj = groups.createNestedObject();
		groupObj.addCopy("nameId", groupController.groupList[i].nameId);
		groupObj.addCopy("languageMain", groupController.groupList[i].languageMain);
		groupObj.addCopy("languageOther", groupController.groupList[i].languageOther);
	}
	response.setAllowCrossDomainOrigin("*");
	response.sendJsonObject(stream);
	Serial.print("Free size: ");
	Serial.println(system_get_free_heap_size());
}
```
And with every request, I get in serial:
>Free size: 23648
Free size: 23024
Free size: 22144
Free size: 19096
Free size: 18944
Free size: 16816
Free size: 14512
Free size: 11480
Free size: 8872
Free size: 5616

robotiko, 2015-08-25:
Hi, 
I have same issue in tested code since I updated to Commits on Aug 24, 2015 
I rolled back sming to Anakod  merge on Aug 16, 2015  (3c7f1401ce0f18820f0a308692e22915c963492a) and the tested code works. 

Looks like the Debug of printf for floats can be causing trouble.

anakod, 2015-08-25:
@robotiko, can you provide more details please?

robotiko, 2015-08-25:
Hi @anakod, 
I'm hunting this thing down. 
Preliminary tests shows that rolling back Sming, gets back to work existing code. 

However, looks like it could be a side effect of a quite big firmware  & a bigger framework footprint that ends ups in heap corruption.  (current firmware file size is 258.346 bytes)

I added some code in the working Sming version and has similar crash. 
It might be related to https://github.com/anakod/Sming/issues/221 however it crashes during code execution, not in boot (looks like a coincidence where it crashes).

I found a scary issue in esp arduino that smell like and might be also related to this (https://github.com/esp8266/Arduino/issues/428) .

I will slim code and move forward to current Sming release to see if it related to firmware size and not the Sming code itself  to narrow the issue. 

I will post the results here.

kireevco, 2015-08-25:
Good reason for #230 (just saying) :)

robotiko, 2015-08-25:
Ok promised update, 

further testing:  same code in latest Sming crashes. 
In my case always in the same place too:  right after downloading a json file, before parsing it, so no reason for it.  The json is always the same and the json lib and code too, and in previous sming parses, and now right after downloading crashes. 

About the firmware size, it seems sligthly smaller with latest sming 258.346 bytes (old) vs
258.182 bytes (head) so it might not be the reason. 

However, as pointed some days ago, the rom burning order might be causing trouble because boundaries my overlap.  http://tech.scargill.net/esp8266-flash-sizes-and-sdk-issues/

Another weird thing: I'm using a nodemcu dev kit 1.0 with 4MB flash and now the spiff size is  524.288 bytes  however on boot is shows 
"fs.start: size:192 Kb, offset:0x%X
mount res: 0"  ??
192KB?   firmware and spiffy are both bigger than that.

The serial debug on crash 
--
Fatal exception (0):
epc1=0x2020200a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 22696, room 16 
tail 8
chksum 0x8d
load 0x3ffe8000, len 1344, room 0 
tail 0
chksum 0x65
load 0x3ffe8540, len 12624, room 8 
tail 8
chksum 0x80
csum 0x80
Œ‚5MEM CHECK FAIL!!!
är
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90

hreintke, 2015-08-27:
@STASIUK : What class is your groupController ?
If you only get posts and no remove's could that be increasing heap usage ?

hreintke, 2015-08-27:
Did some testing with this in the application : 
```
int previousFreeMem = 0;
Timer memTimer;

void memoryPrint()
{
	int curFreeMem = system_get_free_heap_size();
    if (curFreeMem != previousFreeMem)
    {
		uint32_t curMillis = millis();
		Serial.printf("Dbg %4d.%03d : ", curMillis/1000, curMillis % 1000);
        Serial.printf("----------- > Free size : %d\r\n",curFreeMem);
        previousFreeMem = curFreeMem;
    }
}
```
and calling in init()
```
	memTimer.initializeMs(250,memoryPrint).start();
```
This shows heap usage over time.
With me it has the following results : 
- Initiating "slowly" http requests (initiating next request when previous is finished 
 during processing free heap gets lower, after closure returns to initial value
- Initiating "high speed request" (starting wget as speedy as possible by hand)
 Free heap gets lower during processing requests, however after stopping sending requests, the heapsize returns to it's initial value (takes some time)
- The heap size only gets lower very slowly
Possibly to the limited POST requests I am generating

@STASIUK : 
- At what rate are your POST request coming to the ESP ?
- Can you provide me a POST request you are processing ?
- Could you do a simular test in your environment using your POSTs ?

anakod, 2015-08-27:
> The heap size only gets lower very slowly Possibly to the limited POST requests I am generating

Is it restored back after some time?

hreintke, 2015-08-27:
@anakod : Yes, in my testcase it is restored after some time.
That's why I asked @STASIUK the questions above as maybe he has different experience.
I am thinking of pcb's (and other data) kept in lwip until TIME_WAIT expires.
Do you have other first thoughts ?

anakod, 2015-08-27:
> I am thinking of pcb's (and other data) kept in lwip until TIME_WAIT expires.

Yes, I think so too, but additional tests needed, I think.

piperpilot, 2015-09-01:
Yeah, I'm seeing some similar things here...it seems that heap is used but reclaimed very slowly.  I have an ajax refresh on my main page, it runs every 5 seconds...over time, the system will run out of heap and reboot.  I'm not understanding why the heap isn't freed up properly when the connection closes.

closed 2016-04-20 by hreintke