SmingHub/Sming #227: TimerDelegate to support class methods WITH parameters (arguments)

avr39-ripe, 2015-08-22:
Is there any possibility for TimerDelegate to accept class methods with arguments? For one of my tasks I need start some class method with one uint8_t parameter after some amount of time. With class method without any arguments - no problem, TumerDelegate(&myClass::some_method,this) does the trick, but if i need to call myClass::some_method(uint8_t id) in timer - no way..
I found one, but I think it ugly and dirty hack..
I subclass Timer..

typedef Delegate<void(uint8_t)> RoomTimerDelegate;                                                            

class roomTimer : public Timer                                                                                
{
public:                                                                                                       
        roomTimer(uint8_t room_id);                                                                           
        roomTimer& IRAM_ATTR initializeMs(uint32_t milliseconds, RoomTimerDelegate delegateFunction = NULL);  
        roomTimer& IRAM_ATTR initializeUs(uint32_t microseconds, RoomTimerDelegate delegateFunction = NULL);  
        void IRAM_ATTR setCallback(RoomTimerDelegate delegateFunction);                                       
        void IRAM_ATTR start(bool repeating = true);                                                          
        void __forceinline IRAM_ATTR startOnce() { start(false); }
        int room_id;                                                                                          
        static void IRAM_ATTR processing(void *arg);                                                          
        RoomTimerDelegate delegate_func = nullptr;                                                            

};   

most of roomTimer implementation is just copy-paste one-bu-one of Timer code, changeing TimerDelegate to RoomTimerDelegate and adding room_id to constructor.. and then in roomTimer::processing:

copy-paste..

 if (ptimer->callback)
                {
                        ptimer->callback();
                }
                else if (ptimer->delegate_func)
                {
                        ptimer->delegate_func(ptimer->room_id);
                }
         so Delegate routine called WITH room_id parameter..
It works.. Really.. but it is as I can see very bad idea to solve my original task like this..
Moreover, at first, I do not implement roomTimer::start method.. without implementing it (by copy-pasting Timer::start) all my roomTimer stuff do not work...
So may be some one with solid C++ can impement things so that TimerDelegate can be any function with  void to return and ANY count and type of arguments? I look inside Delegate.h and there is plenty of typename ... Paramlist.. I feel like this is the way to go, but my C++ is too weak.. :(

Or may be there is some VERY different way to start some function WITH parameters after some time?

kireevco, 2015-08-23:
@anakod where is this on the priority list? What would it take to accomplish this?

hreintke, 2015-08-24:
I think other solution based on object oriented would be better. will provide example on Wednesday

hreintke, 2015-08-26:
@avr39-ripe : I checked the code as you referred to in chat. Please make sure that this github issue holds all information to discuss. Otherwise it is always a guess whether we talk about the same. 
My first thoughts are also : The callback is an object function so you should have Room defined as an object. Now you are holding information of a room in a structure (= class without constructor) and in _rooms table. With a complete room class I think the addition of a changed timer class is not required.
BTW : I presume your code is example and not running code. Otherwise I misunderstand the usage of delegate. 
Definition of 
```
typedef Delegate<void(uint8_t)> myTimerDelegate;
```
and the usage 
```
 myTimerDelegate(&HeatingSystem::_room_coldy_lo_t_off_delayed, this, room_id)
```
is not compatible, and should be changed to get it working.
Then on implementing Timerdelegate with addtional parameter. The delegate implementation in sming allows definition of delegatefunctions with a fixed number of (different) parameters. There is no option to have a variable number of parameters.
Saying that, the timer class would need additional functions to make the use of an additional parameter (which I think should be "(void*) arg ") usable in a generic way. If @anakod is accepting the enhancement request I will propose a solution.

anakod, 2015-08-27:
You can make simple solution for accepting arguments for timer listener instance (in pseudocode):

```
class Room
{
	int id;
public:
	Room(int rid) : id(rid) {}
	
	void timerCallback()
	{
	// Your timer processing code here
	// Please, use "id" as needed
	...
	}
}

room = new Room(125);
timer.setCallback(room.timerCallback);
```

avr39-ripe, 2015-08-27:
@hreintke @anakod Thank you gyus! I really appreciate your help! In my original model i made huge mistake by trying to use one timer instance to serve several async behaved objects (rooms) moreover, rooms vas implementet as only struct not class so thats how I came to problem of arguments to timer delegate. Even with them i wouldn't work. And after figuring out impossibility of using just one timer and your advices it can be done in simple and elegant manner.

But! Original question is still interesting by itself - it will be wery handy to pass args to timer delegate.. even just one void* argv.. so looking forvard for c++ monsters to implement it.. ;)

alon24, 2015-08-28:
I would like this simple feature too, I think its better than starting to create classes just for this passing of argument 
Just my 2c

closed 2016-04-20 by hreintke