SmingHub/Sming #228: Spiffy crash during build

tprochazka, 2015-08-23:
Very weird thing happen to me. If I run "ALL" target from Eclipse.
Spiffy crash 

```
MSYS-1.0.12 Build:2012-07-05 14:56
Exception: STATUS_ACCESS_VIOLATION at eip=00000000
eax=00000000 ebx=60EA28AC ecx=004E3506 edx=60EA28AC esi=00000000 edi=60EA00D4
ebp=0028FEC8 esp=0028FE9C program=MinGW\bin\spiffy.exe
cs=0023 ds=002B es=002B fs=0053 gs=002B ss=002B
Stack trace:
Frame     Function  Args
End of stack trace
```

But if I run from the command line directly ``` spiffy 524288   web/build```
It works correctly.

Does anybody have the same problem?

kireevco, 2015-08-23:
@tprochazka What is the spiffy source/version? Did you use one from [here](https://bintray.com/kireevco/generic/spiffy/1.0.4/view) ?

piperpilot, 2015-08-24:
@kireevco which version should b supported with Sming?  I have found many different versions/forks out there...we should recommend the proper one in the wiki or readme so everyone is using the proper version.

tprochazka, 2015-08-24:
I used this links to download spiffy
https://github.com/kireevco/chocolatey-packages/blob/master/manual/spiffy/tools/chocolateyInstall.ps1
yesterday

I hope, that it should be right version.
I will test version from bintray.com

tprochazka, 2015-08-24:
So it is exactly the same file.

alonewolfx2, 2015-09-28:
@tprochazka default spiffy size is 196608(192KB). Propably your files bigger than 192kb and thats why its crashing. I am trying to add auto size calculation in spiffy. i will push shen its ready. 
BTW: we need to add some information on wiki. here is spiffy repo link https://github.com/alonewolfx2/spiffy any pull request are welcome :)

flexiti, 2015-09-28:
@alonewolfx2 
You can add support for the spiffy compilation in a Windows environment? :)

alonewolfx2, 2015-09-28:
@flexiti i am building on windows. just change your msys mingw order in your system path. msys should first

flexiti, 2015-09-29:
@alonewolfx2 
 I have errors like this:
..\src\spiffs_hydrogen.c: In function 'SPIFFS_errno':
..\src\spiffs_hydrogen.c:104:14: error: expected identifier before '(' token
   return fs->errno;
              ^
..\src\spiffs_hydrogen.c: In function 'SPIFFS_creat':
..\src\spiffs_nucleus.h:250:11: error: expected identifier before '(' token
     (fs)->errno = SPIFFS_ERR_NOT_MOUNTED; \
           ^
you can help me a little bit? :)
I am using Eclipse environment and CHERTS

btw. spiffs.h It should be the same as those in the Sming spiffs directory?

kireevco, 2015-09-29:
looks like a msys/mingw problem to me. Make sure to change their paths
order in your PATH variable.

flexiti, 2015-09-29:
its ok now
I have not valid Makefile-project.mk, another reason to make less branches ;) 

thanks

hreintke, 2015-09-29:
@flexiti : Wait a few days and you will see the workflow as we see it now :smile: 
For makefile-project-mk : is that the one from current master branch ?
If so please submit an issue so it is documented

flexiti, 2015-09-29:
@hreintke 
Wait, now compiles correctly (nodemcu-spiffy), but still have old files in esp8266 flash - I still have to work it out and check all directory.
(using @alonewolfx2 source still have errors as above)
btw. can we add spiffy to Sming ? it would be easier

flexiti, 2015-09-29:
I changed in spiffs.h in @alonewolfx2  spiffy 
 // last error
  s32_t errno;  to  s32_t errno1;   and also in other files
Spiffy from alonewolfx2 also compiles correctly now

Generate firmware images successully in folder out/firmware.
Done 
# Generating spiffs_bin
Checking for spiffs files
files directory exists. Creating spiff_rom.bin
Creating rom spiff_rom.bin of size 196608 bytes
Adding files in directory files
application.txt added to spiffs (299 bytes)
spiff_rom.bin---------->0x4C000

but in both cases (modemcu and alonewolfx2 spiffy)  the file size application.txt = 0 (read by ftp)
(i am working with standard boot)

kireevco, 2015-09-29:
@flexiti
This is not a right solution, spiffy won't work correctly after this.
Please paste follow
https://github.com/SmingHub/Sming/wiki/Troubleshooting-Windows and paste us
the output.

flexiti, 2015-09-29:
C:\Users\mariuszb>echo %PATH%
C:\mingw\msys\1.0\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Java\jdk1.8
.0_11\bin;C:\Program Files\Common Files\Borland Shared\BDE;C:\Program Files\ATI
Technologies\ATI.ACE\Core-Static;C:\Program Files\AMD\ATI.ACE\Core-Static;C:\Pro
gram Files\Java\jdk1.8.0_11\bin

C:\Users\mariuszb>echo %SMING_HOME%
C:\tools\sming\Sming

C:\Users\mariuszb>echo %ESP_HOME%
c:\Espressif

C:\Users\mariuszb>dir %SMING_HOME%
 Wolumin w stacji C to Seven
 Numer seryjny woluminu: B81F-BE40

 Katalog: C:\tools\sming\Sming

2015-09-01  11:13    <DIR>          .
2015-09-01  11:13    <DIR>          ..
2015-09-05  13:17             6 723 .cproject
2015-08-31  16:18               815 .project
2015-09-01  11:11    <DIR>          .settings
2015-09-01  11:11    <DIR>          appinit
2015-09-01  11:11    <DIR>          apptest
2015-09-01  11:11    <DIR>          compiler
2015-09-01  11:11    <DIR>          include
2015-09-05  13:17    <DIR>          Libraries
2015-08-31  16:18             7 687 License.txt
2015-08-31  16:18             6 504 Makefile
2015-08-31  16:18               576 Makefile-bsd.mk
2015-08-31  16:18               567 Makefile-linux.mk
2015-08-31  16:18               567 Makefile-macos.mk
2015-08-31  16:18            10 975 Makefile-project.mk
2015-08-31  16:18               512 Makefile-windows.mk
2015-09-08  11:38    <DIR>          out
2015-09-01  11:11    <DIR>          Services
2015-09-01  11:11    <DIR>          SmingCore
2015-09-01  11:11    <DIR>          system
2015-09-01  11:11    <DIR>          Wiring
               9 plik(ów)             34 926 bajtów
              13 katalog(ów)  98 314 981 376 bajtów wolnych

C:\Users\mariuszb>dir %ESP_HOME%
 
 Katalog: c:\Espressif

2015-08-25  09:19    <DIR>          .
2015-08-25  09:19    <DIR>          ..
2015-07-09  22:34            17 319 changelog.txt
2015-05-21  21:52             2 061 Command line Unofficial Development Kit for
Espressif ESP8266.lnk
2015-04-11  23:37               186 DevKit web site.url
2015-08-25  09:18    <DIR>          docs
2015-08-12  13:38    <DIR>          esp8266-github
2015-08-25  09:19    <DIR>          ESP8266_MESH_SDK
2015-08-25  09:19    <DIR>          ESP8266_RTOS_SDK
2015-08-12  13:36    <DIR>          ESP8266_SDK
2015-06-10  15:46    <DIR>          ESP8266_SDK_094
2015-06-10  15:46    <DIR>          ESP8266_SDK_095
2015-06-10  15:46    <DIR>          ESP8266_SDK_101
2015-06-10  15:46    <DIR>          ESP8266_SDK_110
2015-08-25  09:18    <DIR>          ESP8266_SDK_120
2015-04-11  23:37               180 Espressif web site.url
2015-08-26  13:21    <DIR>          examples
2015-06-10  15:46    <DIR>          extra
2015-03-25  09:16           137 750 Install.ico
2015-07-06  16:41    <SYMLINKD>     sdk [c:\Espressif\ESP8266_SDK]
2015-08-10  22:56           628 701 unins000.dat
2015-06-10  15:44         1 312 112 unins000.exe
2015-08-10  22:56            23 001 unins000.msg
2014-11-21  10:45           147 878 Uninstall.ico
2015-08-12  13:38    <DIR>          utils
2015-06-10  15:44    <DIR>          xtensa-lx106-elf
               9 plik(ów)          2 269 188 bajtów
              17 katalog(ów)  98 314 981 376 bajtów wolnych

alonewolfx2, 2015-09-29:
ok it seems there is no mingw in your path.

flexiti, 2015-09-29:
I added, but unfortunately no change
(without this change also compilation was ok)
See yours source compilation error without any modifiaction:

In file included from src/main.c:1:0:
./src/spiffs.h:183:9: error: field '_errno' declared as a function
   s32_t errno;
         ^
mingw32-make.exe: *** [build/main.o] Error 1
C:/Espressif/examples/nodemcu-spiffy/Makefile:68: recipe for target 'build/main.o' failed

flexiti, 2015-09-29:
in nodemcu-spiffy this line looks in this way:
 // last error
  s32_t errnos;    (not errno but errnos)
Maybe because  of   errno.h  ??
But it is first problem, if I correct this all seams to work but files are corrupted
[image]

index and my-file are generated by my program after startup

alonewolfx2, 2015-09-29:
i dont have errno.h and here is my result on windows
```
C:\Users\**\Desktop\spiffy-master>mkdir build

C:\Users\**\Desktop\spiffy-master>make
... depend build/spiffs_check.d
... depend build/spiffs_cache.d
... depend build/spiffs_hydrogen.d
... depend build/spiffs_gc.d
... depend build/spiffs_nucleus.d
... depend build/main.d
... compile build/main.o
... compile build/spiffs_nucleus.o
... compile build/spiffs_gc.o
... compile build/spiffs_hydrogen.o
... compile build/spiffs_cache.o
... compile build/spiffs_check.o
... linking

C:\Users\**\Desktop\spiffy-master>

C:\Users\Bekir Kaba\Desktop\spiffy-master\build>ls
main.d          spiffs_check.d  spiffs_hydrogen.d  spiffy.exe
main.o          spiffs_check.o  spiffs_hydrogen.o
spiffs_cache.d  spiffs_gc.d     spiffs_nucleus.d
spiffs_cache.o  spiffs_gc.o     spiffs_nucleus.o

```

alonewolfx2, 2015-09-29:
@flexiti your error seems your system still using mingw first. you can try restart machine. you need msys first for building spiffy

raburton, 2015-09-29:
>in nodemcu-spiffy this line looks in this way:
// last error
s32_t errnos; (not errno but errnos)

If you renamed the variable you'd also need to rename it where it's used. Otherwise those uses of the variable will be referring to the existing errno function, not the newly named local variable.

This reminds me of a problem @robotiko fixed for someone the other day, random compile errors that didn't occur for anyone else (building esptool2 on windows). Turned out the person had half a dozen different sdks and compilers installed and stuff was getting picked up from all over the place. Might be worth rationalising your environment and making sure you have a single good base to work from.

flexiti, 2015-09-29:
@alonewolfx2 
Yes, I see
But compile is smaller problem, I can avoid it.
( @raburton yes, I renamed all in spiffy ; single good base , yes, good idea, I will try)
I don't know why I do not have this files on ftp, if I open spiff_rom I even see in text editor that this files are there.
@alonewolfx2  Can You send me Yours spiffy.exe (for test only)  ??
In this way I can see when and where problem is created :)

C:\mingw\msys\1.0\bin;C:\mingw\bin;    ---  is it not ok ? (system was restarted)

flexiti, 2015-09-29:
btw.
according to me
Spiffy should be done from the same sources as Sming SpiffFS (just add main.c and make appropriate makefile). and placed in SmingFramework.
Otherwise, problems may arise.
What is your opinion? @alonewolfx2 @raburton

kireevco, 2015-09-29:
Just curious why don't you want to use a binary?

flexiti, 2015-09-29:
Could you explain more?

hreintke, 2015-09-29:
@alonewolfx2 : When I clone spiffy from xlfe/spiffy (think the version you adapted) I can compile with mingw\bin\gcc -> looks like the issue is caused by your updates.
Would be good to update your spiffy so it is not that dependent on a specific (older ?) gcc version.

flexiti, 2015-09-29:
changing s32_t errno to errnos  ( in all spiffy files) makes the compilation is correct ... that does not change the fact that I still does not have the files on the ftp :)

alonewolfx2, 2015-09-29:
@flexiti we had same issue with @kireevco and my lastest repo compiling fine on windows with msys. @hreintke i didnt tried lastest xlfe's spiffy. sming using different flash block sizes and thats why we cant use if xlfe not changed that. but if he change its not working on nodemcu. if we need main xlfe's repo to use (its not necessary i think) we need to push all changes into the both xlfe and nodemcu repo. 
if everyone need this i can push changes both repo this weekend. 
@flexiti i uploaded spiffy.exe into the gitter sming chat (i builded  now)

alonewolfx2, 2015-09-29:
@flexiti you dont need any changes in my repo including errno.h. if you changed errno's it cant run on chip. i mean it can run but wrong. i saw all what you saw. i had same issues what you had.

flexiti, 2015-09-29:
@alonewolfx2 if I do not change in Your repo my compilation is with errors. 
So what is your suggestion what to do about it?

my PATH: C:\mingw\msys\1.0\bin;C:\mingw\bin;......

alonewolfx2, 2015-09-29:
it should be ok with this orders but is it system path or user path?

flexiti, 2015-09-29:
system, user is empty

flexiti, 2015-09-29:
@alonewolfx2 with Yours spiffy.exe works ok 
( strange difference in the the size : your Spiffy approximately 800 kb, my only 180 kb why?)

hreintke, 2015-09-29:
@alonewolfx2 : 
I am not saying : let's use the spiffy from xlfe repo but..
I can compile xlfe spiffy without problems, your spiffy only with a specific gcc.
My conclusion is then : The compile issues are caused by the updates you made to spiffy in your repo.
I think the best is that you check your spiffy, and hopefully can solve the issue.

kireevco, 2015-09-29:
@hreintke we can't use xlfe spiffy because it doesn't have enough features (specify directory, specify spiffs size), but we need to see if we can make it work with all compilers.

But is this really such a priority?

flexiti, 2015-09-29:
@kireevco virtually any WWW application with internet access required web pages,other files - You want always to manually load each via FTP ??

hreintke, 2015-09-29:
@kireevco : If you read my post above you see that I know we cannot use xlfe spiffy.
However there is an issue with compiling alonwolfx2's version -> a very specific compiler has to be used.
Questions/issues about this popup regularly in the issues here and on gitter.
That is why I asked if he can solve that issue in his repo. 
When that can and is been done, we will have a better user-experience for first time sming users.
And no, this is not top priority, but I would like to have it on a list to be done sometime as now it is not.

kireevco, 2015-09-29:
@flexiti What do you mean? Spiffy takes 'files' directory and makes a filesystem bin. Uploading via FTP is not a production way.

@hreintke I do read posts. 
I would appreciate @alonewolfx2 to find time to fix this too. 
But it has to be done right: xlfe has to get our changes, so we can build from the original repo.

flexiti, 2015-09-29:
@kireevco You are reading posts ? When I read your response I'm not sure

kireevco, 2015-09-29:
@flexiti > You are reading posts ? When I read your response I'm not sure
> @kireevco virtually any WWW application with internet access required web pages,other files - You want always to manually load each via FTP ??

The answer is no. And I don't get why do you think I thought so?

flexiti, 2015-09-29:
At least 4 examples need Spiffy, and I all the day write here that  I (see not only I) have the problems with spiffy so @kireevco  "But is this really such a priority?"  - exactly what you mean by writing this ?

You previously written:
"Just curious why don't you want to use a binary?" 
and as I asked you to explain what you mean you replied nothing.

kireevco, 2015-09-29:
@flexiti you seem very frustrated, sorry to hear that.
> But is this really such a priority

What I mean is that spiffy has been working OK with mingw gcc for months. Fixing it to work in all possible environments takes time, which could be solved somewhere else in development, so "is this really such a priority?"

> Just curious why don't you want to use a binary?
and as I asked you to explain what you mean you replied nothing.

Normally you should do @flexiti when you talk to someone specifically :)

I meant, why don't you download spiffy from [here](https://bintray.com/artifact/download/kireevco/generic/spiffy-1.0.4-win32-x86_64.zip)?

alonewolfx2, 2015-09-29:
@flexiti priority mean update all code for xlfe's repo isn't necessary . We know spiffy necessary for many projects. You can use prebuilded spiffy binary.also we know we need some more information for usage. But.. Almost all spiffy problems come with "file sizes". You need to select right spiff size for correct working and I will add automatic spiff size calculation when I free time or if anyone can pull request it will be fine. 
@hreintke I tried xlfe repo again but I can't build if mingw comes first. How can you compile on Windows with mingw?

kireevco, 2015-09-29:
@hreintke and can you please build from the command line, with `echo %PATH%` printout?

hreintke, 2015-09-29:
```
C:\Userdata>echo %path%
c:\MinGW\bin;c:\mingw\msys\1.0\bin;C:\ProgramData\Oracle\Java\javapath;C:\Utilities\Python27\;C:\Uti
lities\Python27\Scripts;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client
\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0
\;C:\Program Files (x86)\Intel\OpenCL SDK\3.0\bin\x86;C:\Program Files (x86)\Intel\OpenCL SDK\3.0\bi
n\x64;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(
R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Componen
ts\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Sony\
VAIO Improvement;C:\Program Files (x86)\Sony\VAIO BIOS Data Transfer Utility\;C:\Program Files (x86)
\Sony\VAIO Startup Setting Tool;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Utilities\T
ortoiseSVN\bin;C:\Utilities\Calibre2\;c:\utilities\ffmpeg\bin;C:\Utilities\nodejs\;C:\Users\Herman\A
ppData\Roaming\npm

C:\Userdata>cd spiffy

C:\Userdata\spiffy>make clean
... removing build files in build

C:\Userdata\spiffy>make
... depend build/spiffs_check.d
... depend build/spiffs_cache.d
... depend build/spiffs_hydrogen.d
... depend build/spiffs_gc.d
... depend build/spiffs_nucleus.d
... depend build/main.d
... compile build/main.o
In file included from src/main.c:1:0:
./src/spiffs.h:183:9: error: field '_errno' declared as a function
   s32_t errno;
         ^
make: *** [build/main.o] Error 1

C:\Userdata\spiffy>cd ..\spiffy_xlfe

C:\Userdata\spiffy_xlfe>make clean
... removing build files in build

C:\Userdata\spiffy_xlfe>make
... depend build/spiffs_check.d
... depend build/spiffs_cache.d
... depend build/spiffs_hydrogen.d
... depend build/spiffs_gc.d
... depend build/spiffs_nucleus.d
... depend build/main.d
... compile build/main.o
... compile build/spiffs_nucleus.o
... compile build/spiffs_gc.o
... compile build/spiffs_hydrogen.o
... compile build/spiffs_cache.o
... compile build/spiffs_check.o
... linking

C:\Userdata\spiffy_xlfe>
```

alonewolfx2, 2015-09-29:
@hreintke i can build both
```
C:\Users\Suskun\Desktop\spiffyxlfe>mkdir build

C:\Users\Suskun\Desktop\spiffyxlfe>make
... depend build/spiffs_check.d
... depend build/spiffs_cache.d
... depend build/spiffs_hydrogen.d
... depend build/spiffs_gc.d
... depend build/spiffs_nucleus.d
... depend build/main.d
... compile build/main.o
... compile build/spiffs_nucleus.o
... compile build/spiffs_gc.o
... compile build/spiffs_hydrogen.o
src/spiffs_hydrogen.c: In function `SPIFFS_mount':
src/spiffs_hydrogen.c:69: warning: cast from pointer to integer of different siz
e
src/spiffs_hydrogen.c:81: warning: cast from pointer to integer of different siz
e
... compile build/spiffs_cache.o
... compile build/spiffs_check.o
... linking

C:\Users\Suskun\Desktop\spiffyxlfe>cd ..

C:\Users\Suskun\Desktop>cd spiffy-master

C:\Users\Suskun\Desktop\spiffy-master>make
... linking

C:\Users\Suskun\Desktop\spiffy-master>make clean
... removing build files in build

C:\Users\Suskun\Desktop\spiffy-master>make
... depend build/spiffs_check.d
... depend build/spiffs_cache.d
... depend build/spiffs_hydrogen.d
... depend build/spiffs_gc.d
... depend build/spiffs_nucleus.d
... depend build/main.d
... compile build/main.o
... compile build/spiffs_nucleus.o
... compile build/spiffs_gc.o
... compile build/spiffs_hydrogen.o
... compile build/spiffs_cache.o
... compile build/spiffs_check.o
... linking

C:\Users\Suskun\Desktop\spiffy-master>
```

hreintke, 2015-09-29:
with the gcc from mingw\bin as the first in the path ?

alonewolfx2, 2015-09-29:
msys first

hreintke, 2015-09-29:
try the other way around

tprochazka, 2015-11-03:
From the version which has spiffy as make target, I have no problem.
It works great now.

closed 2015-11-09 by raburton