SmingHub/Sming #229: SPIFF format fails with 4M Flash (ESP-12)

gernoteger, 2015-08-23:
With my ESP-12 and Sming 1.2.0 (installed via chocolatey)  the ESP rebooted during first startup.
The reason: during spiffs_format_internal() the watchdog produced a wdt reset.
Workaround: I reset the watchdog timer during initial format with:
```    
pp_soft_wdt_stop();
pp_soft_wdt_restart();
```

inside the init loop.

kireevco, 2015-08-23:
@gernoteger I'm sorry, I didn't quite get, what is the question?

gernoteger, 2015-08-24:
This question + workaround in one place, sorry for not beeing clear.

gernoteger, 2015-08-24:
Used Versions:
Espressif sdk: esp_iot_sdk_v1.2.0_15_07_03
Sming: 1.2.0
Tested withBasic_Blink example

alonewolfx2, 2015-09-28:
@gernoteger i am using SDK 1.3 with lastest sming and its working fine on esp12 4MB version. can you try to update both sdk snd sming manually?

pvvx, 2015-10-06:
Sming + 4Mbytes Flash + SDK 1.4.0 = rebooted (WDT) during first startup.
http://esp8266.ru/forum/threads/sming-open-source-framework-dlja-nativnoj-razrabotki-proshivok-esp8266.167/page-11#post-10945
sming_patch.zip https://yadi.sk/d/YaLBiet9jYuZU

hreintke, 2015-10-30:
I expect this is fixed by the Sming 1.4.0 Release where the integratio of spiffs/spiffy was restructured.
Please retest and if there are still problems submit a new issue.

closed 2015-10-30 by hreintke