SmingHub/Sming #230: Community proposal: Git flow for Sming development

kireevco, 2015-08-23:
I propose switching Sming framework to git flow. This will highly improve stability for the release, will allow us implement backwards-incompatible changes (like upgrading SDK) easily. Currently we can't do that at all without breaking Sming and causing issues for users.

__master__: Branch that contains latest production (stable) release
__develop__: Main development branch: contains all current new features 
__feature/websocket-client__. Gets deleted when merged to __develop__: Branch for a new feature while in development process.
__release/1.2.5__: Special release branch that exists only when develop gets merged to master. This branch is used when a feature-set in __develop__ is frozen, and the branch is used to send latest fixes, before it gets merged into master and published to [sming/releases](https://github.com/anakod/Sming/releases).

This will mean that __all contributors__ will have to submit a PR to a feature branch, it will be tested and then merged to __develop__ for automated integration testing (via TravisCI, Jenkins or SemaphoreCI), as well as manual testing on a real device. 
__Proposed flow__:
- Fork _Sming_ repo
- Create a branch off the master: __feature/websocket-client__
- Build, test your code
- Commit changes
- Push your changes to your fork on github
- Submit PR to the main Sming repo, __feature/websocket-client__ branch.

This is the most common approach for a git-flow:
http://nvie.com/posts/a-successful-git-branching-model/

Everyone that cares, let's chat :)

hreintke, 2015-08-24:
agree

ADiea, 2015-08-24:
looks to me that this will involve more work for admins(ones with write rights) to get our PRs merged into master.

do we have enough resources now to handle the extra project management work?

piperpilot, 2015-08-24:
+1  something like this is the only way to have a stable Master or release branches while allowing work to be tested.  At a minimum we should implement a development branch where any changes get merged into for testing...that way any new users that come along don't get unstable code.  The usage of feature branches and pull requests allows an individual developer to continue to develop a feature self contained and then when its ready to be merged over to master, the whole branch can be rebased and merged in.

hreintke, 2015-08-26:
Related to this. We now have three communication paths. Chat, Github and Forum.
In all of those we have questions, issues, bugs,... And searching on earlier info is difficult.
Maybe we should get as much as possible to a differentiation of usage.
- Chat : Used for quick information exchange, questions are expected to be answered/solved within a day. History deleted after one week. Mostly used by frequent users/developers.
- Forum : Used for Information exchange on larger subjects, enhancement request and possible bugs.
Also entry point for new users. History is kept. Well searchable.Maybe we need some subforums.
- Github : Used for registration and follow up on Identified/Verified bugs and enhancement requests. Also for Push Requests.

kireevco, 2015-08-26:
@ADiea amount of administrative work is not more than it is right now. Feature pull requests will be merged same way, releases will be more clear. Which part do you think will have extra project-management?

anakod, 2015-08-27:
I agre about switching to gitflow. I'm afraid what it may require some additional resources. But I'm sure what Sming is really a large and serious project now, so it's requried and very important for all of us.

ADiea, 2015-08-28:
super, then it's +1 from me also :)

hreintke, 2015-09-06:
@kireevco :
Also mentioned in gitter but here to keep visbility

The new procedure that you are proposing has the following drawbacks :

- one cannot start a PR without having another (now you or anakod ?) creating a feature branch
- after that another has to merge that into the feature branch
- when merged the PR is closed
- that means : It is not visible anymore as Pull request _ one has to look at which current feature branches are active

and as such there is

- No "standard"way to comment on a PR
- No-one (even the Original creator of the patch) change update the patch according to the comments
- So if you want to update the PR you have to start the whole procedure again with the same drawbacks 

And a remark
- Should the feature branches be based on develop instead of master ?

flexiti, 2015-09-06:
Who will test future brunches and when they will be merged to master?

kireevco, 2015-09-06:
@flexiti it's your responsibility to ask people to test/code review features. So work with @alonewolfx2  @robotiko @hreintke @hrsavla and other contributors - I'm sure they would love to make a code review / test for your PR.
(Before it was just @anakod who's time is very very limited).

The idea is to involve community much more, so changes will go out quicker.
Later we will add basic code syntax, climate and other testing automatically.

kireevco, 2015-09-06:
@anakod please don't merge anything other than a release branch to master:
https://github.com/anakod/Sming/commit/3be5c6ef3670d790fbe4cd279a81a8750be58bb6
https://github.com/anakod/Sming/commit/28d60583573ca6d206d78e4c71d01f1b10dfce59

flexiti, 2015-09-06:
I think it will be in this way: changes now will be tested only by 1 or 2 person ,  and after ??? month ? 2? will be merged - we don't have testers,  so the only efekts we will have , it is delayed comits in master branch :) but it is only my opinion

kireevco, 2015-09-06:
@flexiti I don't think you're right let's chat tomorrow in gitter?

flexiti, 2015-09-06:
Ok

kireevco, 2015-09-06:
@hreintke replying here:
(Certain things adjusted)

> Also mentioned in gitter but here to keep visbility
The new procedure that you are proposing has the following drawbacks :
one cannot start a PR without having another (now you or anakod ?) creating a feature branch

- if you are new commiter - submit a PR to develop, then a contributor (like you) will create a feature branch (if required).
- if you are a contributor - you should have write rights, you can create a branch 
after that another has to merge that into the feature branch

> when merged the PR is closed. that means : It is not visible anymore as Pull request One has to look at which current feature

Not true, if you have access to the repository, you push your chages to a feature branch directly and PR is created from the feature branch against develop.

> No "standard"way to comment on a PR. No-one (even the Original creator of the patch) can change or update the patch according to the comments. So if you want to update the PR you have to create a PR for the original PR and start the whole procedure again with the same drawbacks

Not true, when your branch is ready to be merged to _develop_ you create a PR, and you can update your changes in the branch.

> And a remark
Should the feature branches be based on develop instead of master ?

Of course, that is a mistake. Thank you.

flexiti, 2015-09-06:
My voice in the discussion
This is not a commercial project (so far, although I wish you and @anakod to be), 
we have not also a test department .
I suggest start from small steps forward:  
eg.a solution like this:
- New tag every  "feature code"  with the description of which was added
- "future branch" do not exist, ( if they exist their tests will be symbolic)
- persons authorized to decide whether to merge,create a tag and add a new feature or ask the creator of correction ( also can modify source before merge - to maintain consistency code)
- all publish in the main branch, persons authorized to decide (reject, accept, do tag)
- the persons authorized (all together) decide which version (tag) will be exposed as the most tested.All tags above will be considered as "beta" :)
- if a merge was withheld because of concerns, persons authorized  need to ask in the forum about the assessment and testing. In this situation should be temporarily created a new branch  by authorized persons.However, this should be a rare situation.Such a branch should not be held for longer than 3-4 weeks, otherwise we drown in the branches.

I know that this is not professional, but please remember that many people are doing this for fun, entertainment.
If people are reported to change and these changes will "vanish" somewhere, they quickly discouraged. The more branches, the less will be tested.
The advantage of this project, despite the fact that is very often  improvement, is very well tested ... because so far we all are testing one version.

kireevco, 2015-09-07:
@flexiti 
First of all I appreciate your comment here. Discussion with specific arguments is the only way to find right path!

> This is not a commercial project (so far, although I wish you and @anakod to be)
This is LGPL, it will be open forever :)

> We have not also a test department.

That's right, but we don't need it. You are responsible for your code quality. Branches help you not break other's code. Contributors are friendly enough to review your code and test it, just ask :)
This will make the whole process much more collaborative, when we all will know what's going on in the project. 

> New tag every "feature code" with the description of which was added

Please explain more? Examples?

> "future branch" do not exist, ( if they exist their tests will be symbolic)

What is a future branch? master?

> Persons authorized to decide whether to merge,create a tag and add a new feature or ask the creator of correction ( also can modify source before merge - to maintain consistency code)

Agree. Does make sense to me.

> All publish in the main branch, persons authorized to decide (reject, accept, do tag)

It won't work with large changes that are not backwards - compatible. 
For example, rboot. Are you ready to cut-off current stable version in favor of a new boot loader without knowing it is working and good for all of us? How do you know at the point of merging to one main branch if this kind of change is good? - That's the problem. It causes fear to merge (reasonable fear most of the times). This is what slowing big things down. Branching into a separate feature isolates some issues and speeds up acceptance.

> The persons authorized (all together) decide which version (tag) will be exposed as the most tested. All tags above will be considered as "beta" :)

Git flow works this way: when development branch is good _(=The persons authorized (all together) decide which version (tag) will be exposed)_, we cut a release, which becomes a release branch. We test this release thoroughly, submit final fixes to this branch and then merge back to master and at the same time produce a zip file/github release. Release Branch will look like __release/1.4__ and will be deleted after we merge it back to master / create zip file. 
With your approach there is no way to do code-freeze and submit final fixes. With release branch we can still continue working on new features (in _features_ or  _develop_ branches). Please let me know if that makes sense.

> If a merge was withheld because of concerns, persons authorized need to ask in the forum about the assessment and testing. In this situation should be temporarily created a new branch by authorized [...]

Feature branch, in general, should be merged into _develop_ within a week or less, this way we will keep everyone up-to-date. And again, if this is not an actual feature, we can PR agains development.
Other Contributors and Committers will work off development, and will get your feature only when you are ready to merge it. With your scenario there is a high chance that we will break the build very often.
Think about features, as about something that will be listed in release notes. Clearly, not every PR is a feature.

> I know that this is not professional, but please remember that many people are doing this for fun, entertainment.

I completely agree. So do I - use sming for fun, but I'm tired of lack of clarity - no documentation, it takes forever to find a solution to a problem that someone already had, or maybe not, or just simply was forgotten. I want all potential bugs to be isolated in their feature branches from the beginning, if we can do that. Remember, it is still ok PR against _develop_ if this is not a feature really.

> If people are reported to change and these changes will "vanish" somewhere, they quickly discouraged. The more branches, the less will be tested.

This is really sad to hear. They didn't vanish, they just went to a branch. Also, you can always see your PRs in the history. Changes never vanish.

> The advantage of this project, despite the fact that is very often improvement, is very well tested ... because so far we all are testing one version.

I will disagree with you here. It is not very well tested. There are bugs, and I'm not certain if I want to use Sming + Esp8266 as a platform for Enterprise area (simply because it is not clear sometimes which issues come from Espressif, which are from Sming Core, or my "a little bit better then terrible" implementation. Example: #154 @hreintke seems to figure out the issue, but no one else was able to test his suggestion. Why?

@flexiti I see all your points, but they don't really solve the our issues the way they should.
What do you think about switching to git-flow gradually, when feature branches will be optional, while _develop_/_master_/_release_ mandatory?

hreintke, 2015-09-07:
@All : 
There are two item I fully agree with : 
- We should get a better/faster/clearer  process/procedure for getting bugfixes/features in sming
- It should be done gradulary. There is no reason to go for the perfect solution in one shot and as such we are nor looking for a solution that "solve the our issues the way they should"

As said earlier, I am not experienced in github and the details of working with it so if have have a wrong view of usage just let me know. I am a "happy learner". 
I also looked at our companion project esp8266/arduino. What I see there is only one branch (called esp8266 and on first sight looks like our master) and several releases. Is that the same as @flexiti is proposing with tags ?
The statement from @flexiti regarding limiting number of branches sounds familiar to remarks I heard when managing projects.
Then, remember we are not a multi million project with dozens of contributors and numerous bugs and new features. On the contrary. If, by the use of a simple, not perfect solution, we can attract more people to actaully work on sming it'is beneficial to all.

Conclusion from my side : Take the proposal from @flexiti with maybe some small enhancements and start with it.

@kireevco : 
On follow-up on issues/PR's/documentation. With only a little more effort we can make much more use of communication paths. Then we will keep a better overview of questions/issues/features. See my proposal from (now) 12 days ago.
On actual followup from my side. I already mentioned that I am wiling to document, but am not the person to lay down the structure of it. If someone does that, I will produce input.

PS : on #154. I asked @kireevco to test and feedback on my work in progress but got no response :smile:

kireevco, 2015-09-07:
> As said earlier, I am not experienced in github and the details of working with it so if have have a wrong view of usage just let me know. I am a "happy learner". 

I know that you are open to new things, and so far, no issues on view of usage :)

> I also looked at our companion project esp8266/arduino. What I see there is only one branch (called esp8266 and on first sight looks like our master) and several releases. Is that the same as @flexiti is proposing with tags ?

_esp8266/arduino_ is not a perfect example. Look at their number of issues, and it's growing. It is just more popular because it is sort of "official". One branch is a dead-end IMHO.

>The statement from @flexiti regarding limiting number of branches sounds familiar to remarks I heard when managing projects.

There are multiple points of view, and these remarks sometimes come from laziness and not seeing a global scope of the whole thing. In my experience, when people stepped into branching (at least with one foot) - they never go back. Because this is much more efficient and structured. And not hard at all.

> Then, remember we are not a multi million project with dozens of contributors and numerous bugs and new features. On the contrary. If, by the use of a simple, not perfect solution, we can attract more people to actaully work on sming it'is beneficial to all.

I agree on that one, that's why I'm not demanding 100% unit test coverage, implementing QA environment (Hardware + Software). I am also OK with submitting PRs to _develop_ branch, if it is minor. On the other hand, I want big features like rboot to able to be tested quickly without disrupting current users. Let's find a comfortable balance - that's it.

> Conclusion from my side : Take the proposal from @flexiti with maybe some small enhancements and start with it.

So, what stops you from submitting small PRs to _develop_ now? And all releases and everything else will be taken care by me or someone else who wants to help :)

> On follow-up on issues/PR's/documentation. With only a little more effort we can make much more use of communication paths. Then we will keep a better overview of questions/issues/features. See my proposal from (now) 12 days ago.

Sorry, I didn't respond to that comment, although it is important. I see github as the ultimate source for large discussions (like that) on the development. Chat is for quick talks and forum - more as a front-face for new users, where all actual issues should be mirrored to github, and all user-related questions answered right there, on the forum.

> On actual followup from my side. I already mentioned that I am wiling to document, but am not the person to lay down the structure of it. If someone does that, I will produce input.

> PS : on #154. I asked @kireevco to test and feedback on my work in progress but got no response :smile:

I simply didn't look. Partially communication issue (yes), this is a simple patch, could've been applied myself. But also, that could've been a branch with this patch applied, that anyone can test, not just me. And then when this branch is 'good' from multiple people, it simply gets merged. No double-work.

hreintke, 2015-09-12:
LS,
First of all : I do agree to get an improved/quicker/more confident process to get updates into sming.
However I do see in the current proposed workflow as being complex and having a number of drawbacks.

I have been away for a couple of days and reading what happened on PR's and discussions.
I especially noticed a few remarks in the conversations :
- Need to create PR on PR because the original PR is closed and the committer cannot update/change
- probably because of the above, testers where unclear which version to use (or were available)
- even that much that was suggested to used Richards github version, a version we don't know status and sming version of -> so what was really tested ?
- unclearity on how to inplement, what and how should go where in sming (makefile, core,service,..)
- someone mentioning "too many branches to keep overview"

In my opinion we should structure the discussion on PR's, merging and releases.
I propose to open an issue for a number of possibilities at github with the text below.

```
@ ALL
We are reviewing the process of Pull Requests for integration of Bugfixes and New features within Sming.
Main issue with several discussions/opinions is #230

-----------------------------------

Proposal description

-----------------------------------
Please review this proposed solution and reply with (multiple possible) :  
+1 : This aspect I find very positive on the proposal, it would even get better with "the following improvements"
-1 : This aspect is very negative. If we change "this" to it, it could overcome my concerns
?? : I don't understand this part of the solution, can you provide some more information.

Remember : 
- Do not only reply with the +1/-1/?? but also with an explanation following
- Keep the discussion in this issue solely on this subject, other solutions should get discussed in a separate issue.
- Forget "personal preferences" comment just on this solution.

We will collect opinions/questions until and including monday september 21.
After that, next steps will be defined and executed depending on the outcome.
```
Two options have been proposed until now 
1/ Kireevco proposal as stated at the start of this issue
2/ flexity proposal using tags, added to this issue around sept 6.

Anyone having alternative proposals, please reply to this issue (#230) and give a short description.

kireevco, 2015-09-12:
@hreintke welcome back, first of all,

> Need to create PR on PR because the original PR is closed and the committer cannot update/change
What is still missing here from @anakod is write permissions to the repo for Contributors. Either @flexiti should have them or submit a PR to _develop_

> Probably because of the above, testers where unclear which version to use (or were available)
It is either  _feature/feature_name_ or _develop_ if no feature branch is present. Disadvantage of having PR to _develop_ is that you can't really test it until merging it

> even that much that was suggested to used Richards github version, a version we don't know status and sming version of -> so what was really tested ?

I personally see that Richards, @alonewolfx2 and others who test don't have any issues with testing feature/rboot right now.

> Unclearity on how to inplement, what and how should go where in sming (makefile, core,service,..)
Please elaborate how this is related to branches and git-flow?

> Someone mentioning "too many branches to keep overview"
This is just personal opinion of someone. This is normal practice, and just get used to it. A lot of projects have much more branches, while we won't have much, maybe 5 at a time (the amount of features worked on.

I've read all your (@flexiti) arguments, and I don't really see them valid so far. Also, I'm not sure why @flexiti is not taking a part in this discussion here.

As an addition, other uses don't have issues submitting to _develop_ (https://github.com/anakod/Sming/pull/276 https://github.com/anakod/Sming/pull/277 https://github.com/anakod/Sming/pull/272 https://github.com/anakod/Sming/pull/264 for instance) - it's just you who still remains grumpy :)

hreintke, 2015-09-13:
LS,
Just noticed another drawback of the proposed way of working.
I wanted to submit a remark on the websocket client PR.
But.. the websocket client PR is closed because it is merged into the features/websocket-client branch.
But.. I cannot comment on a branch.

Only option to comment is open a new "generic issue" 
This makes that there is no overall overview in current/past comments on a proposed PR

hreintke, 2015-09-13:
@kireevco : 

> I personally see that Richards, @alonewolfx2 and others who test don't have any issues with testing 
> feature/rboot right now.

If you read the comments/questions during the process you notice the trouble.
Why do they not comment on the procedure ? -> They are fully occupied by testing/updating and focusing on that

> I've read all your (@flexiti) arguments, and I don't really see them valid so far. 
> Also, I'm not sure why @flexiti is not taking a part in this discussion here.

Nice to hear that you think they are not valid. What is the opinion of others ?
That is exactly what I am trying to achieve.

> As an addition, other uses don't have issues submitting to develop (#276 #277 #272 #264 for instance)
> it's just you who still remains grumpy :)

I think they just are using a small part of the proposal and are not into the details complete procedure. (Yes, I know it is an assumption).
And maybe...  using "just a development branch"is a good idea.
We should get into this questioning instead of keeping fixed to one proposal.
BTW : I am not grumpy. I try to get a overall view of the proposals, send my remarks
 and believe in a decision by consensus instead of a forced one man decision.

> PS. I'm not sure how this work, but you keep moving 'the date'

As you have seen, I had to send the message twice to get a response. I updated the date to use a date which is reasonable. Otherwise I would get the remark : "how can this be done in two days"

hreintke, 2015-09-15:
LS,
Seeing the discussion above, the way the rboot PR was handled and showing that I am prepared to think about a better way of getting Sming stable. 
This is an alternate proposal for improved workflow.

Branches/usage in "global sming repo"

- Develop
All new features will PR to this branch
PR's stay open until finished/ready for release
Merge into develop means will be included in next release

- For each release there will be a "Release branch"
At regular intervals develop is merged into "Current release branch" creating "Next release branch"
This finalizes the creation of release and "users" can use/pull this release as current.

- Bugfixes are PR'ed against the "Current Release Branch" 

In a real life example that would mean that that the following branches exist :

Release_1.2.0  -> old release 
Release_1.3.0  -> old release
Release_1.4.0  -> current release + added bugfixes, PR's open for unfinished bugfixes
Develop        -> Release_1.4.0 + added functionality for Release_1.5.0, PR's open for unfinished new features and bugfixes to merged functionality

This would provide a procedure with :
- Known and agreed functionality for each release
- Stable releases
- Regular releases

- Users know which functionality is available is stable in each release
- Users are free to use current or old release -> upgrade in their projects timing
- Users are free to use develop for reasonable stable new functionality

- Developers without write access can initiate work/update/finish PR's without intervention from "main repo committers"
- Developers have clear and complete overview of discussions/issues on PR's
- Developers have clear overview of open "functionality in progress" and "Not yet complete bugfixes"

In my opinion the above creates a process which has the benefits of stability and predictabilty, without being overly complex.

Please comment !

ADiea, 2015-09-15:
I also feel this is a more clear way of doing things.

It is true that I have not contributed lately due to focus on other areas of my hobby project but when I come back SMINGing I'd love to find things as straightforward as before :)

So from my side +1 for @hreintke s proposal as it seems more clean and straightforward

flexiti, 2015-09-15:
@kireevco write:
"I've read all your (@flexiti) arguments, and I don't really see them valid so far. Also, I'm not sure why @flexiti is not taking a part in this discussion here."

I wrote my opinion and that's it.
Someone must decide, otherwise we will exchange views for years :)

@kireevco  write:
"I personally see that Richards, @alonewolfx2 and others who test don't have any issues with testing feature/rboot right now"

yes it's true, but at the same time do not test other changes - the next month will deal with the next branch?. For my own purposes I made myself my own "develop" because I will not be test each branch separately - I do not have much time, have You??
I am sure that today there is only rboot branch is tested by several people (3?), and the master branch by people who take sming. (btw. develop today is almost a master) 
Other changes lie somewhere

In brief:
  - Too much barnch (acc. To me today ( according to what I wrote earlier)  should be 'master' as this stable, develop as a development and only rboot as it essentially affects the entire project.
No matter if it will be called  a tag or branch, it's important not to have this much.
Rapid tests additional branch (in this case rboot), and immediately transfer to develop.
All of which they use Sming take master, those who  test Sming  take the  develop.

If we take into account that, @kireevco metod can be adapted to this and @hreintke also.

piperpilot, 2015-09-15:
The only comment I would like to add is a decision needs to be made and move forward with more than one person being able to commit changes into develop and/or master.  rboot has been in works for weeks now and is all disjointed because only one person can accept the PR.  The feature branch is broken, but there is a PR for a fix specifically for that... because of the current process, there are multiple PRs and you can't really tell where things are.  Its just frustrating when you update and then spend hours realizing that there are outstanding PRs that haven't been pulled into the feature branch in a timely manner.

kireevco, 2015-09-15:
So, from what I see here, what is the main issue? Need to create a separate branch? As I mentioned before, let's skip this, unless we all feel like it is required for some feature.

We will convert a PR to a feature branch, either by asking a committer to re-submit it or merging. 

@hreintke I don't see any essential difference, other than deemphasizing feature branches and keeping old releases. It looks simple, but in reality it is essentially the same. What are main differences you see that "change the game"?

>For each release there will be a "Release branch"
At regular intervals develop is merged into "Current release branch" creating "Next release branch"
This finalizes the creation of release and "users" can use/pull this release as current.

@hreintke Can you please describe the whole flow you see it, starting from _develop_? 

I agree that we should minimize PRs to one, which is already happening when submitting to develop.

flexiti, 2015-09-16:
@hreintke 
if I suggest do it now:
- Develop, is ok
- The master can be,
- Remove any branch except rboot and their contents move to develop
- Rename future/rboot to test/rboot

(the name "future" will tempt occasional visitors to download and see how it work, because it does not ,they can discourage to the use sming)

From time to time (of course, if the version is stable) you will be moved to develop the master and gave another tag.

This is important because a lot of people come here sometimes, rarely
So soon they will be clear for them:
- Develop, you know, the development version: there may be errors
- Master (better Stable) version:  this version people will download  frequently
 (You can also select older versions by the tag's)
- test/rboot, a test, only the persons concerned and informed will know what it is.
(It's a temporary branch)

Since then, All (contrary to appearances, we are not too many :)) we will test develop and assist in moving the test/rboot to develop.

After this You tell us who will decide whether to reject the change, add to develop or create test branch (I hope this will be rare). This should be done a maximum of 1-2 days.

hreintke, 2015-09-16:
@kireevco : @flexiti : 
I do not have the possibility to reply in detail today but will do tomorrow (Thursday CET).

robotiko, 2015-09-16:
Hi,
joining late to this party, I won't focus on the procedure but wanted to share some experience with the feature/rboot and an opinionated generalization from a high level perspective that might help and probably will get blamed for:   :D  

1) Whatever method is chose, should be agreed and should be some transition time (as any change management  project will tell you), and followed during adoption. 
The way the procedural change was managed was far from ideal. 
This issue got opened, some discussion, time goes by and suddenly there is no other way to work and everybody is requested to change. (Including some PR that were waiting for more than 2 month for PR).
The effect was kind of chaotic: instead of a transition time and accept last PR as previous method, all contributors asked how to do it  and had to learn the new procedure and reissue PR (extra work).
A lot of noise and main problems started to pop up: no clear view of PR and issues for newcomers since most people is not used to browse branches to see status. 
As @hreintke pointed, not seeing contributions here doesn't mean that everyone agrees and everything is running smoothly.   Sming is receiving most interaction from esp8266.com forum and Gitter in a different scale github issues.   
Since gitter is quite noisy, many contributors are no longer following it (this includes most of big wolves here :D) but is where most questions and issues pop up after the change and was not used to measure the impact of change.

2) Should always be realistic about resources, efforts and goals: 
A procedure is just that. If resources are low, it should not be an overload. 
Community is growing  with current  lack of documentation and step learning curve, the number of contributors is quite reduced (normal) but they highly compromised. Should take advantage of it. 

Right now after "having fun" and being involved with the /feature/rBoot integration/ testing I can tell that the biggest issue is the lack of real community and collaboration.   
There is no roadmap, and clear goals with polls, etc but the most relevant thing is the merge BOTTLENECK. This is really dangerous. 

3) Goals: Resource optimization and stability with agile approach.
If the git workflows was introduced with this goal, the result was totally the opposite. 
Being so agile one side (contribution) and nothing at the other (merging), it became overkilling task  and totally useless effort.  
Example: rBoot: quite relevant development, with many people willing to give it a try and we were lucky enough to have an heroic @raburton in fast mode deeply involved in getting rBoot into Sming.  

Result: Fast code iterations for stability with no repo control and  the new flow ended in a total repo mess (I do support 200%  @piperpilot comment)  PR over PR .. not being merged, people reporting several times the same fixed issues because  part of the BGFX PR where merged and some other were pending as PR, and not very clear vision in the branch way and how they relate with issues.
I had a "treasure map" with branch, issues and PR and merges.. but methodology/ tool should prevent from this flood.  As proof, today after main rboot merge, many issues were starting to close reflecting that were due to using the wrong branch, not all the PR, etc  
Have fun recovering  gitter conversations from pasts days..  people checking out @raburton repo, some giving up.. some sharing how to cherry pick the different PR to make something stable, and even some people sharing to integrate different branches on top of the PR merges.. to be able to get a couple functionalities. 
Stability and Agile.. maybe :D  we ended testing part of changes with patches to not get crazier. 
Next step in this procedure : send code zipped by email.  (sarcastic).

I think all this is much worse and scarier for newcomers  than previous method.
Not blaming anyone, just exposing. 

4) Empiric conclusion:
-Sming community should be empowered.  A week gap to merge BGFX PR is way too much and generated too much trouble.  We are all too  busy so there should be more people with merge rights. 
Contributors usually send fixes very fast once detected, but if times passes by this gets harder so should have fast reaction there. 
At least in the last couple of months, most Sming improvements (if not all), came from community contributors so don't see why they should not have more rights here (IMHO). 
-Not having a clear global picture of development status (all PR waiting to be merged, all issues and where they belong to, in the same place)  is bad to control the project but also to show how active it is and potentially increase the community. 
-The past experience showed that most people is not used to the git flow adopted, and this change was justified to improve their experience. 

I'm with @hreintke and @flexiti in simplifying flows. 

In most projects head is not stable.. is development, if you want a stable one, get a release (as it was before).  If want to add development one and tag and merge PRs for a release.. it is ok  BUT we need more people with merge rights.  Cannot have such delays. 
This is not a big project with tens of developers full time in different functionalities and advanced version planing deciding what to integrate in next version and what to keep for later. 

Keep is simple!  :dancer:

hreintke, 2015-09-17:
@kireevco @flexity @robotiko @others

As promised a new description of my proposal for the workflow of sming.

I will use a "real life example" to avoid generalized naming and terminology.
Current version of Sming is 1.2, example PR is websocket_client
No "history branches" present.

-------------------------------------------
In this situation there will be the following branches in Sming Main Repo :
- Release_1.2 -> Stable release 
- Develop     -> After release identical to release, will be used for development of next release.

Process for new functionality : 
- @hrsavla writes a websocket_client in private environment
- When appropriate (he is finished/want to discuss further progress with others) he initiates a PR against Develop Branch (PR numbered 111)
- This PR is not merged into develop but stays open for further progress.
- @others can test/use this by : clone main_repo_branch develop, fetch PR-111 (git fetch origin pull/111/head), compile and test.
- @others will comment, find issues, propose updates to the PR, all will be communicated within the open PR-111 at Github
- @hrsavla adapts the websocket_client code according to the remarks
- @hrsavla pushes the additional commits to the outstanding (still open) PR-111
- As soon as these commits are pushed, they are visible and usable by all (no action required by "main_repo_Admins/committers)
- @others can access/test/comment on the updated PR as mentioned above.
- When websocket_client is tested, complete (ready), "Main_Repo_Admin/committer" merges PR-111 into develop. PR-111 is closed.

At this moment there will be the following branches in Sming main repo :
- Release_1.2 -> Stable release
- Develop     -> Release_1.2 + added websocket_client (which is (close to) production ready)

Of course as multiple PR's can be in progress/finished in parallel so Develop will hold all finished functionality/PR's 

When decision to go to a new release.
- Add branch Release_1.3, based on Release_1.2
- Merge Develop into Release_1.3
- Developers rebase PR's on Release_1.3

At this moment there will be the following in Sming Main Repo
- Branch Release_1.2 -> old stable release
- Branch Release_1.3 -> new (latest) stable release
- Branch Develop     -> Bases on Release_1.3 .) ready for new functionality as above
- Open PR's for unfinished new functionality based on Release_1.3

-------------------------------------------------------
If we additionally agree that : 
- When development on a PR is finished (ready to merge) that will be done within a week.
- There will be regular Sming releases (once a month ?)

There will be the an environment which has : 

Generic : 
- Known and agreed functionality for each release
- Stable releases
- Regular releases
- A Develop release which is close to stable and can be used/tested by users with less detailed (internal) knowledge of included changes.
- The open PR's provide a complete overview of new functionality in progress 

Users : 
- Know which functionality is available is stable in each release
- Are free to use current or old release -> upgrade in their projects timing
- Can use develop for reasonable stable new functionality

Developers :
- Can initiate work/update/finish PR's without intervention from "main_repo_admin/committers"
- Have clear and complete overview of discussions/issues on their PR's
- Have clear overview of open "functionality in progress" and "Not yet complete bugfixes"
- Have full access to complete and latest source code of new functionality without combining several PR's

Main Repo Committers :
- Will do a merge to develop only when PR is finished
- Will merge finished PR's to develop within agreed time interval
- Have no time pressure to merge all updates on open PR's
- Will do a merge to new release on regular intervals

----------------------------------------------------
 
Generic remarks :
- The above is a procedure for most of the new functionality. 
- For extra large/complex new functionality a "one time" specific process can be created, probably with use of special additional branches.
- Although less "time pressure" on PR updates, there is still the need for more "Main_Repo_Admin/Committers"
- The steps above are compatible with Travis-CI which, when setup properly, will test open PR after each update. 
- The above is close to the process @flexity proposed (use of Branch instead of Tag). I stayed at branches as (I think) for the less experienced (github) user it is more common to download branches.
- We should make sure that a "Release Branch" has the correct libsming library included. It should be usable without users compiling sming core.
- I am thinking on an extension to the above by asking "Main_Repo_Admins/Committers" to provide a quick initial thought on the PR. That would prevent activity/work on PR's which concern/are against Sming philosophy.

And as always : Please comment, I will answer and explain any unclearity 

Happy reading

kireevco, 2015-09-17:
Hi,

Im' on a long vacation and I don't really have time and good internet.
@hreintke, everything looks good to me. Feel free to take a lead and
implement it!

avr39-ripe, 2015-09-17:
@hreintke Nice Ideas! Me personally like them. Main thing new to me and important base of whole your process is ability to fetch pull request (git fetch origin pull/111/head) so no extra branches on main repo, and consistent workflow of PR till it ready to be merged. +1 from me.

hreintke, 2015-09-17:
@kireevco : 
Thanks for the positive feedback.
But "Feel free to take a lead and implement it!" -> Is not possible without "Main_Repo_Admin/Commiter"

@kireevco : @anakod : 
As you know, the both of you are the only one have privileges on the Main Sming repo.
Kireevco on a "long holiday" and Anakod "busy with (too ?) little time to be really active on sming.
What is the idea/progress in the coming time on 
- Managing PR's ? 
- Managing bugfixes ?
- Implementing rBoot, a major implementation of new functionality
- Merging to master/other branches ?

flexiti, 2015-09-17:
@hreintke 

"When websocket_client is tested, complete (ready), "Main_Repo_Admin/committer" merges PR-111 into develop. PR-111 is closed."

it must be clarified, 
how and when  to report a readiness to merge  eg. a situation that nobody test for two or more  weeks because eg. no one has such a Flash, sensor, LCD or ....time :)

otherwise we will have again a lot of "branch" which now will be called the PR-111, etc.

raburton, 2015-09-17:
@hreintke - sounds very reasonable.

I didn't realise it was possible to checkout a pull request, do many people know that? I like the idea of having a single PR for the feature (instead of a branch) which is accessible via the main repo yet updatable by the developer automatically from their fork.

I was starting to think that the best way was just to go back to the old fashioned method of doing it all in your own fork and only PRing once complete, but the problem with that is when you are doing something non-trivial (rBoot integration has turned out to be a lot more complicated than expected - just the integration mind you, rBoot itself I'm pleased to say has been very well behaved) you need lots of testers and they aren't all that keen on switching to stuff running out of forks. Your way simply allows them to switch to the PR branch in the main repo, like the the current branch system, but giving the developer control over what's actually in there.

piperpilot, 2015-09-17:
Wow, I didn't know it was possible to check out a PR either.  I believe if you update the branch that the PR is created against, it will all be included when it is merged in.  I think this is a great solution.

+1 for giving it a try!  Soon!

hreintke, 2015-09-17:
@raburton : @piperpilot 
I am not that experienced in github so started searching for a way to do what I wanted instead of making use of the (small/partial) knowledge I had.
Found comments like these : 
```
Fortunately, GH creates a remote ref for each PR, so you can use the following syntax to create a local branch that reflects the PR :

git fetch origin refs/pull/PR_NUMBER/head:LOCAL_BRANCH

Elegant and convenient!

For example:

git fetch origin refs/pull/611/head:pull_611
git checkout pull_611

Change /head to /merge to get the ref to the pull request merged with its intended target branch.  
```
And on the github site, hidden in a help on working on stale pull request.
https://help.github.com/articles/checking-out-pull-requests-locally/

If you, as experienced github users, can do a quick check on this option and feedback your results would be great.

hreintke, 2015-09-17:
@flexiti : 
Yes, I agree that we need a way to decide on readiness/completeness of PR's
- Who decides
- Based on what criteria
- In what timeframe

I started a (little) with that in my generic remarks.
> I am thinking on an extension to the above by asking "Main_Repo_Admins/Committers" to provide a quick initial thought on the PR. That would prevent activity/work on PR's which concern/are against Sming philosophy

But definitely needs to be extended.
Probably/Maybe with a different options for "Sming Core" vs "Supplied libraries" as there is more or less dependency on specific hardware.

My thoughts on this however were : This is an open point in every Workfow we design. Let's finalize that discussion first before getting into the next discussion

kireevco, 2015-09-17:
@hreintke <https://github.com/hreintke> Unfortunately I don't have any
permissions to repo management. That's why I asked for migrating the repo
to a organization.

hreintke, 2015-10-02:
@kireevco @flexity @robotiko @raburton @piperpilot @others

As promised a new description of my proposal for the workflow of sming.
I adapted my previous proposal to a more standard Github flow. 

-------------------------------------------

In "normal" situation there will be the following branches in Sming Main Repo :
- Main		-> Stable release 
- Development  	-> After release identical to release, will be used for development of next release.

And a Release in Github repository called "Sming_1.2.0" 

**Process for new functionality to be added :** 
- @FunctionaltyProvider writes "New/adapted functionality" in private environment
- When appropriate (he is finished/want to discuss further progress with others) he initiates a PR against Development Branch (PR numbered 111)
- The PR should only contain 1 commit (squashed local commits)
- This PR is not merged into develop but stays open for further progress.
- @others can test/use this by : clone main_repo_branch develop, fetch PR-111 (git fetch origin pull/111/head), compile and test.
- @others will comment, find issues, propose updates to the PR, all will be communicated within the open PR-111 at Github
- @FunctionalityProvider adapts the "New/adapted functionality" code according to the remarks
- @FunctionalityProvider pushes the additional commits to the outstanding (still open) PR-111
- As soon as these commits are pushed, they are visible and usable by all (no action required by "main_repo_Admins/committers)
- @others can access/test/comment on the updated PR as mentioned above.
- When "New/adapted functionality" is tested, complete (ready), "Main_Repo_Admin/committer" merges PR-111 into branch Development. PR-111 is closed.

At that moment there will be the following branches in Sming main repo :
- Master	 -> Stable release
- Development    -> Master (= Release 1.2) + added websocket_client (which is (close to) production ready)

Of course as multiple PR's can be in progress/finished in parallel so Develop will hold all finished functionality/PR's 

When decision to go to a new release.
- Merge Development into Master
- Create release in Github called "Sming_1.3.0"
- Open PR's are rebased against "New Master"

At that moment there will be the following in Sming Main Repo
- Branch Master	 	-> Contains Release 1.3.0
- Branch Development    -> Based on Release_1.3.0,  ready for new functionality as above
- Open PR's for unfinished new functionality based on Release_1.3.0
- Two Releases in Github directory -> "Sming_1.2.0" and "Sming_1.3.0"

**Process for Bugfixes :**
There will be two types of Bugfixes
- Non Urgent fixes which can wait until next release
These will be handled using the "new functionality" process -> incorporated in next release

- Urgent fixes which have to be applied asap (hotfix).
A (squashed) PR against Master is submitted.
This PR is tested in the same way as "new functionality" only using Master as base to test against.
When testing is completed. The PR is merged in both Master and Development (eventually fixing compatiblity issues with Development)
Decision is made whether "non urgent bugfixes" will also be included -> Merged to Master also.
A Release is created in Github called "Release_1.3.1"

At that moment there will be the following in Sming Main Repo
- Branch Master	 	-> Contains Release 1.3.1
- Branch Development    -> Based on Release_1.3.1,  including finished functionality for "Sming_1.4.0"
- Open PR's for unfinished new functionality based on Release_1.3_1
- Three Releases in Github directory -> "Sming_1.2.0", "Sming_1.3.0" and "Sming_1.3.1"

**Introduction of "Large Impact" functionality**
This process which should only be used when necessary. Expection is very limited usage. 
It can/will be adapted when specific requirements are needed.

For these kind of PR's in general the following procedure is taken.
- @FeatureProvider writes "Feature functionality" in private environment
- When appropriate (he is finished/want to discuss further progress with others) he request a "Featurte Implementation"
- A new branch "Feature/Desription" is created, based on Master
- A (squased) PR is submitted against "Feature/Description"
[... 14 lines cut ...]
- When development on a PR is finished (ready to merge) that will be done within a week.
- There will be regular Sming releases (once a month ?)

----------------------------------------------------
**Next steps**
If we agree with the proceduere above the following steps should be taken.
- Create release "Sming_1.3.0" from current Master
- Create new branch "Development" based on Master
- Identify Bugfixes/functionality PR's already included into Develop
Decide on status of those PR's
Finished/accepeted ones -> Merge into "Development"
Unfinished PR's re-submitted against Development, finish following new procedure.

- Identify open "feature/functionalty" branches currently open
Decide on which process they will follow to complete
"New/adapted functionality" -> re-submit against Development
"Feature" ones -> rebase against Master, follow new procedure

- Document this process on Sming Wiki
- Create a Wiki pages which describes functionality of available releases
- Update Main Wiki page to include latest release information.

----------------------------------------

Please comment

piperpilot, 2015-10-02:
All of this sounds pretty sensible to me.  I think we need to document 2 separate things...

1.  What the process from a maintainer is
2.  A simplified version of the process for a contributor that just wants to get a simple fix/feature in

If the process/documentation is too long and complex, we won't get people to contribute fixes.  I don't think this is THAT complex...it just seems so due to the long explanation.

kireevco, 2015-10-02:
@hreintke I fully trust you!

jedahan, 2015-10-02:
In an ideal world, wouldn't releases happen every time a new feature/functionality is merged into develop, because at that point a feature is merged into develop, that code has already been vetted?

I am just a bit unsure of the reasoning of having a separate develop and main branch, as the biggest workflow change is all PRs must start in a separate feature branch / PR.

### Simplified proposal
* Fork master to a new feature branch
* Develop locally
* Make a pull request with the local commits squashed
* Maintainers and testers test the change, and give feedback until they are comfortable giving :+1:
* After **n** :+1:s, merge into master
* Release new version, using semver or ferver to reduce overhead

**Upsides**
  * easier to contribute
  * less maintenance for project maintainers
  * encourages testing of code before merging

**Downside**
  * developing against multiple unmerged pull requests is trickier

hreintke, 2015-11-29:
@kireevco : Can we close this issue ?

kireevco, 2015-11-29:
I don't know, feel free if you think so.

hreintke, 2015-11-29:
A process for contributions is agreed.
See https://github.com/SmingHub/Sming/blob/develop/CONTRIBUTING.md for process.

closed 2015-11-29 by hreintke