SmingHub/Sming #232: Makefile:20: c:\tools\sming\Sming: Permission denied

Airji3d, 2015-08-26:
Hi just installed Sming framework and trying to compile Basic_Blink but I get 2 errors and 2 warnings.
This has probably been answered by now but I couldn't find any solution that would help in my case.

echo %ESP_HOME%
c:\Espressif

echo %SMING_HOME%
c:\tools\sming\Sming

echo %PATH%
C:\Users\Rolland>echo %PATH%
C:\mingw\bin;C:\mingw\msys\1.0\bin;C:\tools\mingw32\bin;C:\tools\mingw64\msys\1.
0\bin;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Wind
ows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\ProgramData\cho
colatey\bin;C:\tools\mingw32\bin;C:\tools\mingw32\bin

dir %ESP_HOME%
C:\Users\Rolland>dir %ESP_HOME%
 Volume in drive C has no label.
 Volume Serial Number is 846E-FA05

 Directory of c:\Espressif

24/08/2015  04:49 PM    <DIR>          .
24/08/2015  04:49 PM    <DIR>          ..
09/07/2015  10:34 PM            17,319 changelog.txt
14/05/2015  10:37 PM             2,061 Command line Unofficial Development Kit f
or Espressif ESP8266.lnk
11/04/2015  11:37 PM               186 DevKit web site.url
24/08/2015  04:48 PM    <DIR>          docs
24/08/2015  04:49 PM    <DIR>          ESP8266_RTOS_SDK
24/08/2015  04:49 PM    <DIR>          ESP8266_SDK
24/08/2015  04:49 PM    <DIR>          ESP8266_SDK_094
24/08/2015  04:49 PM    <DIR>          ESP8266_SDK_095
11/04/2015  11:37 PM               180 Espressif web site.url
24/08/2015  04:49 PM    <DIR>          examples
24/08/2015  04:49 PM    <DIR>          extra
11/04/2015  11:38 PM           137,750 Install.ico
24/08/2015  04:49 PM    <SYMLINKD>     sdk [c:\Espressif\ESP8266_SDK]
24/08/2015  04:49 PM           628,693 unins000.dat
24/08/2015  04:48 PM         1,312,112 unins000.exe
24/08/2015  04:49 PM            23,001 unins000.msg
11/04/2015  11:38 PM           147,878 Uninstall.ico
24/08/2015  04:48 PM    <DIR>          utils
24/08/2015  04:48 PM    <DIR>          xtensa-lx106-elf
               9 File(s)      2,269,180 bytes
              12 Dir(s)  109,866,876,928 bytes free

dir %SMING_HOME%
C:\Users\Rolland>dir %SMING_HOME%
 Volume in drive C has no label.
 Volume Serial Number is 846E-FA05

 Directory of c:\tools\sming\Sming

24/08/2015  04:55 PM    <DIR>          .
24/08/2015  04:55 PM    <DIR>          ..
24/06/2015  11:53 PM             8,597 .cproject
24/06/2015  11:53 PM               815 .project
25/08/2015  06:25 PM    <DIR>          .settings
24/08/2015  04:55 PM    <DIR>          appinit
24/08/2015  04:55 PM    <DIR>          apptest
24/08/2015  04:55 PM    <DIR>          compiler
24/08/2015  04:55 PM    <DIR>          include
24/08/2015  04:55 PM    <DIR>          Libraries
24/06/2015  11:53 PM             7,687 License.txt
24/06/2015  11:53 PM             8,323 Makefile
24/06/2015  11:53 PM               554 Makefile-bsd.mk
24/06/2015  11:53 PM               541 Makefile-linux.mk
24/06/2015  11:53 PM               533 Makefile-macos.mk
24/06/2015  11:53 PM            13,841 Makefile-project.mk
24/06/2015  11:53 PM               540 Makefile-windows.mk
24/08/2015  04:55 PM    <DIR>          Services
24/08/2015  04:55 PM    <DIR>          SmingCore
24/08/2015  04:55 PM    <DIR>          system
24/08/2015  04:55 PM    <DIR>          Wiring
               9 File(s)         41,431 bytes
              12 Dir(s)  109,866,967,040 bytes free

C:\Users\Rolland>where make
C:\tools\mingw32\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe

Tried a make from the command line:
c:\tools\sming\Basic_Blink>make
Makefile:20: c:\tools\sming\Sming: Permission denied
Makefile:20: /Makefile-project.mk: No such file or directory
make: *** No rule to make target '/Makefile-project.mk'.  Stop.

Clues to resolve these errors would be greatly appreciated and thanks in advance.

kireevco, 2015-08-26:
I really appreciate steps you took to troubleshoot this issue. 

You have 2 make files in the path:

C:\tools\mingw32\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe

Please remove one from mingw32 abs try again. You can adjust your path variable to prioritize mingw64 as well.

Sent from my iPhone

> On Aug 25, 2015, at 6:06 PM, Rolland Jones <notifications@github.com> wrote:
> 
> C:\tools\mingw32\bin\make.exe
> C:\tools\mingw64\msys\1.0\bin\make.exe

Airji3d, 2015-08-26:
Hi Dimitry,
Thanks for the quick reply. I removed the mingw32 make and left the mingw64. It helped but not completely, now I have 1 error and 100 warnings.

c:\tools\sming\Basic_Blink>make
make: *** c:\tools\sming\Sming: Invalid request code.  Stop.

c:\tools\sming\Basic_Blink>echo %PATH%
C:\tools\mingw64\msys\1.0\bin;C:\ProgramData\Oracle\Java\javapath;C:\Windows\sys
tem32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\
v1.0\;C:\ProgramData\chocolatey\bin;c:\tools\mingw64\msys\1.0\bin

Eclipse Console output:
23:02:50 **** Build of configuration Sming for project Basic_Blink ****
make all 
make: *** c:\tools\sming\Sming: Invalid request code.  Stop.

23:02:51 Build Finished (took 1s.700ms)

Eclipse errors and warnings:
[image]

Thanks for the help,
Rolland

kireevco, 2015-08-26:
Please try building from command line. If that works, proceed further.

I also suspect eclipse cached some wrong values. Try creating a new
workspace.

Airji3d, 2015-08-26:
Do you mean reinstalling Eclipse ?

This is the command line make result:
c:\tools\sming\Basic_Blink>make
 make: *** c:\tools\sming\Sming: Invalid request code. Stop.

Rolland

Airji3d, 2015-08-26:
Might be of interest, I get the error:

make: *** c:/tools/sming/Sming: Invalid request code.  Stop.

only when I uncomment the Sming_Home variable in the Makefile-user.mk.

Rolland

hreintke, 2015-10-05:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-05 by hreintke