SmingHub/Sming #234: Invalid arguments in application.cpp

Airji3d, 2015-08-26:
I get these errors when I build Basic_Blink (see picture attached).
The build and flash to ESP-01 worked and the led blinks but why these errors.

echo %ESP_HOME%
 c:\Espressif

echo %SMING_HOME%
 c:\tools\sming\Sming

echo %PATH%
C:\tools\mingw64\msys\1.0\bin;C:\ProgramData\Oracle\Java\javapath;C:\Program Fil
es (x86)\Common Files\Intel\Shared Files\cpp\bin\Intel64;C:\Program Files\Common
 Files\Microsoft Shared\Windows Live;C:\Program Files (x86)\Common Files\Microso
ft Shared\Windows Live;C:\Program Files\Dell\DW WLAN Card;;c:\Program Files (x86
)\Intel\iCLS Client\;c:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\
Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;c:\
Program Files\WIDCOMM\Bluetooth Software\;c:\Program Files\WIDCOMM\Bluetooth Sof
tware\syswow64;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files\Trend
 Micro\AMSP;C:\Program Files (x86)\Skype\Phone\;C:\ProgramData\chocolatey\bin;C:
\Program Files\Eclipse Foundation\4.4.2\eclipse;C:\Program Files\Java\jre1.8.0_5
1\bin;C:\tools\mingw64\msys\1.0\bin;

[image]

Thanks in advance for the help, 
Rolland

kireevco, 2015-08-26:
@Airji3d did you try building from a command line? `make` in your project directory?

Airji3d, 2015-08-26:
Yes I did try make from command line in my projects folder and it builds with no error.
My question is how do I fix the project path for the includes not found?
Why the invalid arguments in the app.cpp?

Rolland

kireevco, 2015-08-26:
Did you re-create eclipse workspace? (New Workspace)

Airji3d, 2015-08-26:
No I didn't do that, I'll try it and reply.

Airji3d, 2015-08-26:
Changed workspace and it removed the invalid path errors, thanks.
I still have the warnings
Invalid project path: Include path not found (\usr\include) etc.

Rolland

Airji3d, 2015-08-27:
OK found that it was a bad Eclipse workspace metadata setup.
Thanks Dmitry.

closed 2015-08-27 by Airji3d

perttierkkila, 2015-10-09:
I have same error quite randomly. It seems to be problem with C/C++ Indexes. Make works fine.

Point the project, right-click, Index -> Rebuild did the trick for me to get rid of those "Invalid arguments"-errors.