SmingHub/Sming #235: Do not turn off system Debug Output

STASIUK, 2015-08-27:
I ordered:
>Serial.systemDebugOutput(false); // Debug output to serial
yet still see output system information
~~~
fs.start: size:256 Kb, offset:0x%X
mount res: 0
Station configuration was updated to: PG
I'm CONNECTED
MQTT start connection
connect to: test.mosquitto.org
timeout updating: 70 -> 65535
subscription 'rx/module/#' registered
realloc 23 -> 170
DNS record found: test.mosquitto.org = 85.119.83.194
TcpConnection::connect result:, 0
TCP connection error: -8
MQTT closed previous connection
MQTT start connection
+TCP connection
connect to: test.mosquitto.org
timeout updating: 65535 -> 65535
subscription 'rx/module/#' registered
realloc 23 -> 170
DNS record found: test.mosquitto.org = 85.119.83.194
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 41
onReadyToSendData: 2
> MQTT status: MQTT_MSG_CONNACK (len: 4)
TCP received: 4 bytes
onReadyToSendData: 1
> MQTT status: MQTT_MSG_SUBACK (len: 5)
TCP received: 5 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
~~~

How do I turn it off?
The latest version sming

anakod, 2015-08-27:
It can be related to Debug implementation and https://github.com/anakod/Sming/pull/195
@hreintke can you look it, please?

hreintke, 2015-08-28:
It's related to the "Float support in printf, also for Wiring class"  #188 
In there the debugf definition is changed from
```
#define debugf(fmt, ...) os_printf(fmt"\r\n", ##__VA_ARGS__)
```
to
```
define debugf(fmt, ...) m_printf(fmt"\r\n", ##__VA_ARGS__)
```
The systemDebugOutout(false) redirects os_printf so no effect anymore on debugf.
Short term solution might be to undo this specific redefine. 
@ADiea : Can you have a look ?

ADiea, 2015-08-28:
@hreintke @STASIUK I will submit a pr to correct this

ADiea, 2015-08-28:
opened pr #241 for this issue

hreintke, 2015-09-08:
@STASIUK : a PR for thiswas merged. Please retest and close issue when solved

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke