SmingHub/Sming #236: SDK API functions available

hreintke, 2015-08-27:
@anakod : 
When answering a question on the esp8266/sming forum I discovered that sdk api functions like 
```
	wifi_station_set_reconnect_policy(true);
```
Are directly accessible by sming applications.
Earlier work from others on PWM sdk usage needed additional effort to get them linked.
And some of the wifi functions are made available thru  WifiStation and WifiAccesPoint classes.

What is/was your intention on this ?
Having direct access to sdk functions might give compatibility issues with new SDK versions

alonewolfx2, 2015-09-28:
actually its good thing when some functions not implemented yet but finally you ara right. it might give compatibility issues.

kireevco, 2015-10-30:
@anakod @raburton @robotiko 
What do you think?

raburton, 2015-10-30:
I don't see a big problem with users being able to call sdk functions, and I'm not sure how you'd stop them either!

hreintke, 2015-10-30:
At that time I was thinking of the following.
For the sdk api's we "really support" we create functions in the sming framework.
Like the proposed from flexiti in PR #371 .
```

sint8 StationClass::getRssi()
{
	debugf("Rssi: %d dBm", wifi_station_get_rssi());
	return wifi_station_get_rssi();
}
```
And  also f.e. the hw_pwm functions by implementing the HW_PWM class 
In that way all those functions are 
- put together in a logical way.
- easy accessible when using an IDE with code completion (just type Station. and you see the options)

For all other they are are available for the adventurous but not supported

closed 2015-11-14 by hreintke