SmingHub/Sming #240: Firmware size limited to 256KB

robotiko, 2015-08-27:
Flashing firmware bigger that 256KB causes exceptions in execution.
According to specs an other frameworks (rboot, arduino, etc)  the maximum size can go up to 1MB (fs limitation).
 
Related to  #221

anakod, 2015-08-27:
Can you provide test case, please?

> maximum size can go up to 1MB
I think it can be larger, just right implementation needed here.

alon24, 2015-08-27:
This is my code https://github.com/alon24/esp8266_projects 
it is currently over 256k

robotiko, 2015-08-27:
Thinking out loud. 
Can this also be related to not erased regions in flash?
I'm not sure if the flash erase is taking into account all the area that is available in big modules.

anakod, 2015-08-27:
Which flash modules you are using?

robotiko, 2015-08-27:
Using nodemcu DevKit 0.9 and 1.0.
esp-12 and esp-12e .

alon24, 2015-08-29:
If I use qio,  i do not get why reset. 
It is very important to not have delays in code of more than 250-300 ms,  because that causes a lot of war reset on its own,  my code now is 266 uses qio and does not reset

robotiko, 2015-08-29:
@alon24  I just checked that was using qio. 
Tried dio too with no luck. 
The code is not the issue, is blockless and there are no issues. It is just that after certain size in code, it starts to reset in a code that doesn't reset if reduced codebase. 
I know that this will be hard to find and fix.

robotiko, 2015-09-02:
This looks like the reason for this: 
http://www.esp8266.com/viewtopic.php?f=6&t=4131
"esp8266 crashes when code in .irom.text over 192k"

robotiko, 2015-09-04:
This seems to be solved with rboot version that is in PR

closed 2015-09-04 by robotiko