SmingHub/Sming #242: Interupt not working, when 3 weeks ago it did

alon24, 2015-08-28:
I have code for Rotary which uses interrupts.
In sming from 3 weeks ago it works perfectly, but now with latest it does not print anything, and does not work well.

I am thinking something changed, maybe with defing pins or something like that

#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define encoderA 13
#define encoderB 12

#define say(a) ( Serial.print(a) )
#define sayln(a) (Serial.println(a))

volatile int lastEncoded = 0;
volatile long encoderValue = 0;

long lastencoderValue = 0;
int lastValue =0;

int lastMSB = 0;
int lastLSB = 0;

Timer procTimer;

void IRAM_ATTR updateEncoder(){

  int MSB = digitalRead(encoderA); //MSB = most significant bit
  int LSB = digitalRead(encoderB); //LSB = least significant bit

  int encoded = (MSB << 1) |LSB; //converting the 2 pin value to single number
  int sum  = (lastEncoded << 2) | encoded; //adding it to the previous encoded value

  if(sum == 0b1101 || sum == 0b0100 || sum == 0b0010 || sum == 0b1011) encoderValue ++;
  if(sum == 0b1110 || sum == 0b0111 || sum == 0b0001 || sum == 0b1000) encoderValue --;

  lastEncoded = encoded; //store this value for next time
//  String s= "l=" + lastEncoded;
//  Serial.println(s);
}

void printStatus() {
	sayln(lastValue);
}

void displayChange() {
	if(lastValue != encoderValue/4) {
	    //Serial.println(lastValue);
	    printStatus();
	    lastValue = encoderValue/4;
	  }
//	printStatus();
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE); // 115200 or 9600 by default

//	delay(3000);
	say("======= Start Rotary Test ==========");
//	say( INT_PIN);
//	say( " low to trigger interrupt(s) =======");
	Serial.println();

	pinMode(encoderA, INPUT);
	pinMode(encoderB, INPUT);
	digitalWrite(encoderA, HIGH); //turn pullup resistor on
	digitalWrite(encoderB, HIGH); //turn pullup resistor on

	attachInterrupt(encoderA, updateEncoder, CHANGE);
	attachInterrupt(encoderB, updateEncoder, CHANGE);

	procTimer.initializeMs(100, displayChange).start();
}

If connecting a rotary to pins 12,13 (a,b) and the middle of the rotary to GND 
This code should print something like this

-1
-3
-4
-8
-12
-14
-17
-18
-16

Withe latest sming it does not, checked with SDK 1.2 and 1.3 (and with old sming using sdk1.2)

robotiko, 2015-08-28:
Hi @alon24,
can you put the commit id or dates of both sming versions mentioned?

alon24, 2015-08-28:
So this is from new code .git dir
```
0000000000000000000000000000000000000000 b51d8442ddd30f68acef2c5f0a76da68814bb632 alon24 <htpcfreak@gmail.com> 1431459991 +0300	clone: from https://github.com/anakod/Sming.git
b51d8442ddd30f68acef2c5f0a76da68814bb632 d410569b4e958104d3966ec6dabdf28e0b94f9cd alon24 <htpcfreak@gmail.com> 1431545944 +0300	pull --progress --rebase --prune origin master: checkout d410569b4e958104d3966ec6dabdf28e0b94f9cd
d410569b4e958104d3966ec6dabdf28e0b94f9cd d410569b4e958104d3966ec6dabdf28e0b94f9cd alon24 <htpcfreak@gmail.com> 1431545945 +0300	rebase finished: returning to refs/heads/master
d410569b4e958104d3966ec6dabdf28e0b94f9cd aba1287f50981a414c55bae99f3d2f572925e5cd alon24 <htpcfreak@gmail.com> 1433346951 +0300	pull --progress --rebase --prune origin master: checkout aba1287f50981a414c55bae99f3d2f572925e5cd
aba1287f50981a414c55bae99f3d2f572925e5cd aba1287f50981a414c55bae99f3d2f572925e5cd alon24 <htpcfreak@gmail.com> 1433346952 +0300	rebase finished: returning to refs/heads/master
aba1287f50981a414c55bae99f3d2f572925e5cd 40050b746e91d7ed34375f8562af7604c8a49921 alon24 <htpcfreak@gmail.com> 1434865589 +0300	pull --progress --rebase --prune origin master: checkout 40050b746e91d7ed34375f8562af7604c8a49921
40050b746e91d7ed34375f8562af7604c8a49921 40050b746e91d7ed34375f8562af7604c8a49921 alon24 <htpcfreak@gmail.com> 1434865590 +0300	rebase finished: returning to refs/heads/master
40050b746e91d7ed34375f8562af7604c8a49921 48cc0a46529ead3b3f2c199df84dec0985e5b078 alon24 <htpcfreak@gmail.com> 1435313648 +0300	pull --progress --rebase --prune origin master: checkout 48cc0a46529ead3b3f2c199df84dec0985e5b078
48cc0a46529ead3b3f2c199df84dec0985e5b078 48cc0a46529ead3b3f2c199df84dec0985e5b078 alon24 <htpcfreak@gmail.com> 1435313649 +0300	rebase finished: returning to refs/heads/master
48cc0a46529ead3b3f2c199df84dec0985e5b078 150c8c3ba6dd9420624eced50272e75f9ab2d23a alon24 <htpcfreak@gmail.com> 1436032318 +0300	pull --progress --rebase --prune origin master: checkout 150c8c3ba6dd9420624eced50272e75f9ab2d23a
150c8c3ba6dd9420624eced50272e75f9ab2d23a 150c8c3ba6dd9420624eced50272e75f9ab2d23a alon24 <htpcfreak@gmail.com> 1436032318 +0300	rebase finished: returning to refs/heads/master
150c8c3ba6dd9420624eced50272e75f9ab2d23a d2bb56b91ae315170c3c35800e9a3f2592998bb8 alon24 <htpcfreak@gmail.com> 1436328816 +0300	pull --progress --rebase --prune origin master: checkout d2bb56b91ae315170c3c35800e9a3f2592998bb8
d2bb56b91ae315170c3c35800e9a3f2592998bb8 d2bb56b91ae315170c3c35800e9a3f2592998bb8 alon24 <htpcfreak@gmail.com> 1436328818 +0300	rebase finished: returning to refs/heads/master
d2bb56b91ae315170c3c35800e9a3f2592998bb8 9482e6445e19277586b37f72ac78b659ac40a25a alon24 <htpcfreak@gmail.com> 1439150899 +0300	pull --progress --rebase --prune origin master: checkout 9482e6445e19277586b37f72ac78b659ac40a25a
9482e6445e19277586b37f72ac78b659ac40a25a 9482e6445e19277586b37f72ac78b659ac40a25a alon24 <htpcfreak@gmail.com> 1439150900 +0300	rebase finished: returning to refs/heads/master
9482e6445e19277586b37f72ac78b659ac40a25a 09828192c850d5f98a20196dc218cf77c733f315 alon24 <htpcfreak@gmail.com> 1440683204 +0300	pull --progress --rebase --prune origin master: checkout 09828192c850d5f98a20196dc218cf77c733f315
09828192c850d5f98a20196dc218cf77c733f315 09828192c850d5f98a20196dc218cf77c733f315 alon24 <htpcfreak@gmail.com> 1440683204 +0300	rebase finished: returning to refs/heads/master
```

and this is from old:
```
0000000000000000000000000000000000000000 b51d8442ddd30f68acef2c5f0a76da68814bb632 alon24 <htpcfreak@gmail.com> 1431459991 +0300	clone: from https://github.com/anakod/Sming.git
b51d8442ddd30f68acef2c5f0a76da68814bb632 d410569b4e958104d3966ec6dabdf28e0b94f9cd alon24 <htpcfreak@gmail.com> 1431545944 +0300	pull --progress --rebase --prune origin master: checkout d410569b4e958104d3966ec6dabdf28e0b94f9cd
d410569b4e958104d3966ec6dabdf28e0b94f9cd d410569b4e958104d3966ec6dabdf28e0b94f9cd alon24 <htpcfreak@gmail.com> 1431545945 +0300	rebase finished: returning to refs/heads/master
d410569b4e958104d3966ec6dabdf28e0b94f9cd aba1287f50981a414c55bae99f3d2f572925e5cd alon24 <htpcfreak@gmail.com> 1433346951 +0300	pull --progress --rebase --prune origin master: checkout aba1287f50981a414c55bae99f3d2f572925e5cd
aba1287f50981a414c55bae99f3d2f572925e5cd aba1287f50981a414c55bae99f3d2f572925e5cd alon24 <htpcfreak@gmail.com> 1433346952 +0300	rebase finished: returning to refs/heads/master
aba1287f50981a414c55bae99f3d2f572925e5cd 40050b746e91d7ed34375f8562af7604c8a49921 alon24 <htpcfreak@gmail.com> 1434865589 +0300	pull --progress --rebase --prune origin master: checkout 40050b746e91d7ed34375f8562af7604c8a49921
40050b746e91d7ed34375f8562af7604c8a49921 40050b746e91d7ed34375f8562af7604c8a49921 alon24 <htpcfreak@gmail.com> 1434865590 +0300	rebase finished: returning to refs/heads/master
40050b746e91d7ed34375f8562af7604c8a49921 48cc0a46529ead3b3f2c199df84dec0985e5b078 alon24 <htpcfreak@gmail.com> 1435313648 +0300	pull --progress --rebase --prune origin master: checkout 48cc0a46529ead3b3f2c199df84dec0985e5b078
48cc0a46529ead3b3f2c199df84dec0985e5b078 48cc0a46529ead3b3f2c199df84dec0985e5b078 alon24 <htpcfreak@gmail.com> 1435313649 +0300	rebase finished: returning to refs/heads/master
48cc0a46529ead3b3f2c199df84dec0985e5b078 150c8c3ba6dd9420624eced50272e75f9ab2d23a alon24 <htpcfreak@gmail.com> 1436032318 +0300	pull --progress --rebase --prune origin master: checkout 150c8c3ba6dd9420624eced50272e75f9ab2d23a
[... cut ...]  alon24 <htpcfreak@gmail.com> 1436032318 +0300	rebase finished: returning to refs/heads/master
150c8c3ba6dd9420624eced50272e75f9ab2d23a d2bb56b91ae315170c3c35800e9a3f2592998bb8 alon24 <htpcfreak@gmail.com> 1436328816 +0300	pull --progress --rebase --prune origin master: checkout d2bb56b91ae315170c3c35800e9a3f2592998bb8
d2bb56b91ae315170c3c35800e9a3f2592998bb8 d2bb56b91ae315170c3c35800e9a3f2592998bb8 alon24 <htpcfreak@gmail.com> 1436328818 +0300	rebase finished: returning to refs/heads/master
```

ADiea, 2015-08-29:
@alon24 can you isolate this better by halving the commits interval? 3weeks/2, 3weeks/4 etc

alon24, 2015-08-29:
I will try to use gut bisect soon

alon24, 2015-08-30:
so run bisect and got

```
C:\tools\Sming [(32d4357...)|BISECTING +1 ~1 -0 !]> git bisect bad
32d4357a091dde067d7eb2844cd487ebe3f69b18 is the first bad commit
commit 32d4357a091dde067d7eb2844cd487ebe3f69b18
Author: unknown <diea_andrei@yahoo.com>
Date:   Sun Jul 5 23:28:28 2015 +0300

    -default pullup; +pullup config compatible w/ Arduino ver > 1.0.1 and earlier versions

:040000 040000 f581f15c4bb1ac188285f72d8e7bbf8367070401 014f7fc29c95766b5a1464c49fcbd5134b5bb8f9 M      Sming
```

found previsous commit (and that worked):
```
C:\tools\Sming [(32d4357...) +1 ~0 -0 !]> git log
commit 32d4357a091dde067d7eb2844cd487ebe3f69b18
Author: unknown <diea_andrei@yahoo.com>
Date:   Sun Jul 5 23:28:28 2015 +0300

    -default pullup; +pullup config compatible w/ Arduino ver > 1.0.1 and earlier versions

commit 150c8c3ba6dd9420624eced50272e75f9ab2d23a
Merge: a3ff0b2 0964962
Author: Skurydin Alexey <anakod@gmail.com>
Date:   Thu Jul 2 19:23:08 2015 +0500

    Merge pull request #153 from shansted/master

    + Add Capsense Library and Example

```

so commit 150c8c3ba6dd9420624eced50272e75f9ab2d23a works ok.

ADiea, 2015-08-30:
The commit activates pullups when you write HIGH to a input pin just like on an arduino. 

Is it possible you don't drive the pin hard enough so that a pullup is enabled and the pin state doesn't change? 

Are you writing a HIGH to the input pin?

alon24, 2015-08-30:
of course :)

look at the code

digitalWrite(encoderA, HIGH); //turn pullup resistor on
    digitalWrite(encoderB, HIGH); //turn pullup resistor on

alon24, 2015-08-30:
maybe i missread?
the whole idea of a rotary is that its connected to GND and the 2 connectors (a,b) connected to a pullup'd io pin.

Example:
http://www.hobbytronics.co.uk/arduino-tutorial6-rotary-encoder

so this should work, and did work before that commit

alon24, 2015-08-30:
I changed my code to different pins - no good.
I added some more printouts, and see this:
```
======= Start Rotary Test ==========
starting with a=1, b=1
mode : softAP(1a:fe:34:a6:77:23)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
updateEncoderd a=0, b=1
updateEncoderd a=0, b=0
```
so it starts as high,high then a turn makes it 0,0 (because a turn is actually 4 states), it just does not get HIGH again.

could be the way its initialized in code. the pullup should always be there, could it be removed somehow after a revolution

alon24, 2015-08-30:
so I gather I answered my question:
```
void digitalWrite(uint16_t pin, uint8_t val)
{
	//make compatible with Arduino < version 100
	//enable pullup == setting a pin to input and writing 1 to it
	if (isInputPin(pin))
	{
		if(val == HIGH)
			pullup(pin);
		else
			noPullup(pin);
	}
	else
	{
		if (pin != 16)
			GPIO_REG_WRITE((((val != LOW) ? GPIO_OUT_W1TS_ADDRESS : GPIO_OUT_W1TC_ADDRESS)), (1<<pin));
		else
			WRITE_PERI_REG(RTC_GPIO_OUT, (READ_PERI_REG(RTC_GPIO_OUT) & (uint32)0xfffffffe) | (uint32)(val & 1));

		//GPIO_OUTPUT_SET(pin, (val ? 0xFF : 00));
	}
}
'''

This seems to remove the pullup, when it gets a low, and that kills my rotary....

CAn you take a look at the code there - digital.cpp

alon24, 2015-08-30:
this is the origin of my rotary code
http://bildr.org/2012/08/rotary-encoder-arduino/

ADiea, 2015-08-30:
hi, you are correct, i will try making a PR maybe tonight

alon24, 2015-08-30:
I tried to fix this,  but it seems to be another issue,  it is not called when the rotary changes  so it's not being put to nopullup. 
This needs to be fixed,  but it is not the fix I need,  do u know what happens on interrupt?  When move g to low,?  I could not find it.

alon24, 2015-08-31:
I found the other problem, but this one needs some thought.
It seems that in Interrups.cpp, it sets mode to input:

```
void attachInterruptHandler(uint8_t pin, GPIO_INT_TYPE mode)
{
	ETS_GPIO_INTR_DISABLE();

	if (!_gpioInterruptsInitialied)
	{
		ETS_GPIO_INTR_ATTACH((void*)interruptHandler, NULL); // Register interrupt handler
		_gpioInterruptsInitialied = true;
	}

	pinMode(pin, INPUT);

	gpio_pin_intr_state_set(GPIO_ID_PIN(pin), mode); // Enable GPIO pin interrupt

	ETS_GPIO_INTR_ENABLE();
}

void interruptMode(uint8_t pin, GPIO_INT_TYPE type)
{
	ETS_GPIO_INTR_DISABLE();

	pinMode(pin, INPUT);

	gpio_pin_intr_state_set(GPIO_ID_PIN(pin), type);

	ETS_GPIO_INTR_ENABLE();
}
```
I checked this when using the original commit that broke, so it could be a bit different in latest code.
How to solve? Just have it ask the digital what is the pin mode?

alon24, 2015-08-31:
I realy do not understand how to create a PR :( so here is the patch I made, can you please check it, and check it in?
```
diff --git a/Sming/SmingCore/Digital.cpp b/Sming/SmingCore/Digital.cpp
index bf5bdf6..86f657a 100644
--- a/Sming/SmingCore/Digital.cpp
+++ b/Sming/SmingCore/Digital.cpp
@@ -8,6 +8,8 @@
 #include "../SmingCore/Digital.h"
 #include "../Wiring/WiringFrameworkIncludes.h"
 
+static bool gIsPinPullUp[TOTAL_PINS+1] = {0};
+
 void pinMode(uint16_t pin, uint8_t mode)
 {
 	if (pin < 16)
@@ -23,6 +25,10 @@
 	}
 	else if (pin == 16)
 	{
+		if(mode == INPUT_PULLUP) {
+			gIsPinPullUp[pin] = true;
+		}
+
 		if (mode == INPUT  || mode == INPUT_PULLUP)
 		{
 		    WRITE_PERI_REG(PAD_XPD_DCDC_CONF,
@@ -59,6 +65,18 @@
 	}
 }
 
+uint8_t getPinMode(uint16_t pin) {
+	if(!isInputPin(pin)) {
+		return OUTPUT;
+	}
+
+	if (gIsPinPullUp[pin]) {
+		return INPUT_PULLUP;
+	}
+
+	return INPUT;
+}
+
 //Detect if pin is input
 bool isInputPin(uint16_t pin)
 {
diff --git a/Sming/SmingCore/Digital.h b/Sming/SmingCore/Digital.h
index 149aa75..d06b62d 100644
--- a/Sming/SmingCore/Digital.h
+++ b/Sming/SmingCore/Digital.h
@@ -23,6 +23,7 @@
 void noPullup(uint16_t pin);
 
 bool isInputPin(uint16_t pin);
+uint8_t getPinMode(uint16_t pin);
 
 unsigned long pulseIn(uint16_t pin, uint8_t state, unsigned long timeout =
 		1000000L);
diff --git a/Sming/SmingCore/Interrupts.cpp b/Sming/SmingCore/Interrupts.cpp
index 3977e3d..6d8bdd5 100644
--- a/Sming/SmingCore/Interrupts.cpp
+++ b/Sming/SmingCore/Interrupts.cpp
@@ -51,7 +51,9 @@
 		_gpioInterruptsInitialied = true;
 	}
 
-	pinMode(pin, INPUT);
+	if (getPinMode(pin) == OUTPUT) {
+		pinMode(pin, INPUT);
+	}
 
 	gpio_pin_intr_state_set(GPIO_ID_PIN(pin), mode); // Enable GPIO pin interrupt
 
@@ -75,7 +77,9 @@
 {
 	ETS_GPIO_INTR_DISABLE();
 
-	pinMode(pin, INPUT);
+	if (getPinMode(pin) == OUTPUT) {
+		pinMode(pin, INPUT);
+	}
 
 	gpio_pin_intr_state_set(GPIO_ID_PIN(pin), type);
 
```

ADiea, 2015-08-31:
@alon24 there is another improvement on initial pullup code here #163 I think this one checks for the pin to be input so the part for digitalpin fix is there, only interrupt needs to be thought over, right?

ADiea, 2015-08-31:
@alon24 sure i'll take a look on your patch

alon24, 2015-08-31:
yes I think so, master after integrate with 195, I am still geting used to GIT

ADiea, 2015-08-31:
@alon24 i think there is a better way.

Please try to modify Digital.cpp and remove these lines from void pinMode(uint16_t pin, uint8_t mode)

```
	else if (mode == INPUT)
	{
		noPullup(pin);
	}
```

I think it is wrong to always setup the pullup on/off every time the pin is set to input. Is your issue solved if you do this modification?

alon24, 2015-08-31:
1. I agree it's badba but I did not know why it was done so i flowed with it. 
2.attschinterrupt  redefines the pin and that is not good  I gave a solution for that too.

ADiea, 2015-08-31:
i think we need as little code modified to avoid introducing other bugs. Now, even if attachinterrupt sets the pin as input again it will not matter the pullups will not be changed, also less code in my opinion...

alon24, 2015-08-31:
What about:
```
void digitalWrite(uint16_t pin, uint8_t val)
{
	//make compatible with Arduino < version 100
	//enable pullup == setting a pin to input and writing 1 to it
	if (isInputPin(pin))
	{
		if(val == HIGH)
			pullup(pin);
		else
			noPullup(pin);
	}
```
This needs to come out too, but you guys put it there on purpose?

alon24, 2015-08-31:
new patch
```
diff --git a/Sming/SmingCore/Digital.cpp b/Sming/SmingCore/Digital.cpp
index bf5bdf6..d222ea9 100644
--- a/Sming/SmingCore/Digital.cpp
+++ b/Sming/SmingCore/Digital.cpp
@@ -53,10 +53,6 @@
 	{
 		pullup(pin);
 	}
-	else if (mode == INPUT)
-	{
-		noPullup(pin);
-	}
 }
 
 //Detect if pin is input
@@ -77,24 +73,24 @@
 
 void digitalWrite(uint16_t pin, uint8_t val)
 {
-	//make compatible with Arduino < version 100
-	//enable pullup == setting a pin to input and writing 1 to it
-	if (isInputPin(pin))
-	{
-		if(val == HIGH)
-			pullup(pin);
-		else
-			noPullup(pin);
-	}
-	else
-	{
+//	//make compatible with Arduino < version 100
+//	//enable pullup == setting a pin to input and writing 1 to it
+//	if (isInputPin(pin))
+//	{
+//		if(val == HIGH)
+//			pullup(pin);
+//		else
+//			noPullup(pin);
+//	}
+//	else
+//	{
 		if (pin != 16)
 			GPIO_REG_WRITE((((val != LOW) ? GPIO_OUT_W1TS_ADDRESS : GPIO_OUT_W1TC_ADDRESS)), (1<<pin));
 		else
 			WRITE_PERI_REG(RTC_GPIO_OUT, (READ_PERI_REG(RTC_GPIO_OUT) & (uint32)0xfffffffe) | (uint32)(val & 1));
 
 		//GPIO_OUTPUT_SET(pin, (val ? 0xFF : 00));
-	}
+//	}
 }
 
 uint8_t digitalRead(uint16_t pin)
diff --git a/Sming/compiler/lib/libsming.a b/Sming/compiler/lib/libsming.a
index 612bf37..0fc457f 100644
--- a/Sming/compiler/lib/libsming.a
+++ b/Sming/compiler/lib/libsming.a
Binary files differ

```

ADiea, 2015-08-31:
No, the stuff in digitalWrite is ok. This is arduino default behavior when you have an input pin if you write to it 1/0 it affects it's pullup. Newer arduino versions have the INPUT_PULLUP value defined but older versions only have INPUT/OUTPUT 

This is needed by some plugins that were ported from arduino to work as intended.

Is this working ok fro you if you only remove the else in the pinMode()?

```
-   else if (mode == INPUT)
-   {
-       noPullup(pin);
-   }
```

alon24, 2015-08-31:
no it does not work, because when attachinterrupt is called, I think that when it cals the setting of the pin to inout (again) it somehow resets the pullup state

alon24, 2015-08-31:
@ADiea I had a different isue, this minimal change seems to work ok.

so can you please check it in?

ADiea, 2015-08-31:
Just submitted PR #247 

Steps to contribute are like this:
1. fork git repository - done once
Next, you want to work on a feature
2. sync your master branch https://help.github.com/articles/syncing-a-fork/
3. create new branch and work in it (git checkout -b new_feature)
4. when you are happy with your fix, synch your working branch like on 2. except you will be working in your feature branch and the merge will be performed in your feature branch ( so git checkout new_feature instead of git checkout master)
5. make final commit/push to your github forked repository, go to github.com you will see a green button "compare and make pull request" -> clik it add a description and start the PR
6. after discussing if you need to work on the fix some more make the changes in your feature branch then commit + push and the pull request will be updated automatically
7. someone with writing rights like anakod will decide to merge your fix into the original anakod/Sming repository 
8. now you must resynch your master of your forked repo (2)

PR #247 (2015-08-31): input fix: remove clear of pullup when not explicitly cleared
Pullup is cleared when pin set to input even if not explicitly cleared by digitalWrite(LOW).

This affects interrupts and other places where 
1. pin is input pullup is enabled 
2. pin is set to input again

Fixes #242
closed 2015-08-31 by anakod