SmingHub/Sming #243: Choco install and manual wiki links in windows install point to latest UDK that is not Sming compatible

robotiko, 2015-08-28:
Choco and wiki links for windows install point to "programs74.ru/get.php?file=EspressifESP8266DevKit"
Downloaded version is not controlled and now is getting version 2.0.7 that is not compatible with Sming.
Should point to 2.0.6 or a later compatible version when it comes.

kireevco, 2015-08-28:
Unfortunately there is no way to control the version of udk, unless the author of udk provides such functionality. 

This is one of the reasons why I wish we could have esp open sdk.

kennpie, 2015-08-28:
Just wondering - what is the compatibility problem with v2.07 of Mikhail's UDK? 

If it's the Espressif SDK version, it's very easy to change that manually... if we knew what version Sming wants.

robotiko, 2015-08-28:
@kennpie   i recall the problem with 2.0.7 is that it doesn't support c++ 
and needs some changes in sming make 
>"Updated compiler xtensa-lx106-gcc and xtensa-lx106-g++ up to version 5.1 (big thanks jcmvbkbc); ATTENTION! For correct build of your firmware, you must add the option -std=gnu90 in directive CFLAGS in Makefile"

@kireevco   I would point to the build in the repo:  
https://github.com/CHERTS/esp8266-devkit/releases/download/v2.0.6/Espressif-ESP8266-DevKit-v2.0.6-x86.exe

kireevco, 2015-08-28:
Sure, we can point it. no problem.

anakod, 2015-08-31:
Is that compiler attribute back compatible with old compiler version?

hreintke, 2015-11-29:
@kireevco : Can you check choco and if done, close this issue ?

hreintke, 2015-12-21:
Done

closed 2015-12-21 by hreintke