SmingHub/Sming #245: SD Card does not detect SD card

williamesp2015, 2015-08-30:
SD Card does not detect SD card and SPI read and write needs to change to digitalRead and digitalWrite. In disk initializing the SPI speead should be 100-400 khz and needs to be considered.

open-home-automation, 2015-09-08:
Can you try the 2 changes below? It works for me. The first change is because the wrong variable is returned in the function disk_initialize, the second is to slow down the SPI speed.
When those 2 are applied and you rebuild Sming (do a make rebuild in the inner Sming directory), followed by a make clean in the app directory to link with the new Sming library, you will probably be able to access your SD card.

```
diff --git a/Sming/Libraries/SDCard/SDCard.cpp b/Sming/Libraries/SDCard/SDCard.cpp
index f8dd0a1..b912dc5 100644
--- a/Sming/Libraries/SDCard/SDCard.cpp
+++ b/Sming/Libraries/SDCard/SDCard.cpp
@@ -403,7 +403,7 @@ DSTATUS disk_initialize (
 
        SDCardSPI->setDelay(SCK_NORMAL);
 
-       return s;
+       return Stat;
 }

diff --git a/Sming/SmingCore/SPI.cpp b/Sming/SmingCore/SPI.cpp
index dcdd2e2..183711d 100644
--- a/Sming/SmingCore/SPI.cpp
+++ b/Sming/SmingCore/SPI.cpp
@@ -47,7 +47,7 @@ void SPIClass::begin()
        // time length LOW level = (CPU clock / 10 / 2) ^ -1
        // Frequency calculation: 80Mhz / predivider / divider
        uint16 predivider = 2;  // (1 ... 8192)
-       uint8 divider = 4;              // (1 ... 64)
+       uint8 divider = 8;              // (1 ... 64)
        WRITE_PERI_REG(SPI_FLASH_CLOCK(id),
                (((predivider-1) & SPI_CLKDIV_PRE) << SPI_CLKDIV_PRE_S) |
                (((divider-1) & SPI_CLKCNT_N) << SPI_CLKCNT_N_S) |
```

ADiea, 2015-09-18:
ta the moment, I believe software spi is used for accessing the sdcard

hreintke, 2015-12-21:
No information for a long time. Closing the issue.

closed 2015-12-21 by hreintke