SmingHub/Sming #248: out/build/app_app.a(application.o):(.text._Z9connectOkv+0xc): undefined reference to `m_printf'

kteza1, 2015-09-02:
I'm trying `BasicWIFI` example on my mac with `esp-open-sdk-1.2` which is resulting in the above linking error. `debugf` seems to be the culprit. With all `debugf`s removed, example is working fine.

hreintke, 2015-09-02:
@kteza1 : Can you try rebuilding sming and use debugf again ?
@anakod : libsming.a in github is from 18 days ago. Should it be rebuild and stored on github after each commit ?

anakod, 2015-09-02:
WIll be fine to implement automatic rebuild (as depended project or somehow else).

hreintke, 2015-09-02:
But wouldn't that force all Sming "users" to compile the core/framework themselves ?
By including a libsming.a in the github, you (at least) give the impression that all can be used "out of the box". That is also true of compile/use/run the examples in github, of which some/most depend on the latest version of sming.

As you are the only one to commit to the "original" sming, you are the only possible one to update the libsming.a in the github directory. Or am I wrong ?

gustavobreunig, 2015-09-02:
I'm having this problem too. How can I solve?

kteza1, 2015-09-02:
@hreintke Rebuilding worked. Like you said, this is not an out of box experience as @anakod might have intended this to be.  I suppose rebuilt `libsming.a` needs to be pushed.

gustavobreunig, 2015-09-02:
By the way, rebuild worked too. Maybe we have to put this on the Wiki?

closed 2015-09-04 by kteza1