SmingHub/Sming #251: Undefined reference when adding a library

jmgiacalone, 2015-09-04:
Hello,

I am having trouble adding a library to my project. I can build all the example projects just fine.

I cloned https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library inside the Sming/Libraries folder and included in my project with:

#include <Libraries/Adafruit-PWM-Servo-Driver-Library/Adafruit_PWMServoDriver.h>
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

When I make the project I get the following error:
undefined reference to `Adafruit_PWMServoDriver::Adafruit_PWMServoDriver(unsigned char)'

I can include any of the existing Sming libraries into my project and it compiles just fine. I also tried creating a library with a simple class definition, but I get the same error.

I expect I have overlooked some simple setting, and would be grateful if someone could point me in the right direction to fix this.

Thanks,
Jean-Marc.

Tavalin, 2015-09-04:
First off, did you rebuilt sming-core after adding the library?

jmgiacalone, 2015-09-04:
Brilliant! Thank you very much. Should that be explained somewhere?

closed 2015-09-04 by jmgiacalone