SmingHub/Sming #259: out/build/app_app.a(application.o):(.text._Z17onMessageReceived6StringS_+0x20): undefined reference to `ArduinoJson::JsonObject::at(char const*) const'

kteza1, 2015-09-06:
I'm facing the above linker error with json library while checking if a key is present with the below function call

```
if(root.containsKey("test") == false)
```

Usage:

```
 StaticJsonBuffer<200> json_buff;
     JsonObject& root = json_buff.parseObject(switch_message);
 
     if(!root.success()){
         Serial.println("Parsing failed");
         return;
     }
 
     if(root.containsKey("test") == false){
         Serial.println("@@@@ No test key");
         return;
     }
```

But the library doc says that the method is defined.

https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON#check-values

Is sming library getting compiled with out this definition?

hreintke, 2015-09-06:
I don't know why the error message comes, probably because of type conversion.
Someone with more knowledge on that should investigate.

I could compile/link (not run) with changing : 
```
if(root.containsKey("test") == false)
```
to 
```
if(!root.at("test").success())
```
Which is identical functionality, so should be a workaround until solved by someone
```

kteza1, 2015-09-07:
@hreintke Thanks for the workaround. I'll use `at` till the error with `containsKey` is fixed. I'll leave this issue open.

piperpilot, 2015-09-21:
I think this is possibly because we are using a really outdated version of Arduino Json.  We are on 4.2 and the latest released is 5.0.3.  I see some specific references in the changeLog to fixing problems with ESP-8266.

kteza1, 2015-09-30:
May be we need to update. I'll try it out and send a pull request this weekend.

piperpilot, 2015-09-30:
Issue https://github.com/SmingHub/Sming/issues/159 is to update ArduinoJson.  Its probably worth the effort.  I was going to work on it, but if @kteza1 has time, go for it.

hreintke, 2015-11-29:
Arduinojson is upgraded to V5.05. If issue persists, please submit a new issue

closed 2015-11-29 by hreintke