SmingHub/Sming #267: HttpServer_WebSockets disconnect when ESP try to send binary data

williamesp2015, 2015-09-07:
I tested HttpServer_WebSockets example and I found two main issues 1) disconnecting ESP when EXP try to send binary data and 2) can not continuously send data string more than 5 times/sec the result is:"ERROR: undefined" and consequently disconnected.

williamesp2015, 2015-09-07:
Websocket is capable to send and receive data very fast (https://www.youtube.com/watch?v=8ISbmQTbjDI) please see https://github.com/cnlohr/ws2812esp8266 If Sming developers be interested, Sming Websocket could be the best platform for Websocket server and client

hreintke, 2015-11-12:
You will need to supply more information and log information to be able to check

I will close this issue, I the problems persist please submit a new one with additional info.

closed 2015-11-12 by hreintke