SmingHub/Sming #270: Request for COAP server and client

kteza1, 2015-09-08:
I believe CoAP should be present in Sming.

robotiko, 2015-09-08:
+1  CoAP is a great addition to IOT platform.

flexiti, 2015-09-12:
By the way :)

http://www.slideshare.net/paolopat/internet-ofthingsprotocolswar

HaknCo, 2015-12-09:
I started to look into this issue.
Lobaro CoAP Server (https://github.com/Lobaro/lobaro-coap) worked almost out-of-the-box.
You can find my code here https://github.com/HaknCo/sming-coap
By the way, currently I am working on a CoAP client, which is based on the NodeMCU code, which again seems to be based on the microcoap project.

kteza1, 2015-12-09:
@HaknCo That would be a great contribution to Sming

HaknCo, 2015-12-11:
Just finished the CoAP client !!!

Server and Client can be found here : https://github.com/HaknCo/sming-coap.
GET, POST, PUT, DELETE work.
OBSERVE and BLOCKWISE will follow hopefully.

Let me know if it works for you.

ystrem, 2015-12-11:
Hi, great work ! Currently I'm using MQTT, is there any big think about CoAP that I want to change it to CoAP ?

HaknCo, 2015-12-11:
It depends, as long as you are interested in PubSub, MQTT is a good protocol. But if you are interested in Internet resources, meaning if you have URIs,
you definitely need CoAP or HTTP.

CoAP follows REST principles, thus it easy to integrate into Web Applications through proxies.

KR

hreintke, 2016-03-25:
Closing. If cloapp client is finished please submit a PR for further discussion

closed 2016-03-25 by hreintke