SmingHub/Sming #278: Stability Issues with standalone buil: Makefile for Sming relies on SMING_HOME

gernoteger, 2015-09-12:
As mentioned  in PR #268, I had issues correctly defining SMING_HOME: when sming is installed in a nonstandard location, either the Makefiles has to be modified, or the location has to be injected from outside.
In said PR #268 this caused problems when new include directories where added.

gernoteger, 2015-09-12:
This problem can be easily overcome when defining the default SMING_HOME as the location of the Makefile. This would make the handling of multiple Sming installations for e.g. testing of PRs easier.
Something like that could help:
~~~
  MAKEFILE_DIR:= $(dir $(abspath $(lastword $(MAKEFILE_LIST))))
  SMING_HOME ?=$(MAKEFILE_DIR)
~~~

raburton, 2015-09-12:
I assumed that SMING_HOME was correctly set when I added the rBoot includes to the Makefile, but it may not be. In fact that variable isn't needed in the includes anyway, so the fix is trivial. There may be other good reasons for having SMING_HOME defined at this point which would require a broader fix (if so that is not related to the rBoot change), but I don't actually see it used anywhere.

commit c53fbcf8 (2015-09-15): makefile fix closes #278 plus other tweaks
closed 2015-09-27 by anakod