SmingHub/Sming #280: Creating good looking documentation

kteza1, 2015-09-13:
I'm using sming for my personal project and i think it doesn't have professional looking documentation. Good documentation will help take this fantastic work really forward.

###### What I think we should do

* Create a documentation folder which contains all the markdowns.
* Use tools like [mkdocs](http://www.mkdocs.org/) (or similar) to create a documentation site in `github pages` and just point to documentation site in `README` (Because this is the first page which is visible when you open the repo in github)

When ever new changes or features are added, just update your documentation markdowns and we can periodically deploy changes to `gh-pages`and the documentation site will be updated.

In `mkdocs` its as simple as `mkdocs gh-deploy`. I've experimented with `mkdocs` today and we can make the documentation [look similar to this](http://kteza1.github.io/hng-switchboard/).

If we need, maybe we can add github hooks to update documentation with every push we make in `master` branch.

I'm willing to know if there are any better ways to document this project.

kireevco, 2015-09-13:
What should be part of it - doxygen code comments. Every feature should be documented.
Documenting all features separately in markdown - is utopical. Code should speak for itself, IMHO.

> If we need, maybe we can add github hooks to update documentation with every push we make in master branch.

We need to do #156 first.

Related: #136

kteza1, 2015-09-13:
Yeah. Sorry. Not every change and feature. Maybe important stuff like 
* Setup  in different operating system. 
* Example code for important features. 
* APIs (doxygen will take care of it but they should be embedded into documentation)
* How to add new libraries.
* FAQs

kteza1, 2015-09-15:
Documentation, HOW TOs, FAQs might not be a priority now. I see lot of discussion going on about work flow. May be this is for a later time. Closing this.

closed 2015-09-15 by kteza1